espressif/esp-adf #9: ESP32-LyraT battery issues

glennhuber, 2018-04-25:
My ESP32-LyraT board does not work from a battery. It will not charge nor run. Anyone else having this problem?

mlongo95, 2018-04-25:
You need to make sure you're connecting the power pins appropriately, my battery JST connector pins did not match the pin layout on the board.

mlongo95, 2018-04-26:
Luckily it has protection circuitry. I discovered this when trying to connect a Lipo but it would not boot. I checked the pin out and came to find out it was reversed. No damage was done to the board, using Dupont connectors i was able to power the board from a power suplply/battery.

glennhuber, 2018-05-11:
I have two (2) boards and neither work. The battery leds are not illuminating.  Trace....
MONITOR
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5580
load:0x40078000,len:0
load:0x40078000,len:13764
entry 0x40078fb4
[0;32mI (28) boot: ESP-IDF v3.0-rc1-257-g489e98cf 2nd stage bootloader[0m
[0;32mI (28) boot: compile time 17:01:52[0m
[0;32mI (34) boot: Enabling RNG early entropy source...[0m
[0;32mI (34) boot: SPI Speed      : 40MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (72) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (80) boot: End of partition table[0m
[0;32mI (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x25dc0 (155072) map[0m
[0;32mI (147) esp_image: segment 1: paddr=0x00035de8 vaddr=0x3ffb0000 size=0x0356c ( 13676) load[0m
[0;32mI (153) esp_image: segment 2: paddr=0x0003935c vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (154) esp_image: segment 3: paddr=0x00039764 vaddr=0x40080400 size=0x068ac ( 26796) load[0m
[0;32mI (174) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x96320 (615200) map[0m
[0;32mI (390) esp_image: segment 5: paddr=0x000d6340 vaddr=0x40086cac size=0x12868 ( 75880) load[0m
[0;32mI (422) esp_image: segment 6: paddr=0x000e8bb0 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (436) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (436) boot: Disabling RNG early entropy source...[0m
[0;32mI (438) spiram: SPI RAM mode: flash 40m sram 40m[0m
[0;32mI (442) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (449) cpu_start: Pro cpu up.[0m
[0;32mI (453) cpu_start: Starting app cpu, entry point is 0x400813b0[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1352) spiram: SPI SRAM memory test OK[0m
[0;32mI (1352) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1353) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1359) heap_init: At 3FFB95F8 len 00026A08 (154 KiB): DRAM[0m
[0;32mI (1365) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1371) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1378) heap_init: At 40099514 len 00006AEC (26 KiB): IRAM[0m
[0;32mI (1384) cpu_start: Pro cpu start user code[0m
[0;32mI (1389) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (1397) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (77) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (139) HTTP_MP3_EXAMPLE: [ 1 ] Start audio codec chip[0m
[0;32mI (159) HTTP_MP3_EXAMPLE: [2.0] Create audio pipeline for playback[0m
[0;32mI (159) HTTP_MP3_EXAMPLE: [2.1] Create http stream to read data[0m
[0;32mI (159) HTTP_MP3_EXAMPLE: [2.2] Create i2s stream to write data to codec chip[0m
[0;32mI (169) HTTP_MP3_EXAMPLE: [2.3] Create mp3 decoder to decode mp3 file[0m
[0;32mI (169) HTTP_MP3_EXAMPLE: [2.4] Register all elements to audio pipeline[0m
[0;32mI (179) HTTP_MP3_EXAMPLE: [2.5] Link it together http_stream-->mp3_decoder-->i2s_stream-->[codec_chip][0m
[0;32mI (189) HTTP_MP3_EXAMPLE: [2.6] Setup uri (http as http_stream, mp3 as mp3 decoder, and default output is i2s)[0m
[0;32mI (199) HTTP_MP3_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network[0m
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5580
load:0x40078000,len:0
load:0x40078000,len:13764
entry 0x40078fb4
[0;32mI (28) boot: ESP-IDF v3.0-rc1-257-g489e98cf 2nd stage bootloader[0m
[0;32mI (28) boot: compile time 17:01:52[0m
[0;32mI (34) boot: Enabling RNG early entropy source...[0m
[0;32mI (34) boot: SPI Speed      : 40MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (50) boot: ## Label            Usage          Type ST O

krzychb, 2018-05-12:
Hi @glennhuber 

> I have two (2) boards and neither work. 

From the log it looks like you are running `pipeline_http_mp3` example. What do you mean the boards do not work?

>  �[0;32mI (199) HTTP_MP3_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network�[0m
> ets Jun 8 2016 00:22:57

The last log line before reset indicates the example is waiting for Wi-Fi. It will not play anything before connecting to a Wi-Fi access point and retrieving `http://dl.espressif.com/dl/audio/adf_music.mp3`file. 

Have you setup the Wi-Fi connection as suggested in [README.md](https://github.com/espressif/esp-adf/tree/3628b1b5437e84c0c7dd9bcdeba08c36d49b1043/examples/player/pipeline_http_mp3) ?

closed 2018-05-22 by glennhuber

theronic, 2019-03-26:
@glennhuber did you discover any other wiring issues? I'm having a hard time with the DSP chip on the LyraTD, and very few of the examples seem to be compatible. And can you advise on how you went about returning them / getting a refund?