espressif/esp-adf #10: Play_http_mp3_example plays Baidu speech synthesis

Laxiyan, 2018-04-27:
Hello，
I changed the http link to "https://ai.baidu.com/aidemo?type=tns2&idx=1&tex=hello&cuid=baidu_speech_demo&cod=2&lan=zh&ctp=1&pdt=1&spd=5&per=0&vol=5&pit=5" and the following error occurred,

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5604
load:0x40078000,len:0
load:0x40078000,len:13756
entry 0x40078fb4
I (28) boot: ESP-IDF v3.0-rc1-257-g489e98c 2nd stage bootloader
I (28) boot: compile time 09:49:44
I (32) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 40MHz
I (38) boot: SPI Mode       : DIO
I (42) boot: SPI Flash Size : 4MB
I (46) boot: Partition Table:
I (49) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 00100000
I (79) boot: End of partition table
I (83) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1d870 (120944) map
I (135) esp_image: segment 1: paddr=0x0002d898 vaddr=0x3ffb0000 size=0x02778 ( 10104) load
I (139) esp_image: segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x7f664 (521828) map
I (324) esp_image: segment 3: paddr=0x000af684 vaddr=0x3ffb2778 size=0x00aec (  2796) load
I (325) esp_image: segment 4: paddr=0x000b0178 vaddr=0x40080000 size=0x00400 (  1024) load
I (331) esp_image: segment 5: paddr=0x000b0580 vaddr=0x40080400 size=0x1282c ( 75820) load
I (371) esp_image: segment 6: paddr=0x000c2db4 vaddr=0x400c0000 size=0x00000 (     0) load
I (382) boot: Loaded app from partition at offset 0x10000
I (382) boot: Disabling RNG early entropy source...
I (383) cpu_start: Pro cpu up.
I (386) cpu_start: Starting app cpu, entry point is 0x40080ff8
I (0) cpu_start: App cpu up.
I (397) heap_init: Initializing. RAM available for dynamic allocation:
I (404) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (410) heap_init: At 3FFB91B0 len 00026E50 (155 KiB): DRAM
I (416) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (422) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (429) heap_init: At 40092C2C len 0000D3D4 (52 KiB): IRAM
I (435) cpu_start: Pro cpu start user code
I (117) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (180) HTTP_MP3_EXAMPLE: [ 1 ] Start audio codec chip
I (200) HTTP_MP3_EXAMPLE: [2.0] Create audio pipeline for playback
I (200) HTTP_MP3_EXAMPLE: [2.1] Create http stream to read data
I (200) HTTP_MP3_EXAMPLE: [2.2] Create i2s stream to write data to codec chip
I (210) HTTP_MP3_EXAMPLE: [2.3] Create mp3 decoder to decode mp3 file
I (220) HTTP_MP3_EXAMPLE: [2.4] Register all elements to audio pipeline
I (220) HTTP_MP3_EXAMPLE: [2.5] Link it together http_stream-->mp3_decoder-->i2s_stream-->[codec_chip]
I (230) HTTP_MP3_EXAMPLE: [2.6] Setup uri (http as http_stream, mp3 as mp3 decoder, and default output is i2s)
I (240) HTTP_MP3_EXAMPLE: [ 3 ] Start and wait for Wi-Fi network
I (360) phy: phy_version: 366.0, ba9923d, Dec 29 2017, 14:25:06, 0, 0
I (3640) HTTP_MP3_EXAMPLE: [ 4 ] Setup event listener
I (3640) HTTP_MP3_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (3640) HTTP_MP3_EXAMPLE: [4.2] Listening event from peripherals
I (3650) HTTP_MP3_EXAMPLE: [ 5 ] Start audio_pipeline
W (3650) AUDIO_ELEMENT: [http] RESUME:Element has not running,state:3,task_run:1
W (3670) AUDIO_ELEMENT: [mp3] RESUME:Element has not running,state:3,task_run:1
W (3670) AUDIO_ELEMENT: [i2s] RESUME:Element has not running,state:3,task_run:1
I (3680) HTTP_MP3_EXAMPLE: [ * ] Event received: src_type:131072, source:0x3ffb9bd8 cmd:8, data:0xd, data_len:4
I (3690) HTTP_MP3_EXAMPLE: [ * ] Event received: src_type:131072, source:0x3ffbffe8 cmd:8, data:0xd, data_len:4
I (3700) HTTP_MP3_EXAMPLE: [ * ] Event received: src_type:131072, source:0x3ffbde3c cmd:8, data:0xd, data_len:4
I (3710) HTTP_MP3_EXAMPLE: [ * ] Event received: src_type:131072, source:0x3ffbde3c cmd:8, data:0xc, data_len:4
W (4860) HTTP_STREAM: No more data,errno:9
Encountered error reading when MP3 init
E (4870) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN
W (4870) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
E (4880) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_ERROR
I (4880) HTTP_MP3_EXAMPLE: [ * ] Event received: src_type:131072, source:0x3ffb9bd8 cmd:10, data:0x0, data_len:0
I (4890) HTTP_MP3_EXAMPLE: [ * ] Event received: src_type:131072, source:0x3ffb9bd8 cmd:8, data:0xc, data_len:4
I (4900) HTTP_MP3_EXAMPLE: [ * ] Event received: src_type:131072, source:0x3ffbffe8 cmd:8, data:0x1, data_len:4
I (4910) HTTP_MP3_EXAMPLE: [ * ] Event received: src_type:131072, source:0x3ffb9bd8 cmd:8, data:0xe, data_len:4
I (4920) HTTP_MP3_EXAMPLE: [ * ] Event received: src_type:131072, source:0x3ffbde3c cmd:8, data:0xe, data_len:4
I (4930) HTTP_MP3_EXAMPLE: [ 6 ] Stop audio_pipeline
W (4940) AUDIO_PIPELINE: There are no listener registered
W (4960) PERIPH_WIFI: Wi-Fi disconnected from SSID OhhSmart, auto-reconnect enabled, reconnect after 5000 ms

jason-mao, 2018-05-01:
@Laxiyan I have checked your link , it's not mp3 URL.

Laxiyan, 2018-05-02:
**@jason-mao**  This is an audio file of Baidu Voice, a stream without MP3 headers.

MarsPlanet, 2018-08-30:
I have the same issue with AWS Polly, was there any resolution on this?

jason-mao, 2019-01-03:
@Laxiyan @MarsPlanet  Could you test the latest version to help confirm whether there is still have this problem?

closed 2019-05-22 by jason-mao