espressif/esp-adf #17: Support signal messenger protocol

renne, 2018-05-26:
The [Signal Protocol C Library](https://github.com/signalapp/libsignal-protocol-c) is a C-implementation of the Signal protocol used by the Signal messenger from Whisper Systems. Adding this with RTP/Opus as sink/source to the ESP-ADF would allow to stream audio securely and in high quality between the ESP32 and the Signal messenger on Smartphones and Desktops (e.g. ESP32 as door station, ...).

tuanpmt, 2018-06-07:
Thank you for the suggestion about this useful protocol, We will note as a new feature request and find out the capabilities of ESP32 and ADF for this protocol.

rdehuyss, 2019-07-28:
+1! Would buy one right now if there would be webrtc support.

sosaucily, 2020-08-13:
Would be a nice replacement for the SIP systems, which are very complex and difficult to support widely.

sosaucily, 2020-08-25:
Seems Amazon has a tech that is working with it's [Kenesis](https://docs.aws.amazon.com/kinesisvideostreams-webrtc-dg/latest/devguide/what-is-kvswebrtc.html) tech.

https://docs.aws.amazon.com/kinesisvideostreams-webrtc-dg/latest/devguide/kvswebrtc-sdk-c.html

If anyone gets this working on espressif hardware that would be interesting. Would love to see that.

closed 2022-02-08 by jason-mao