espressif/esp-adf #18: No example works from ADF

shaunakJagtap, 2018-05-30:
Hello Team Expressif,

	Recently we have brought a ESP-LyraTD-MSC which works excellent with IDF. I have a technical doubt around which I have tried and read everything but nothing worked out.
I think the issue is with I2C Driver. Can you please look into it and suggest if we are missing on something obvious?
No example works from ADF, for instance you can refer to following logs for Play_MP3 example project given in repository esp-adf/examples/get-started/play_mp3 :

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5568
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:13756
entry 0x40078fb4
[0;32mI (30) boot: ESP-IDF v3.0-rc1-257-g489e98cf 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 19:08:49[0m
[0;32mI (33) boot: Enabling RNG early entropy source...[0m
[0;32mI (36) boot: SPI Speed      : 40MHz[0m
[0;32mI (40) boot: SPI Mode       : DIO[0m
[0;32mI (44) boot: SPI Flash Size : 4MB[0m
[0;32mI (49) boot: Partition Table:[0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (59) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (67) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (74) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (82) boot:  3 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (89) boot:  4 ota_0            OTA app          00 10 00110000 00100000[0m
[0;32mI (97) boot:  5 ota_1            OTA app          00 11 00210000 00100000[0m
[0;32mI (104) boot: End of partition table[0m
[0;32mI (109) esp_image: segment 0: paddr=0x00110020 vaddr=0x3f400020 size=0x1eee8 (126696) map[0m
[0;32mI (162) esp_image: segment 1: paddr=0x0012ef10 vaddr=0x3ffb0000 size=0x01100 (  4352) load[0m
[0;32mI (164) esp_image: segment 2: paddr=0x00130018 vaddr=0x400d0018 size=0x20510 (132368) map[0m
[0;32mI (215) esp_image: segment 3: paddr=0x00150530 vaddr=0x3ffb1100 size=0x010e4 (  4324) load[0m
[0;32mI (217) esp_image: segment 4: paddr=0x0015161c vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (222) esp_image: segment 5: paddr=0x00151a24 vaddr=0x40080400 size=0x09de4 ( 40420) load[0m
[0;32mI (247) esp_image: segment 6: paddr=0x0015b810 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (254) boot: Loaded app from partition at offset 0x110000[0m
[0;32mI (254) boot: Disabling RNG early entropy source...[0m
[0;32mI (257) cpu_start: Pro cpu up.[0m
[0;32mI (261) cpu_start: Starting app cpu, entry point is 0x40080edc[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (271) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (278) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (284) heap_init: At 3FFB2A10 len 0002D5F0 (181 KiB): DRAM[0m
[0;32mI (290) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (296) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (303) heap_init: At 4008A1E4 len 00015E1C (87 KiB): IRAM[0m
[0;32mI (309) cpu_start: Pro cpu start user code[0m
[0;32mI (327) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (329) PLAY_MP3_FLASH: [ 1 ] Start audio codec chip[0m
[0;31mE (329) gpio: gpio_isr_handler_add(377): GPIO isr service is not installed, call gpio_install_isr_service() first[0m
[0;31mE (339) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (349) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (349) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (359) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (359) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (359) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (369) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (369) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (379) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (379) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (389) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (389) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (399) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (399) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (409) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (409) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (419) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (419) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (429) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (429) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (429) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (439) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (439) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (449) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (449) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (459) ES8388_DRIVER: es_read_reg error[0m
[0;31mE (459) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (469) ES8388_DRIVER: es_read_reg error[0m
[0;31mE (469) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (479) ES8388_DRIVER: es_read_reg error[0m
[0;31mE (479) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (489) ES8388_DRIVER: es_read_reg error[0m
[0;31mE (489) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (499) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (499) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (499) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (509) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (509) ES8388_DRIVER: es_read_reg error[0m
[0;31mE (519) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (519) ES8388_DRIVER: es_read_reg error[0m
[0;31mE (529) ES8388_DRIVER: es_write_reg error[0m
[0;31mE (529) ES8388_DRIVER: es_read_reg error[0m
[0;31mE (539) ES8388_DRIVER: es_write_reg error[0m
[0;32mI (539) PLAY_MP3_FLASH: [ 2 ] ------------------- shaunak :-------------------------Create audio pipeline, add all elements to pipeline, and subscribe pipeline event[0m
[0;32mI (559) PLAY_MP3_FLASH: [2.1] Create mp3 decoder to decode mp3 file and set custom read callback[0m
[0;32mI (569) PLAY_MP3_FLASH: [2.2] Create i2s stream to write data to codec chip[0m
[0;32mI (579) PLAY_MP3_FLASH: [2.3] Register all elements to audio pipeline[0m
[0;32mI (579) PLAY_MP3_FLASH: [2.4] Link it together [mp3_music_read_cb]-->mp3_decoder-->i2s_stream-->[codec_chip][0m
[0;32mI (589) PLAY_MP3_FLASH: [ 3 ] Setup event listener[0m
[0;32mI (599) PLAY_MP3_FLASH: [3.1] Listening event from all elements of pipeline[0m
[0;32mI (609) PLAY_MP3_FLASH: [ 4 ] Start audio_pipeline[0m
[0;32mI (629) PLAY_MP3_FLASH: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2[0m
[0;32mI (7409) PLAY_MP3_FLASH: [ 5 ] Stop audio_pipeline[0m
[0;33mW (7409) AUDIO_PIPELINE: There are no listener registered[0m

tuanpmt, 2018-05-30:
Hi @shaunakJagtap 

Currently, ADF does not support ESP-LyraTD-MSC yet. We are trying to support this board as soon as possible. The example that you run the test runs on the Lyra32T board, which have ES8388 codec chip, while ESP-LyraTD-MSC not.

Thanks

shaunakJagtap, 2018-07-10:
Hey @tuanpmt Thanks for the update, can I know approximately down the line when can we expect the support for ADF on ESP-LyraTD-MSC?

btashton, 2018-07-18:
@tuanpmt will it be leveraging the microsemi Timberwolf for wake word detection?

jason-mao, 2018-07-19:
@shaunakJagtap ESP-LyraTD-MSC will be available on two 2-3 weeks.

btashton, 2018-07-19:
@jason-mao Can you comment on how much of the wake word and phrase processing will take place in the microsemi part?  I met with one of their reps this week and am trying to figure out where things stand.

jason-mao, 2018-07-19:
@btashton We only have test the "Alexa" word in microsemi.

krzychb, 2018-10-22:
Hello @shaunakJagtap, support for ESP32-LyraTD-MSC has been added in fa6258204bb8324a21188bd744ed3d1e14a14964

robertoimai, 2018-10-24:
I tried running the play_mp3 example on ESP32-LyraTD-MSC and although the code runs fine, I get no sound.
I have v2.1 of MSC.
Just as a side note, I can hear sound and interact with the board just fine with the alexa_msc_sr example code in the avs-sdk release.

withdra, 2018-10-24:
Hi @robertoimai, please check to see if there is a problem with the version of the firmware. Try erasing the firmware of MSC board before running the code.

shaunakJagtap, 2018-10-24:
@withdra, @krzychb even after erasing the firmware, make clean and then flashing of the play_mp3 results same, no driver error after ADF support but no sound. I have ESP-LyraTD-MSC v2.1 . Here are the logs:

--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---

--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---

I (7863) PLAY_MP3_FLASH: [ 5 ] Stop audio_pipeline

W (7863) AUDIO_PIPELINE: There are no listener registered

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)

flash read err, 1000

ets_main.c 371 

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:2

load:0x3fff0018,len:4

load:0x3fff001c,len:5552

ho 0 tail 12 room 4

load:0x40078000,len:0

load:0x40078000,len:13756

entry 0x40078fb4

I (30) boot: ESP-IDF v3.0.1 2nd stage bootloader

I (30) boot: compile time 14:31:17

I (44) boot: Enabling RNG early entropy source...

I (44) boot: SPI Speed      : 40MHz

I (44) boot: SPI Mode       : DIO

I (46) boot: SPI Flash Size : 4MB

I (50) boot: Partition Table:

I (54) boot: ## Label            Usage          Type ST Offset   Length

I (61) boot:  0 nvs              WiFi data        01 02 00009000 00006000

I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000

I (76) boot:  2 factory          factory app      00 00 00010000 00100000

I (84) boot: End of partition table

I (88) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x4bfec (311276) map

I (206) esp_image: segment 1: paddr=0x0005c014 vaddr=0x3ffb0000 size=0x02230 (  8752) load

I (210) esp_image: segment 2: paddr=0x0005e24c vaddr=0x40080000 size=0x00400 (  1024) load

0x40080000: _iram_start at /Users/mmt/esp/esp-adf/esp-idf/components/freertos/./xtensa_vectors.S:1685

I (213) esp_image: segment 3: paddr=0x0005e654 vaddr=0x40080400 size=0x019bc (  6588) load

I (224) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0x24950 (149840) map

0x400d0018: _stext at ??:?

I (283) esp_image: segment 5: paddr=0x00084970 vaddr=0x40081dbc size=0x08df8 ( 36344) load

0x40081dbc: uart_tx_wait_idle at /Users/mmt/esp/esp-adf/esp-idf/components/esp32/./system_api.c:360

I (298) esp_image: segment 6: paddr=0x0008d770 vaddr=0x400c0000 size=0x00000 (     0) load

I (305) boot: Loaded app from partition at offset 0x10000

I (305) boot: Disabling RNG early entropy source...

I (308) cpu_start: Pro cpu up.

I (311) cpu_start: Starting app cpu, entry point is 0x40080f14

0x40080f14: call_start_cpu1 at /Users/mmt/esp/esp-adf/esp-idf/components/esp32/./cpu_start.c:215

I (0) cpu_start: App cpu up.

I (322) heap_init: Initializing. RAM available for dynamic allocation:

I (329) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM

I (335) heap_init: At 3FFB2A78 len 0002D588 (181 KiB): DRAM

I (341) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM

I (347) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM

I (354) heap_init: At 4008ABB4 len 0001544C (85 KiB): IRAM

I (360) cpu_start: Pro cpu start user code

I (42) cpu_start: Starting scheduler on PRO CPU.

I (0) cpu_start: Starting scheduler on APP CPU.

I (44) PLAY_MP3_FLASH: [ 1 ] Start audio codec chip

W (1044) SPI: MCS ret:0,Status:1

I (1044) PLAY_MP3_FLASH: [ 2 ] Create audio pipeline, add all elements to pipeline, and subscribe pipeline event

I (1044) PLAY_MP3_FLASH: [2.1] Create mp3 decoder to decode mp3 file and set custom read callback

I (1054) PLAY_MP3_FLASH: [2.2] Create i2s stream to write data to codec chip

I (1064) PLAY_MP3_FLASH: [2.3] Register all elements to audio pipeline

I (1064) PLAY_MP3_FLASH: [2.4] Link it together [mp3_music_read_cb]-->mp3_decoder-->i2s_stream-->[codec_chip]

I (1084) PLAY_MP3_FLASH: [ 3 ] Setup event listener

I (1084) PLAY_MP3_FLASH: [3.1] Listening event from all elements of pipeline

I (1094) PLAY_MP3_FLASH: [ 4 ] Start audio_pipeline

I (1134) PLAY_MP3_FLASH: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2

I (7864) PLAY_MP3_FLASH: [ 5 ] Stop audio_pipeline

W (7864) AUDIO_PIPELINE: There are no listener registered

withdra, 2018-10-24:
HI， @shaunakJagtap，from the logs, you did not re-burn the firmware of zl38063. put it another way, you did not  erase it successfully, because if it was detected that there was no firmware in zl38063, the code would re-burn the firmware, which would take about a dozen seconds, but if it was detected that the firmware already exists, it would print "W (1041) SPI: MCS ret: 0, Status: 1"

shaunakJagtap, 2018-10-24:
@withdra Thanks for quick response, for re-burning of the firmware of zl38063 **make erase_flash** command don't work is there any alternative ?

withdra, 2018-10-24:
I'm sorry I'm not clear enough. I mean you should erase the firmware of zl38063 instead of using make flash_erase to erase the flash of ESP32, but that requires software to erase it. That would be very troublesome. So I suggest you delete line 55 of / components / audio_hal / driver / zl38063 / examples_app / tw_spi_access. c and run the code again, which will burn the new firmware. If you can succeed after running, please give me feedback in time. Thank you!

withdra, 2018-10-24:
Hi, @shaunakJagtap I'm sorry I'm not clear enough. I mean you should erase the firmware of zl38063 instead of using make flash_erase to erase the flash of ESP32, but that requires software to erase it. That would be very troublesome. So I suggest you delete line 55 of / components / audio_hal / driver / zl38063 / examples_app / tw_spi_access. C and run the code again, which will burn the new firmware. If you can succeed after running, please give me feedback in time. Thank you!

shaunakJagtap, 2018-10-24:
@withdra I have removed line no 55 having  **return 0;** but then make flash is failing with following errors:

CC build/audio_stream/fatfs_stream.o
CC build/audio_stream/http_stream.o
CC build/audio_stream/i2s_stream.o
/Users/mmt/esp/esp-adf/components/audio_stream/./i2s_stream.c: In function '_i2s_close':
/Users/mmt/esp/esp-adf/components/audio_stream/./i2s_stream.c:149:9: error: implicit declaration of function 'i2s_write' [-Werror=implicit-function-declaration]
         i2s_write(i2s->config.i2s_port, (char *)buf, i2s->config.i2s_config.dma_buf_len * 4, &bytes_written, portMAX_DELAY);
         ^
/Users/mmt/esp/esp-adf/components/audio_stream/./i2s_stream.c: In function '_i2s_read':
/Users/mmt/esp/esp-adf/components/audio_stream/./i2s_stream.c:168:5: error: implicit declaration of function 'i2s_read' [-Werror=implicit-function-declaration]
     i2s_read(i2s->config.i2s_port, buffer, len, &bytes_read, ticks_to_wait);
     ^
cc1: some warnings being treated as errors
make[1]: *** [i2s_stream.o] Error 1
make: *** [component-audio_stream-build] Error 2

withdra, 2018-10-24:
@shaunakJagtap I don't think deleting this line will cause these problems. Would it work if you added return 0? I use this example to run normally in MSC_V2.1.

shaunakJagtap, 2018-10-24:
@withdra No in both cases it wont work. I deleted that line because you suggested to delete line 55 of / components / audio_hal / driver / zl38063 / examples_app / tw_spi_access. C

sagar-patil-52, 2018-10-24:
@withdra I am not able to hear the audio as well. As per your suggestions , I have removed line no 55 and now I am getting following errors these were not there earlier.

I (359) cpu_start: Pro cpu start user code
I (42) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (44) PLAY_MP3_FLASH: [ 1 ] Start audio codec chip
E (44) ES8388_DRIVER: es_write_reg error
E (44) ES8388_DRIVER: es_write_reg error
E (54) ES8388_DRIVER: es_write_reg error
E (54) ES8388_DRIVER: es_write_reg error
E (64) ES8388_DRIVER: es_write_reg error

E (304) gpio: gpio_install_isr_service(406): GPIO isr service already installed

robertoimai, 2018-10-24:
When you say re-burning the firmware, are you talking about the zl38063 that goes into the spi flash that is connected red to the spi master of zl38063?

Sincerely,
Roberto Imai

robertoimai, 2018-10-25:
I have to withdraw my issue.

The problem I had was that my adf repo is modified to use a 4th type of board, which is a custom  board I use with another codec chip.

This caused the example code to break because it was trying to initialize index 2 in this line:

audio_hal_handle_t hal = audio_hal_init(&audio_hal_codec_cfg, 2);

Well, index 2 is my custom board.

So, when I changed it to 3, everything started working fine.

Sorry for the red alert. My issue is resolved.

Sincerely,

Roberto Imai

robertoimai, 2018-10-25:
I have another problem with the zl38063 driver though.
Should I open a new thread?
On VprocTwolfGetAppStatus, the function VprocTwolfHbiRead(0x030, 1, &buf) puts value 0x6000 into buf, but because status gets only the first digit, it returns status=0, which causes the firmware to be loaded again.
So, in my case, the firmware gets loaded everytime I reboot ESP32.
Is this expected behavior or there is something that should allow the firmware to be bypassed if everything is properly loaded?
Just as a test, I returned this:
*status  = buf;
And I got sound just fine without the extra firmware upload overhead.
It was an instant sound play, which leads me to believe there is something wrong with the firmware check.
Can you comment, please?

withdra, 2018-10-25:
@robertoimai In fact, it returns status = 1 when it detects that firmware already exists, but sometimes it's a playback exception due to firmware reasons, so I recommend deleting retrun 0 and running the code in the comments above. In this way, the program will burn the correct firmware into the board. If there is already the correct firmware in the board, then it is not necessary to burn it again every time.The function will return in one second rather than burn the firmware again.

robertoimai, 2018-10-25:
I never get there anyway.
when VprocTwolfGetAppStatus(&vol) is called, I get vol=0, which doesn't let the code go to line 55 and return 0.
The reason I get vol=0 is because VprocTwolfHbiRead(0x030, 1, &buf) always puts 0x6000 into buf.
The *status = buf &0xf makes status be set to 0 and therefore vol=0.
Is 0x6000 the wrong value to get from VprocTwolfHbiRead(0x030, 1, &buf)?

theronic, 2019-03-05:
I bought two ESP32-LYRATD-MSC v2.2 boards specifically to capture audio, but the audio recording examples don't work (`es_write_reg` errors). Has anyone gotten this working with the LYRATD-MSC?

jabadabado, 2019-03-05:
I bought it too, but it speaks Chinese!!?? Useless.

RemyEE, 2019-03-06:
@theronic Have you tried to select your board in `make menuconfig`? The library works fine with Lyratd-msc for me.

theronic, 2019-03-06:
@RemEE, I did configure via `make menuconfig`. The problem was the codec. Posted [solution here](https://stackoverflow.com/a/55026181/198927) which I lifted from another example.

shaunakJagtap, 2019-03-07:
> Hi, @shaunakJagtap I'm sorry I'm not clear enough. I mean you should erase the firmware of zl38063 instead of using make flash_erase to erase the flash of ESP32, but that requires software to erase [...]

@withdra , sorry for late reply. This solution works fine. Older version(v2.1 in my case) for first time we need to forcefully flash the zl38063 firmware as you mentioned, after that we can clear the change and everything works absolutely fine!

closed 2019-03-07 by shaunakJagtap

theronic, 2019-03-26:
@shaunakJagtap how do you flash the zl38063 firmware? I have a LyraTD-MSC v2.2 and am having a hard time with the DSP chip.