espressif/esp-adf #19: Using ESP-ADF with Arduino

HumamHelfawi, 2018-06-05:
Hi,
Is it possible to use ESP-ADF from Arudino IDE with other Arduino libraries? I used to use ESP-IDF from Arduino IDE. However, I could not find any documentation about using ESP-ADF from Arduino.

tuanpmt, 2018-06-07:
I think the better way to use ADF with Arduino is using Arduino for ESP32 as IDF component: https://github.com/espressif/arduino-esp32/blob/master/docs/esp-idf_component.md

It means that you can compile examples of the current ADF, and add Arduino features, libraries to ADF application.

HumamHelfawi, 2018-06-07:
@tuanpmt Thanks for your reply! Yes this the only way it worked with me.

closed 2018-06-09 by HumamHelfawi

unltdsoul, 2019-01-16:
I am trying to use Arduino as component of ADF - but only get errors - I have successfully used Arduino as component of IDF.  
Errors I get:  
execvp: xtensa-esp32-elf-gcc: Argument list too long
make[1]: *** [/home/ameyatma/esp/esp-adf/esp-idf/make/component_wrapper.mk:274: cores/esp32/libb64/cencode.o] Error 127
make: *** [c:/msys32/home/ameyatma/esp/esp-adf/esp-idf/make/project.mk:450: component-arduino-build] Error 2

I have reloaded in ADF - i can build without Arduino, or can use Arduino as component of IDF, but, not with ADF -  any idea?

mgsscorr1, 2019-01-17:
From what ive seen on another forum is that ESP has no plan to make ADF compatibly with Arduino, but seems like the guys above^^^ have found a slight work around

HumamHelfawi, 2019-01-17:
> I am trying to use Arduino as component of ADF - but only get errors - I have successfully used Arduino as component of IDF.
> Errors I get:
> execvp: xtensa-esp32-elf-gcc: Argument list too long [...]

I have faced the same issue at some point while I was updating the version of AFD and Arduino. As far as I remember (I am note sure actually), the problem was that the Arduino version (commit actually) was compatible. So, you have to make sure that ADF, IDF and Arduni-esp32 commits you are using are the same the repos suggest in their sub-modules.

If you clone the last commit of each repo, it won't work.

unltdsoul, 2019-01-17:
Thank, HumanHelfawi, I'll look into that when i get time, and will post here if successful letting others know how, or if, i got it to work.

maharshi114, 2021-12-31:
Hey, @tuanpmt page not found in this comment, link: https://github.com/espressif/esp-adf/issues/19#issuecomment-395490888

Thank you.

Droxx, 2024-06-12:
It's moved here: https://github.com/espressif/arduino-esp32/blob/master/docs/en/esp-idf_component.rst

sheetaljain01, 2024-10-15:
Hi, have u completed successfully?