espressif/esp-adf #23: automatic sample_rate

HumamHelfawi, 2018-06-09:
Hi,
I was playing this example **play_mp3_dac** on ESP32 board. It works. However, I have some questions:

The macro `I2S_STREAM_INTERNAL_DAC_CFG_DEFAULT()` is defined as:
```
#define I2S_STREAM_INTERNAL_DAC_CFG_DEFAULT() {                                     \
    .type = AUDIO_STREAM_WRITER,                                                    \
    .i2s_config = {                                                                 \
        .mode = I2S_MODE_MASTER | I2S_MODE_DAC_BUILT_IN | I2S_MODE_TX,              \
        .sample_rate = 44100,                                                       \
        .bits_per_sample = 16,                                                      \
        .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,                               \
        .communication_format = I2S_COMM_FORMAT_I2S_MSB,                            \
        .dma_buf_count = 3,                                                         \
        .dma_buf_len = 300,                                                         \
        .intr_alloc_flags = ESP_INTR_FLAG_LEVEL2,                                   \
    },                                                                              \
    .i2s_port = 0,                                                                  \
}
```

The sample rate is set to `sample_rate = 44100`. However, this causes problem when the MP3 file is not 44.1 KHZ. It plays the Music much faster if it was not 44.1KHZ. 

My question is: **how to  dynamically set the sample_rate to be exactly as the Mp3 file sample rate without hard-coding it.**

One another question: after `calling audio_pipeline_run(pipeline_);`. The Mp3 starts till it ends. **How can I have control over it? Does it open some kind of thread?**

krzychb, 2018-06-09:
HI @HumamHelfawi,

> My question is: how to dynamically set the sample_rate to be exactly as the Mp3 file sample rate without hard-coding it.

The sample rate of I2S stream is set dynamically be the following code:
https://github.com/espressif/esp-adf/blob/ad50e25e05b37fcaeaac3167ce8d8f5c1b49ea8c/examples/get-started/play_mp3_dac/main/play_mp3_dac_example.c#L94-L101

> One another question: after calling audio_pipeline_run(pipeline_);. The Mp3 starts till it ends. How can I have control over it? Does it open some kind of thread?

Control over played music is demonstrated in [play_mp3_control](https://github.com/espressif/esp-adf/tree/ad50e25e05b37fcaeaac3167ce8d8f5c1b49ea8c/examples/get-started/play_mp3_control) example.

HumamHelfawi, 2018-06-11:
Thank you very much for the fast and detailed answer!

closed 2018-06-11 by HumamHelfawi