espressif/esp-adf #26: audio_element_deinit(i2s_stream_writer); infinite loop

HumamHelfawi, 2018-06-12:
Hi,
I am trying to mimic the play_mp3_dac example (which works correctly Commit: f6d007ae20a8c0e42b76bf7553257c1717b8fd95 ). I am almost sure that I have done everything as it was written in the example. However, after the Mp3 is played and finished, I call :
audio_pipeline_terminate(pipeline_);
```
	audio_pipeline_remove_listener(pipeline_);
	audio_event_iface_destroy(evt_);
	audio_pipeline_deinit(pipeline_);
	audio_element_stop(i2s_stream_writer_);
	audio_element_deinit(i2s_stream_writer_);
	audio_element_deinit(mp3_decoder_);	
```
At this line:
`audio_element_deinit(i2s_stream_writer_);`
The program stops responding. I went into  `audio_element_deinit` code and I found that the problem is exactly at this line:
`audio_element_wait_for_stop(el);`

I commented it out and everything worked fine. However, I am sure that at some points my programm will explode because it is trying to free something while it is still working. 

May you please give me any hint why does this happen?

Thank you very much!

jason-mao, 2018-06-13:
@HumamHelfawi 
Could you please remove the `audio_element_stop(i2s_stream_writer_);` test again. Then I will take a look as your usage, looks like `audio_element_stop ` need to improved.

Thanks.

HumamHelfawi, 2018-06-13:
@jason-mao  I forgot to mention that I had the same problem without `audio_element_stop `. I added it hopping that it may change the behavior but it did not.

ahhfzhang, 2018-06-19:
@HumamHelfawi i cann't reproduce this issue on latest esp-adf,maybe it's already been fixed?

jason-mao, 2019-01-03:
@HumamHelfawi Did you met this problem on latest version ?

closed 2019-05-22 by jason-mao