espressif/esp-adf #27: I2S0 & I2S1  issue

simonlou99, 2018-06-13:
We want to use I2S0 & I2S1  at the same time. I2S0 works at the master mode to play music. and I2S1 works at the slave mode to record ; How to set i2s, is there any reference code?
Please kindly help us to fix the issue .

jason-mao, 2018-06-19:
Hi @simonlou99 

Please refer to
https://github.com/espressif/esp-adf/blob/master/components/audio_stream/include/i2s_stream.h#L48.
`.mode = I2S_MODE_MASTER | I2S_MODE_TX  `  for master playback.
`.mode = I2S_MODE_SLAVE | I2S_MODE_RX` for slave recording.

BTW, ESP32 have to provide MCLK to master at slave mode.

closed 2018-11-09 by jason-mao

crespum, 2019-03-26:
Hi @jason-mao, is there any way to sync the clocks without sending MCLK to the master? In the technical reference there's a mention to a bit named `I2S_TX_DSYNC_SW`, which "synchronizes signals into the transmitter in double sync method". Can you share more info about it?