espressif/esp-adf #30: pipeline_sdcard_mp3 Fail to open file

RemyEE, 2018-06-26:
1. What did I do:
I am trying to play mp3 file from my sd card. Using my own ESP-WROOM-32 module. First, I modified the I2S pins to working with my hardware, it works on play_mp3_dac example. Next, I wired the SD card same as module esp32-lyrat and removed the element codec_chip in pipeline_sdcard_mp3 example's main file.
2. Problem:
The SD card is mounted success. But mp3 file can't be read.
It returns:
`
I (320) SDCARD_MP3_EXAMPLE: [ 1 ] Mount sdcard
I (420) SDCARD_MP3_EXAMPLE: [3.0] Create audio pipeline for playback
I (420) SDCARD_MP3_EXAMPLE: [3.1] Create fatfs stream to read data from sdcard
I (420) SDCARD_MP3_EXAMPLE: [3.2] Create i2s stream to write data to codec chip
I (430) SDCARD_MP3_EXAMPLE: [3.3] Create mp3 decoder to decode mp3 file
I (430) SDCARD_MP3_EXAMPLE: [3.4] Register all elements to audio pipeline
I (440) SDCARD_MP3_EXAMPLE: [3.5] Link it together [sdcard]-->fatfs_stream-->mp3_decoder-->i2s_stream-->[codec_chi
p]
I (450) SDCARD_MP3_EXAMPLE: [3.6] Setup uri (file as fatfs_stream, mp3 as mp3 decoder, and default output is i2s)
I (460) SDCARD_MP3_EXAMPLE: [ 4 ] Setup event listener
I (470) SDCARD_MP3_EXAMPLE: [4.1] Listening event from all elements of pipeline
I (480) SDCARD_MP3_EXAMPLE: [4.2] Listening event from peripherals
I (480) SDCARD_MP3_EXAMPLE: [ 5 ] Start audio_pipeline
E (490) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x109
E (500) diskio_sdmmc: sdmmc_read_blocks failed (265)
E (500) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
E (510) diskio_sdmmc: sdmmc_read_blocks failed (263)
E (520) FATFS_STREAM: Failed to open file /sdcard/test.mp3
E (520) AUDIO_ELEMENT: [file] AEL_STATUS_ERROR_OPEN
W (530) AUDIO_ELEMENT: IN-[mp3] AEL_IO_ABORT
E (530) AUDIO_ELEMENT: [file] AEL_MSG_CMD_ERROR
Encountered error reading when MP3 init
E (540) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN
E (550) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_ERROR
I (550) SDCARD_MP3_EXAMPLE: [ 6 ] Listen for all pipeline events
W (580) SDCARD_MP3_EXAMPLE: [ * ] Stop event received
I (580) SDCARD_MP3_EXAMPLE: [ 7 ] Stop audio_pipeline
W (580) AUDIO_PIPELINE: There are no listener registered
`

tuanpmt, 2018-06-26:
Hi @RemyEE 

Are you sure the sdcard card was working properly? If not, please try to test the sdcard with an example read/write: https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card

RemyEE, 2018-06-26:
I tested the sd_card example. It works well. 
[image]
Do I need to set some specific pins to make pipeline_sdcard_mp3 work on my own module instead of ESP32-Lyrat?

RemyEE, 2018-06-27:
I've solved the problem.
Follow the example https://github.com/espressif/esp-idf/tree/master/examples/storage/sd_card
to set GPIO2 low to enter download mode, I connected GPIO0 and GPIO2 by a jumper.
Some how, connecting GPIO0 and GPIO2 make the mp3 file can't be read. I tried to disconnect the jumper between GPIO0 and GPIO2 the file be read properly.
Summary: 
(1) To enter flash mode, connect GPIO0 and GPIO2 by a jumper. 
(2) In run mode, disconnect that jumper. The SD card should works.

closed 2018-06-27 by RemyEE

dev-techshlok, 2020-05-25:
I am using ESP32 Wroom board. What configuration do i need to use.
My steps till now.
I have checked the ESP32 with SDcard using ESP-IDF example.
Now i am trying to access the SDCard MP3 example "pipeline_sdcard_mp3.c".
Can you help me? @RemyEE