espressif/esp-adf #31: play_mp3

SantoshVema1, 2018-06-26:
Hello,

When running the play_mp3 on esp32-lyrat v4.3, I get the following error. I can't hear any audio even though I connected an apple headphones to the PhoneJack. 

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5564
b4
[0;32mI (28) boot: ESP-IDF v3.0.1 2nd stage bootloader
I (28) boot: compile time 13:48:38
I (30) boot: Enabling RNG early entropy source...
I (33) boot: SP Speed      : 40MHz
I (37) boot: SPI Mode       : DIO
I (41) boot: SPI Flash Size : 8MB
I (45) boot: Partition Table:
I (48) boot: ## Label    6 boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data         01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot: End of partition table
I (82) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x1f238 (127544) map
[0;32mI (136) esp_image: segment 1: paddr=0x0002f260 vaddr=0x3fb0000 size=0x00db0 (  3504) load
I (138) esp_image:segment 2: paddr=0x00030018 vaddr=0x400d0018 size=0x212dc (13590) map
[0;32mI (190) esp_image: segment 3: paddr=0x000512fc vaddr=0x3fb0db0 size=0x01464 (  5220) load
I (193) esp_image:segment 4: paddr=0x00052768 vaddr=0x40080000 size=0x00400 (  104) load
I (197) esp_image: segment 5: paddr=0x0005270 vaddr=0x40080400 size=0x09de4 ( 40420) load
[0;32mI (223) esp_image: segment 6: paddr=0x0005c95c vaddr=0x400c0000 size=0x00000 (     0) load
I (229) boot: Loadd app from partition at offset 0x10000
I (229) boot: Disabling RNG early entropy source...
I (232) cpu_start: Pro cpu up.
I (236) cpu_start: Starting app cpu, entry point is 0x40080edc
I (0) cpu_start: App cpu up.
I (246) heap_init: Initializing. RAM availablE6E0 len 00001920 (6 KiB): DRAM
I (259) heap_init: At 3FFB2A40 len 0002D5C0 (181 KiB): DRAM
I (265) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (272) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
M
I (284) cpu_start: Pro cpu start user code
I (302) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
 (304) PLAY_MP3_FLASH: [ 1 ] Start audio codec chip
E (304) gpio: gpio_isr_handler_add(377): GPIO isr service is not installed, call gpio_install_isr_service() first
[0;32mI (334) PLAY_MP3_FLASH: [ 2 ] Create audio pipeline, add all elements to pipeline, and subscribe pipeline ev                 ent
I (334) PLAY_MP3_FLASH: [2.1] Create mp3 decoder to decode mp3 file and set custom read callback
udio pipelineMP3_FLASH: [2.2] Create i2s stream to write data to codec chip
I (364) PLAY_MP3_FLASH: [2.4] Link it together [mp3_music_read_cb]-->mp3_decoder-->i2s_stream-->[codec_chip]
I (374) PLAY_MP3_FLASH: [ 3 ] Setup event lisener
I (374) PLAY_MP3_FLASH: [3.1] Listening event : [ 4 ] Start audio_pipeline
I (404) PLAY_MP3_FLASH: [ * ] Receive music info from mp3 decoder, sample_rates=44100, bits=16, ch=2
I (7184) PLAY_MP3_FLASH: [ 5 ] Stop audio_pipeline
W (7184) AUDIO_PIPELINE: There are no listener registered

Thanks

Santosh

krzychb, 2018-06-27:
Hi @SantoshVema1,

Thank you for reporting this error. 

I see the same error on ESP32-LyraT v4.3 but can hear the audio using speakers or apple phones through the phone jack without any issues. Have you tried using speakers or another headphones?

The fix for the error itself is in internal review and I hope to get it merged soon.

SantoshVema1, 2018-06-27:
Hello @krzychb,

Thanks, I was just about to reply saying that I can hear the audio through the headphones but get an error. 

Thanks,
Santosh

krzychb, 2018-06-28:
`E (304) gpio: gpio_isr_handler_add(377): GPIO isr service is not installed, call gpio_install_isr_service() first` error has been resolved in b83c7535f73b441f8643ce0033636897fb464d34

SantoshVema1, 2018-06-28:
Excellent thanks.

closed 2019-01-03 by jason-mao