espressif/esp-adf #34: REC.WAV get from sample code pipeline_wav_sdcard, cannot be played

Makeblock-official, 2018-06-29:
The front character of the binary is:
0000 0000 0000 0000 0000 0000 0000 0000
0000 0000 0000 0000 0000 0000 0000 0000
0000 0000 0000 0000 0000 0000 b6ff 1500
baff 2a00 a4ff 1900 a4ff 1600 b2ff 3100
b0ff 3200 c1ff 2a00 c9ff 2300 c6ff 0700
cdff fbff b2ff 0100 adff f5ff d1ff e3ff
d0ff ddff d4ff e5ff ddff eeff c6ff eaff
d2ff cdff f0ff cdff e9ff c8ff e8ff b4ff
daff bbff c7ff abff daff a7ff ddff afff
e6ff b6ff f3ff b9ff dfff b4ff d2ff b0ff
efff caff f7ff dcff 0100 ccff f2ff c2ff
ccff beff f6ff dcff 0f00 f7ff efff eaff
0800 efff 3400 0900 2400 1100 1e00 1000
1f00 2e00 3000 3300 4700 3b00 2900 3d00
1700 3100 2800 4400 2700 4900 2200 2800
2700 4800 2c00 5c00 3b00 3f00 3900 4700
2400 4100 5300 2100 5400 1b00 4400 2d00
6300 2000 6000 2400 4e00 2700 6400 3000
5100 4700 4800 3700 4a00 4400 4c00 4c00
"
Obviously, it does not have the file header information of the standard wav file.

jason-mao, 2018-07-09:
@Makeblock-official 

Thanks. We will reproduced and fixed.

ahhfzhang, 2018-07-11:
@Makeblock-official This issue has been resolved in 3da1c35 .

closed 2018-07-15 by jason-mao