espressif/esp-adf #37: adf的配置跟之前有什么不同吗？

DanielLiMing, 2018-07-07:
git下最近的adf编译运行里面的demo有几个pipeline_sdcard_mp3，pipeline_sdcard_wav等几个demo没有播放出音乐，没有任何错误信息，就是没有声音，另外的一些demo例如element_sdcard_mp3，pipeline_bt_sink都能够通过耳机听到音乐，是配置跟之前有什么不同吗？adf是git的最新的，idf默认的是c2b39f4a5f4没有声音，乐鑫官网上的举例是ca3faa61 也没有声音，是需要其他配置吗？使用之前git的都能够听到音乐的。
[image]

jason-mao, 2018-07-08:
@DanielLiMing 

We are already got this bug, it's fixing. This bug caused by `fa554130bc6920f0e1a3616fa8ce3a39c28a5701` modify the `fatfs_stream` initialization parameters `fatfs_stream_cfg_t `. If you want to test now, please change `play_sdcard_mp3_example.c` as follow.
https://github.com/espressif/esp-adf/blob/717a5b1cac28fd0a00261118672222450c428b56/examples/player/pipeline_sdcard_mp3/main/play_sdcard_mp3_example.c#L68

```c
fatfs_stream_cfg_t fatfs_cfg = FATFS_STREAM_CFG_DEFAULT();
fatfs_cfg.type = AUDIO_STREAM_READER;

```

Thanks.

krzychb, 2018-07-10:
Hi @DanielLiMing this issue has been resolved in 153dc76c771cfd4bcc33538b3a0cc7820ceb8f86

bangdushutong, 2018-07-11:
除了dueros登不上，其它几个例子都没有问题

bangdushutong, 2018-07-11:
profile换了，分区的问题也解决了，wifi连上了，就是dueros登不上。

jason-mao, 2018-07-11:
@bangdushutong We will be reply at #39 . This issue will be closed after @DanielLiMing review.

wenkumk, 2018-07-11:
@bangdushutong 您是怎么解决分区的问题呢，profile是通过后给的那个链接里的内容吗？

dtboy1995, 2018-08-13:
@bangdushutong 请问`duer_profile`这个文件中需要配置什么呢，谢谢您

closed 2018-11-09 by jason-mao