espressif/esp-adf #39: dueros

bangdushutong, 2018-07-11:


jason-mao, 2018-07-11:
@bangdushutong Could you  provide the issue log?

bangdushutong, 2018-07-11:
串口信息 ：
[0;31mE (310611) DUEROS: Func:retry_login_timer_cb[0m
[0;31mE (310618) DUEROS: Recv Que DUER_CMD_LOGIN[0m
[0;31mE (310618) DUEROS: @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@@@@@@@@@@@@@@@@@@@@@@@@@[0m
[0;32mI (310620) DUEROS: duer_start, len:1
[0m
[0;31mE (310625) DUEROS: $$$$$$$$$$$$$$$$$$$$$$$$$$$$$
$$$$$$$$$$$$$$$$$$$$$$$$$$$$$[0m
I (307453,tid:3ffc2f80) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f806f3c, length:1, profile:0x3f80f420
[0;31mE (310633) DUEROS: +++++++++++++++++++++++++++++++++++++++++
+++++++++++++++++++++++++++++++++++++
[0m
E (307471,tid:3ffc2f80) lightduer_ca.c( 607):     create the configuation failed...
I (307488,tid:3ffc2f80) lightduer_ds_log_cache.c(  45): cached tracecode: 0x12010301
I (307496,tid:3ffc2f80) lightduer_connagent.c( 131): =====> event: 1[DUER_EVT_START], status: -18
E (307504,tid:3ffc2f80) lightduer_connagent.c( 178): start fail! status:-18
E (307511,tid:3ffc2f80) lightduer_connagent.c( 255): Action failed: event: 1, status: -18
W (307519,tid:3ffc2f80) lightduer_engine.c( 623): duer_engine_stop has stopped
I (307526,tid:3ffc2f80) lightduer_connagent.c( 131): =====> event: 5[DUER_EVT_STOP], status: -18
I (307535,tid:3ffc2f80) lightduer_connagent.c( 228): connect stopped, status:-18!!
[0;31mE (310714) DUEROS: event: 1[0m
[0;31mE (310717) DUEROS: event: DUER_EVENT_STOPPED[0m
------------------------
-----------------------------
[0;31mE (310722) DUEROS: Dueros reconnect failed,time num:128 [0m
W (307555,tid:3ffc2f80) lightduer_events.c(  80): [lightduer_ca] <== event end = 0x40130444, timespent = 102

其中
 @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
@@@@@@@@@@@@@@@@@@@@@@@@@、
$$$$$$$$$$$$$$$$$$$$$$$$$$$$$
$$$$$$$$$$$$$$$$$$$$$$$$$$$$$和+++++++++++++++++++++++++++++++++++++++++
+++++++++++++++++++++++++++++++++++++
是我加上去的，用ESP_LOGE（）显示的定位代码行位置的信息。

wenkumk, 2018-07-11:
我的编译后出现这样的问题：
[0;32mI (78) boot: End of partition table[0m
[0;32mI (82) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x43f28 (278312) map[0m
[0;32mI (172) esp_image: segment 1: paddr=0x00053f50 vaddr=0x3ffb0000 size=0x03844 ( 14404) load[0m
[0;32mI (177) esp_image: segment 2: paddr=0x0005779c vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (179) esp_image: segment 3: paddr=0x00057ba4 vaddr=0x40080400 size=0x0846c ( 33900) load[0m
[0;32mI (199) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xfa5a4 (1025444) map[0m
[0;32mI (497) esp_image: segment 5: paddr=0x0015a5c4 vaddr=0x4008886c size=0x09070 ( 36976) load[0m
[0;32mI (510) esp_image: segment 6: paddr=0x0016363c vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;31mE (511) esp_image: Image length 1390176 doesn't fit in partition length 1048576[0m
[0;31mE (516) boot: Factory app partition is not bootable[0m
[0;31mE (522) boot: No bootable app partitions in the partition table[0m
user code done

bangdushutong, 2018-07-11:
No bootable app partitions in the partition table�[0m
 user code done 用户无法引导。
这个是分区问题。须在menuconfig中用自己定义的分区。

bangdushutong, 2018-07-11:
我自己建的分区是这样的：
[image]

bangdushutong, 2018-07-13:
现在OK了，但语音识别正确率太低了，不如板子自带的固件识别率高，不知啥原因。

bangdushutong, 2018-07-17:
[0;32mI (53778) AAC_DECODER: New audio song[0m
[0;32mI (53779) AAC_DECODER: This audio is TS AAC[0m
[0;32mI (53895) AAC_DECODER: I2S setup: sample_rate = 44100 : channels = 2[0m
[0;32mI (53895) AAC_DECODER: M4A sampleRate=44100, Channels=2, frameSize=2048[0m
[0;32mI (53896) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5[0m
[0;32mI (53905) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY, line:525[0m
[0;32mI (53911) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0[0m
[0;32mI (53917) AUDIO_ELEMENT: [resample] Element task created[0m
[0;32mI (53927) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:260, MEM Total:4011616 Bytes, Inter:157108 Bytes, Dram:97972 Bytes

[0m
[0;32mI (53935) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (53936) AUDIO_PIPELINE: Pipeline started[0m
[0;32mI (53942) RSP_FILTER: rsp_filter_open, decoder, src:44100,2,dest:48000,2,in:2048, out_pcm:576[0m
[0;32mI (53947) AUDIO_WRAPPER: Get play position 932 of 0[0m
[0;31mE (53949) AUDIO_WRAPPER: duer_dcs_speech_on_finished,2[0m
[0;32mI (53978) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
[0;33mW (58890) HTTP_STREAM: No more data,errno:0[0m
[0;32mI (58890) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0[0m
[0;32mI (58890) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (58986) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
Guru Meditation Error: Core  0 panic'ed (StoreProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4008a9b2  PS      : 0x00050033  A0      : 0x4008a8bb  A1      : 0x3ffe81f0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x4008a83c  
A6      : 0x00000001  A7      : 0x00060423  A8      : 0x800834d2  A9      : 0x3ffb05e0  
A10     : 0x3ffb0c14  A11     : 0x00000001  A12     : 0x3ffe4b44  A13     : 0x00000001  
A14     : 0x000000fe  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x40090a0c  LEND    : 0x40090a3a  LCOUNT  : 0xffffffff  

Backtrace: 0x4008a9b2:0x3ffe81f0 0x4008a8b8:0x3ffe8200

Entering gdb stub now.
$T0b#e6

bangdushutong, 2018-07-17:
[0;33mW (402599) HTTP_STREAM: No more data,errno:0[0m
[0;32mI (402599) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0[0m
[0;32mI (402599) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (402701) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
assertion "res == coreID || res == portMUX_FREE_VAL" failed: file "D:/esp/msys32/home/adf/esp-adf/esp-idf/components/freertos/portmux_impl.inc.h", line 105, function: vPortCPUAcquireMutexIntsDisabledInternal
abort() was called at PC 0x40114abf on core 0

Backtrace: 0x4008c0cb:0x3ffbfb90 0x4008c0fc:0x3ffbfbb0 0x40114abf:0x3ffbfbd0 0x400888d5:0x3ffbfc00 0x4008a1c7:0x3ffbfc30 0x40105f61:0x3ffbfc70 0x4010b80c:0x3ffbfca0 0x40110da5:0x3ffbfcc0 0x40100922:0x3ffbfd10 0x400fdfe9:0x3ffbfd40 0x4010c349:0x3ffbfd60

Entering gdb stub now.
$T0b#e6

bangdushutong, 2018-07-17:
[0;32mI (39093) AAC_DECODER: New audio song[0m
[0;32mI (39094) AAC_DECODER: This audio is TS AAC[0m
[0;32mI (39218) AAC_DECODER: I2S setup: sample_rate = 44100 : channels = 2[0m
[0;32mI (39219) AAC_DECODER: M4A sampleRate=44100, Channels=2, frameSize=2048[0m
[0;32mI (39219) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5[0m
[0;32mI (39232) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY, line:525[0m
[0;32mI (39235) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0[0m
[0;32mI (39241) AUDIO_ELEMENT: [resample] Element task created[0m
[0;32mI (39247) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:260, MEM Total:4029300 Bytes, Inter:157108 Bytes, Dram:97972 Bytes

[0m
[0;32mI (39266) AUDIO_PIPELINE: Pipeline started[0m
[0;32mI (39267) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (39268) AUDIO_WRAPPER: Get play position 1308 of 0[0m
[0;32mI (39271) RSP_FILTER: rsp_filter_open, decoder, src:44100,2,dest:48000,2,in:2048, out_pcm:576[0m
[0;31mE (39270) AUDIO_WRAPPER: duer_dcs_speech_on_finished,2[0m
[0;32mI (39306) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
[0;33mW (44223) HTTP_STREAM: No more data,errno:0[0m
[0;32mI (44224) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0[0m
[0;32mI (44224) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (44502) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
[0;32mI (47331) DUEROS: AUDIO_USER_KEY_VOL_UP [55][0m
[0;32mI (47652) DUEROS: AUDIO_USER_KEY_VOL_UP [64][0m
[0;32mI (48649) DUEROS: AUDIO_USER_KEY_VOL_UP [73][0m
[0;32mI (51030) DUEROS: AUDIO_USER_KEY_VOL_UP [82][0m
[0;33mW (51284) HTTP_STREAM: No more data,errno:0[0m
[0;32mI (51285) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0[0m
[0;32mI (51285) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (51383) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
[0;33mW (58291) HTTP_STREAM: No more data,errno:0[0m
[0;32mI (58292) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0[0m
[0;32mI (58292) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (58385) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
[0;33mW (65362) HTTP_STREAM: No more data,errno:0[0m
[0;32mI (65365) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0[0m
[0;32mI (65366) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (65498) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
[0;33mW (72410) HTTP_STREAM: No more data,errno:0[0m
[0;32mI (72411) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0[0m
[0;32mI (72411) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (74096) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception)
Debug exception reason: BREAK instr 
Register dump:
PC      : 0x40084d90  PS      : 0x00000016  A0      : 0x40080306  A1      : 0x3ffe7840  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x4008a90b  A5      : 0x3ffe11c0  
A6      : 0x3ffb1c04  A7      : 0x3ffb682c  A8      : 0x00000000  A9      : 0x00000000  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x80089458  A13     : 0x3ffe11b0  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000015  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x40090a0c  LEND    : 0x40090a3a  LCOUNT  : 0x00000000  

Backtrace: 0x40084d90:0x3ffe7840 0x40080303:0x003ffd6e

Entering gdb stub now.
$T1f#eb

Chenzejun-Dummkopf, 2018-10-09:
我想问下 duer 登陆失败是怎么回事？我现在遇到这个问题并且无法解决
我用的esp32lyrat 自带的duer_profile，但是始终无法登陆。
我现在将我的log贴出来，大家看下：
rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5900
load:0x40078000,len:7796
load:0x40080000,len:7440
entry 0x400803a0
\0x1b[0;32mI (29) boot: ESP-IDF 2nd stage bootloader\0x1b[0m
\0x1b[0;32mI (29) boot: compile time 09:55:36\0x1b[0m
\0x1b[0;32mI (31) boot: Enabling RNG early entropy source...\0x1b[0m
\0x1b[0;32mI (33) boot: SPI Speed : 80MHz\0x1b[0m
\0x1b[0;32mI (37) boot: SPI Mode : DIO\0x1b[0m
\0x1b[0;32mI (41) boot: SPI Flash Size : 4MB\0x1b[0m
\0x1b[0;32mI (45) boot: Partition Table:\0x1b[0m
\0x1b[0;32mI (49) boot: ## Label Usage Type ST Offset Length\0x1b[0m
\0x1b[0;32mI (56) boot: 0 nvs WiFi data 01 02 00009000 00006000\0x1b[0m
\0x1b[0;32mI (64) boot: 1 phy_init RF data 01 01 0000f000 00001000\0x1b[0m
\0x1b[0;32mI (71) boot: 2 avs WiFi data 01 02 00010000 00006000\0x1b[0m
\0x1b[0;32mI (79) boot: 3 factory factory app 00 00 00020000 00300000\0x1b[0m
\0x1b[0;32mI (86) boot: End of partition table\0x1b[0m
\0x1b[0;32mI (90) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x11b318 (1159960) map\0x1b[0m
\0x1b[0;32mI (437) esp_image: segment 1: paddr=0x0013b340 vaddr=0x3ffb0000 size=0x041a8 ( 16808) load\0x1b[0m
\0x1b[0;32mI (442) esp_image: segment 2: paddr=0x0013f4f0 vaddr=0x3ffb41a8 size=0x00000 ( 0) load\0x1b[0m
\0x1b[0;32mI (443) esp_image: segment 3: paddr=0x0013f4f8 vaddr=0x40080000 size=0x00400 ( 1024) load\0x1b[0m
\0x1b[0;32mI (453) esp_image: segment 4: paddr=0x0013f900 vaddr=0x40080400 size=0x00710 ( 1808) load\0x1b[0m
\0x1b[0;32mI (462) esp_image: segment 5: paddr=0x00140018 vaddr=0x400d0018 size=0xbdc50 (777296) map\0x1b[0m
\0x1b[0;32mI (696) esp_image: segment 6: paddr=0x001fdc70 vaddr=0x40080b10 size=0x18000 ( 98304) load\0x1b[0m
\0x1b[0;32mI (731) esp_image: segment 7: paddr=0x00215c78 vaddr=0x400c0000 size=0x00000 ( 0) load\0x1b[0m
\0x1b[0;32mI (732) esp_image: segment 8: paddr=0x00215c80 vaddr=0x50000000 size=0x00000 ( 0) load\0x1b[0m
\0x1b[0;32mI (753) boot: Loaded app from partition at offset 0x20000\0x1b[0m
\0x1b[0;32mI (753) boot: Disabling RNG early entropy source...\0x1b[0m
\0x1b[0;32mI (754) spiram: SPI RAM mode: flash 80m sram 80m\0x1b[0m
\0x1b[0;32mI (758) spiram: PSRAM initialized, cache is in low/high (2-core) mode.\0x1b[0m
\0x1b[0;32mI (765) cpu_start: Pro cpu up.\0x1b[0m
\0x1b[0;32mI (769) cpu_start: Starting app cpu, entry point is 0x40081484\0x1b[0m
\0x1b[0;32mI (0) cpu_start: App cpu up.\0x1b[0m
\0x1b[0;32mI (1269) spiram: SPI SRAM memory test OK\0x1b[0m
\0x1b[0;32mI (1270) heap_init: Initializing. RAM available for dynamic allocation:\0x1b[0m
\0x1b[0;32mI (1270) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM\0x1b[0m
\0x1b[0;32mI (1276) heap_init: At 3FFBB178 len 00024E88 (147 KiB): DRAM\0x1b[0m
\0x1b[0;32mI (1282) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM\0x1b[0m
\0x1b[0;32mI (1289) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM\0x1b[0m
\0x1b[0;32mI (1295) heap_init: At 40098B10 len 000074F0 (29 KiB): IRAM\0x1b[0m
\0x1b[0;32mI (1302) cpu_start: Pro cpu start user code\0x1b[0m
\0x1b[0;32mI (1306) spiram: Adding pool of 4096K of external SPI memory to heap allocator\0x1b[0m
\0x1b[0;32mI (1314) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations\0x1b[0m
\0x1b[0;32mI (217) cpu_start: Starting scheduler on PRO CPU.\0x1b[0m
\0x1b[0;32mI (0) cpu_start: Starting scheduler on APP CPU.\0x1b[0m
\0x1b[0;32mI (218) duer: ==== VERSION: v1 ====\0x1b[0m

******* nvs_flash_init *******
\0x1b[0;32mI (258) gpio: GPIO[33]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 \0x1b[0m
.........................Initializing audio codec..............................
\0x1b[0;31mE (2268) CODEC_ALC5650: alc5650_read_reg error\0x1b[0m
\0x1b[0;31mE (2268) AudioCodec: Error: Failed to initialize codec.
\0x1b[0m
\0x1b[0;31mE (2268) AudioCodec: Could not read the Device ID.
\0x1b[0m

******* alc5650_init *******
\0x1b[0;31mE (6268) UI_LED: Led driver init failed\0x1b[0m

******* ui_led_init *******
\0x1b[0;32mI (6268) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:3 \0x1b[0m
\0x1b[0;32mI (6268) gpio: GPIO[37]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:3 \0x1b[0m
[... 88 lines cut ...]
{"configures":"{}","bindToken":"08c94f9b5b46c1657b11648bd9f8005f","coapPort":9001,"token":"5YxN55cQ","serverAddr":"device.iot.baidu.com","lwm2mPort":9001,"uuid":"037f0000000001","rsaCaCrt":"-----BEGIN CERTIFICATE-----\nMIIDUDCCAjgCCQCmVPUErMYmCjANBgkqhkiG9w0BAQUFADBqMQswCQYDVQQGEwJD\nTjETMBEGA1UECAwKU29tZS1TdGF0ZTEOMAwGA1UECgwFYmFpZHUxGDAWBgNVBAMM\nDyouaW90LmJhaWR1LmNvbTEcMBoGCSqGSIb3DQEJARYNaW90QGJhaWR1LmNvbTAe\nFw0xNjAzMTEwMzMwNDlaFw0yNjAzMDkwMzMwNDlaMGoxCzAJBgNVBAYTAkNOMRMw\nEQYDVQQIDApTb21lLVN0YXRlMQ4wDAYDVQQKDAViYWlkdTEYMBYGA1UEAwwPKi5p\nb3QuYmFpZHUuY29tMRwwGgYJKoZIhvcNAQkBFg1pb3RAYmFpZHUuY29tMIIBIjAN\nBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAtbhIeiN7pznzuMwsLKQj2xB02+51\nOvCJ5d116ZFLjecp9qtllqOfN7bm+AJa5N2aAHJtsetcTHMitY4dtGmOpw4dlGqx\nluoz50kWJWQjVR+z6DLPnGE4uELOS8vbKHUoYPPQTT80eNVnl9S9h/l7DcjEAJYC\nIYJbf6+K9x+Ti9VRChvWcvgZQHMRym9j1g/7CKGMCIwkC+6ihkGD/XG40r7KRCyH\nbD53KnBjBO9FH4IL3rGlZWKWzMw3zC6RTS2ekfEsgAtYDvROKd4rNs+uDU9xaBLO\ndXTl5uxgudH2VnVzWtj09OUbBtXcQFD2IhmOl20BrckYul+HEIMR0oDibwIDAQAB\nMA0GCSqGSIb3DQEBB [... cut ...]
I (2391,tid:3ffd6804) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f804e24, length:0, profile:0x0
I (2391,tid:3ffd6804) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (4391,tid:3ffd6804) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f804e24, length:0, profile:0x0
I (4391,tid:3ffd6804) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)

然后就一直卡在这里，直到超时重新登陆，我想问下这是为什么？

@bangdushutong @jason-mao @wenkumk @projectgus @kedars 
我现在也没有想到解决办法，希望知道答案的同志告知一声，万分感谢！！！

bangdushutong, 2018-10-11:
可能是你的网络问题。。。

bangdushutong, 2018-10-11:
自带的profile不行

Chenzejun-Dummkopf, 2018-10-11:
解决了这个问题了，谢谢大家

xieshang, 2019-04-02:
> 解决了这个问题了，谢谢大家

@Chenzejun-Dummkopf 请问你的是如何解决的？
看了你前面发出的log，看到有打印duer_profile信息duer_all_login，但是我的log里没看到过这个信息；

下面是我的log信息，烦请帮忙诊断一下是什么问题，谢谢~

> I (13642) DUEROS: PERIPH_WIFI_DISCONNECTED [281]
W (13644) PERIPH_WIFI: Wi-Fi disconnected from SSID XSC_mob, auto-reconnect enabled, reconnect after 1000 ms
I (17073) DUEROS: PERIPH_WIFI_DISCONNECTED [281]
W (17074) PERIPH_WIFI: Wi-Fi disconnected from SSID XSC_mob, auto-reconnect enabled, reconnect after 1000 ms
I (18563) wifi: n:4 0, o:1 0, ap:255 255, sta:4 0, prof:1
I (18564) wifi: state: init -> auth (b0)
I (18571) wifi: state: auth -> assoc (0)
I (18584) wifi: state: assoc -> run (10)
I (18609) wifi: connected with XSC_mob, channel 4
I (18609) wifi: pm start, type: 1<br>
I (19166) event: sta ip: 192.168.43.131, mask: 255.255.255.0, gw: 192.168.43.1
I (19166) DUEROS: PERIPH_WIFI_CONNECTED [276]
E (19168) DUEROS: Recv Que DUER_CMD_LOGIN
I (19172) DUEROS: duer_start, len:1

I (19054,tid:3ffbe720) lightduer_engine.c( 220):duer_engine_start, g_handler:3F8077D0, length:1, profile:3F807864
E (19064,tid:3ffbe720) lightduer_ca.c( 645):    create the configuation failed...
W (19180) DUEROS: duer_callback: type:0, source:0x3f806bd8 data:2, data_len:0
I (19072,tid:3ffbe720) lightduer_ds_log_cache.c(  45):cached tracecode: 0x12010301
I (19116,tid:3ffbe720) lightduer_connagent.c( 160):=====> event: 1[DUER_EVT_START], status: -18
E (19123,tid:3ffbe720) lightduer_connagent.c( 207):start fail! status:-18
E (19130,tid:3ffbe720) lightduer_connagent.c( 286):Action failed: event: 1, status: -18
W (19138,tid:3ffbe720) lightduer_engine.c( 812):duer_engine_stop has stopped
I (19145,tid:3ffbe720) lightduer_connagent.c( 160):=====> event: 5[DUER_EVT_STOP], status: -18
I (19153,tid:3ffbe720) lightduer_connagent.c( 257):connect stopped, status:-18!!
E (19283) DUEROS: event: 1
E (19286) DUEROS: event: DUER_EVENT_STOPPED
W (19291) DUEROS: duer_callback: type:0, source:0x3f806bd8 data:1, data_len:0
W (19169,tid:3ffbe720) lightduer_events.c(  80):[lightduer_ca] <== event end = 4018934C, timespent = 115
I (19335) DUEROS: Dueros DUER_CMD_QUIT reconnect, retry_num:2
E (21343) DUEROS: Func:retry_login_timer_cb
E (21343) DUEROS: Recv Que DUER_CMD_LOGIN
I (21343) DUEROS: duer_start, len:1<br>
W (21346) DUEROS: duer_callback: type:0, source:0x3f806bd8 data:2, data_len:0
I (21224,tid:3ffbe720) lightduer_engine.c( 220):duer_engine_start, g_handler:3F8077D0, length:1, profile:3F807864
E (21241,tid:3ffbe720) lightduer_ca.c( 645):    create the configuation failed...
I (21249,tid:3ffbe720) lightduer_ds_log_cache.c(  45):cached tracecode: 0x12010301
I (21255,tid:3ffbe720) lightduer_connagent.c( 160):=====> event: 1[DUER_EVT_START], status: -18
E (21264,tid:3ffbe720) lightduer_connagent.c( 207):start fail! status:-18
E (21271,tid:3ffbe720) lightduer_connagent.c( 286):Action failed: event: 1, status: -18
W (21279,tid:3ffbe720) lightduer_engine.c( 812):duer_engine_stop has stopped
I (21285,tid:3ffbe720) lightduer_connagent.c( 160):=====> event: 5[DUER_EVT_STOP], status: -18
I (21293,tid:3ffbe720) lightduer_connagent.c( 257):connect stopped, status:-18!!
E (21423) DUEROS: event: 1
E (21426) DUEROS: event: DUER_EVENT_STOPPED
W (21431) DUEROS: duer_callback: type:0, source:0x3f806bd8 data:1, data_len:0
W (21308,tid:3ffbe720) lightduer_events.c(  80):[lightduer_ca] <== event end = 4018934C, timespent = 85
I (21441) DUEROS: Dueros DUER_CMD_QUIT reconnect, retry_num:4
E (25482) DUEROS: Func:retry_login_timer_cb
E (25483) DUEROS: Recv Que DUER_CMD_LOGIN
I (25483) DUEROS: duer_start, len:1<br>
I (25363,tid:3ffbe720) lightduer_engine.c( 220):duer_engine_start, g_handler:3F8077D0, length:1, profile:3F807864
W (25486) DUEROS: duer_callback: type:0, source:0x3f806bd8 data:2, data_len:0
E (25373,tid:3ffbe720) lightduer_ca.c( 645):    create the configuation failed...
I (25418,tid:3ffbe720) lightduer_ds_log_cache.c(  45):cached tracecode: 0x12010301
I (25423,tid:3ffbe720) lightduer_connagent.c( 160):=====> event: 1[DUER_EVT_START], status: -18
E (25431,tid:3ffbe720) lightduer_connagent.c( 207):start fail! status:-18
E (25438,tid:3ffbe720) lightduer_connagent.c( 286):Action failed: event: 1, status: -18
W (25446,tid:3ffbe720) lightduer_engine.c( 812):duer_engine_stop has stopped
I (25452,tid:3ffbe720) lightduer_connagent.c( 160):=====> event: 5[DUER_EVT_STOP], status: -18
I (25461,tid:3ffbe720) lightduer_connagent.c( 257):connect stopped, status:-18!!
E (25591) DUEROS: event: 1
E (25595) DUEROS: event: DUER_EVENT_STOPPED
W (25476,tid:3ffbe720) lightduer_events.c(  80):[lightduer_ca] <== event end = 4018934C, timespent = 114
W (25629) DUEROS: duer_callback: type:0, source:0x3f806bd8 data:1, data_len:0
I (25629) DUEROS: Dueros DUER_CMD_QUIT reconnect, retry_num:8

tanhailong, 2019-04-04:
> 解决了这个问题了，谢谢大家

你这个问题是怎么解决的呀，我使用Dueros一直没有用起来过

jason-mao, 2019-04-04:
@xieshang @tanhailong 参考一下#126. ` lightduer_connagent.c( 257):connect stopped, status:-18!!` 的错误代码是无效的profile。

closed 2019-04-04 by jason-mao