espressif/esp-adf #45: How simply connect ESP32 to Bluetooth Speaker and play sound file?

williamesp2015, 2018-07-16:
If we have Bluetooth Speaker then we do not need I2S/DAC to play sound. In this case how to connect ESP32 with a Bluetooth Speaker and send streaming Audio File, pause and resume it.
Thank you

bangdushutong, 2018-07-17:
[0;32mI (53778) AAC_DECODER: New audio song[0m
[0;32mI (53779) AAC_DECODER: This audio is TS AAC[0m
[0;32mI (53895) AAC_DECODER: I2S setup: sample_rate = 44100 : channels = 2[0m
[0;32mI (53895) AAC_DECODER: M4A sampleRate=44100, Channels=2, frameSize=2048[0m
[0;32mI (53896) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5[0m
[0;32mI (53905) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY, line:525[0m
[0;32mI (53911) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0[0m
[0;32mI (53917) AUDIO_ELEMENT: [resample] Element task created[0m
[0;32mI (53927) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:260, MEM Total:4011616 Bytes, Inter:157108 Bytes, Dram:97972 Bytes

[0m
[0;32mI (53935) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (53936) AUDIO_PIPELINE: Pipeline started[0m
[0;32mI (53942) RSP_FILTER: rsp_filter_open, decoder, src:44100,2,dest:48000,2,in:2048, out_pcm:576[0m
[0;32mI (53947) AUDIO_WRAPPER: Get play position 932 of 0[0m
[0;31mE (53949) AUDIO_WRAPPER: duer_dcs_speech_on_finished,2[0m
[0;32mI (53978) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
[0;33mW (58890) HTTP_STREAM: No more data,errno:0[0m
[0;32mI (58890) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0[0m
[0;32mI (58890) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (58986) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
Guru Meditation Error: Core  0 panic'ed (StoreProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x4008a9b2  PS      : 0x00050033  A0      : 0x4008a8bb  A1      : 0x3ffe81f0  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x4008a83c  
A6      : 0x00000001  A7      : 0x00060423  A8      : 0x800834d2  A9      : 0x3ffb05e0  
A10     : 0x3ffb0c14  A11     : 0x00000001  A12     : 0x3ffe4b44  A13     : 0x00000001  
A14     : 0x000000fe  A15     : 0x00000000  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x40090a0c  LEND    : 0x40090a3a  LCOUNT  : 0xffffffff  

Backtrace: 0x4008a9b2:0x3ffe81f0 0x4008a8b8:0x3ffe8200

Entering gdb stub now.
$T0b#e6

tuanpmt, 2018-07-19:
Hi @williamesp2015 

The current version only supports BT Sink, We will add BT Source feature for next release.

Thanks.,

zeitounian, 2018-09-09:
Good morning, we are a development company of São Paulo, we have a demand of a client that needs to transmit audio and also control some IO's of ESP32, you can help me, we are already developing an APP, but the hardware is more complex, the ESP32 must receive the audio stream and when it receives some command to trigger the IO does not interrupt the audio stream.

jason-mao, 2019-01-03:
@rob-bits BT-Source has been supported on release v1.0. Please refer to https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_bt_source

closed 2019-01-03 by jason-mao

rob-bits, 2019-01-03:
@jason-mao Hice to hear it is implemented but i was wondering the support to connect/stream audio simultaneously to two sink device. I have a proof-of-concept with BTStack from bluekitchen + ESP32.
But it would be nice if the esp-idf or esp-adf would have this functionality.

Please check my issue:
[[TW#23410] A2DP Source stream audio to two Sink device #2060](https://github.com/espressif/esp-idf/issues/2060)

vednildip, 2024-07-05:
For  Arduino  IDE  how  to  implement  for  Bluetooth  speaker  connection ?