espressif/esp-adf #47: Bluetooth HFP support

BobaTrek, 2018-07-24:
Hello,

The readme page says that LyraT will be supporting Bluetooth A2DP and HFP.

I see that the Bluetooth A2DP sink support has been added - fantastic!

When will the Bluetooth HFP support become available?  HFP is required by my product.

Thanks!

JB041, 2018-11-09:
Even I was working on the HFP feature and was able to connect ESP32 with my mobile with HFP enabled. And I could log the HFP connection status on the monitor. But during the call it shows HFP connected on the phone but can't quite hear anything on either end of the call.

JB041, 2018-11-09:
please let me know if you had any luck..

jason-mao, 2019-08-17:
There is exmaple support HFP and A2DP `examples/get-started/pipeline_a2dp_sink_and_hfp`

closed 2019-08-17 by jason-mao