espressif/esp-adf #48: pipeline_wav_sdcard录音

DanielLiMing, 2018-07-26:
最近在lyrat_v4.2上测试录音，发现编译这个例程运行的时候录音只有10秒（理论上应该是20秒的），而且通过其他播放器播放的时候播放速度比录的声音要快，这是什么原因啊？

ahhfzhang, 2018-07-26:
Hi @DanielLiMing  
I have not reproduced this on the latest adf, can you show more detail, like rec.wav mediainfo ?

DanielLiMing, 2018-07-26:
播放语速明显比正常语速要快，而且音质也不是很好！日志没有什么异常信息，我记得之前的是能正常播放的！（原音乐名称：白羊）

DanielLiMing@163.com
 
发件人： Yong Zhang
发送时间： 2018-07-26 15:43
收件人： espressif/esp-adf
抄送： Loophole; Mention
主题： Re: [espressif/esp-adf] pipeline_wav_sdcard录音 (#48)
Hi @DanielLiMing
I have not reproduced this on the latest adf, can you show more detail, like rec.wav mediainfo ?

ahhfzhang, 2018-07-27:
Hi @DanielLiMing
你的意思是你录音的时候在播放一个歌曲，然后录下来之后和原来的音频对比语速快，音质不好，是吗？
我也做了同样尝试，没有发现你说的问题，能否帮忙将录音文件传上来看下。

DanielLiMing, 2018-07-27:
[rec.zip](https://github.com/espressif/esp-adf/files/2235461/rec.zip)
我又在新的环境中重新配置ADF发现结果还是有点快，录下来的文件只有10s，

closed 2018-07-31 by DanielLiMing