espressif/esp-adf #52: dueros无法识别

daiyinger, 2018-08-07:
试了好多次 ，百度语音提示都是提示“抱歉，我不知道你说了什么”，请问调试的话应从什么方向入手呢？

youngda, 2018-08-09:
请问 
`[0;31mE (480) esp_image: Image length 1390112 doesn't fit in partition length 1048576[0m
[0;31mE (485) boot: Factory app partition is not bootable[0m
[0;31mE (491) boot: No bootable app partitions in the partition table[0m
user code done`

怎么解决的？修改了factory,  app,  factory, 0x10000,4M,还是不行

youngda, 2018-08-09:
额，看到首页的说明了，不过回应声音好小 
Note:
DuerOS profile is device unique ID.
This example size of binary image is 1.3MB, please use ADF_APTH/examples/dueros/partitions_singleapp.csv instead of IDF_PATH/components/partition_table/partitions_singleapp.csv.

dtboy1995, 2018-08-13:
请问`duer_profile`这个文件中需要配置什么呢，谢谢您

dtboy1995, 2018-08-14:
@daiyinger  谢谢您，随便一个都可以是吗？

dtboy1995, 2018-08-15:
@daiyinger 您好，按照您说的下载了profile压缩包，解压了其中的一个文件，把里面的内容复制到dueros例子中的dueros_profile里，之后make menuconfig 配置了WiFi，之后make flash 。。。但是程序跑起来了，绿灯和红灯都亮着，按下rec按钮，也有日志，放下也有日志，就是说话没有反馈
```log
0x40081420: call_start_cpu1 at E:/esp-adf/esp-idf/components/esp32/cpu_start.c:2
15

I (0) cpu_start: App cpu up.
I (971) spiram: SPI SRAM memory test OK
I (972) heap_init: Initializing. RAM available for dynamic allocation:
I (972) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (978) heap_init: At 3FFBB450 len 00024BB0 (146 KiB): DRAM
I (984) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (991) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (997) heap_init: At 400918DC len 0000E724 (57 KiB): IRAM
I (1003) cpu_start: Pro cpu start user code
I (1008) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1016) spiram: Reserving pool of 32K of internal memory for DMA/internal alloc
ations
I (142) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (228) DUEROS: ADF version is v1.0-beta1-9-g8ec973c-dirty
W (228) ESP_PERIPH: Peripherals have been initialized already
I (230) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:3
I (239) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldo
wn: 0| Intr:3
E (258) PERIPH_SDCARD: no sdcard detect
I (32448) DUEROS: PERIPH_NOTIFY_KEY_REC
I (34357) DUEROS: PERIPH_NOTIFY_KEY_REC_QUIT
I (52149) DUEROS: PERIPH_NOTIFY_KEY_REC
I (56392) DUEROS: PERIPH_NOTIFY_KEY_REC_QUIT
I (75058) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (75338) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (75518) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (75683) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (75883) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (76058) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (76238) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (76383) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (76538) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (77082) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (77258) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (77638) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (77998) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (79638) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (79958) DUEROS: AUDIO_USER_KEY_VOL_UP [10]
I (83648) DUEROS: PERIPH_NOTIFY_KEY_REC
I (91102) DUEROS: PERIPH_NOTIFY_KEY_REC_QUIT
I (216298) DUEROS: PERIPH_NOTIFY_KEY_REC
I (221179) DUEROS: PERIPH_NOTIFY_KEY_REC_QUIT
```

dtboy1995, 2018-08-15:
插上sd卡后，重新make flash的部分日志如下
```log
I (24434,tid:3ffbd248) lightduer_engine.c( 174): duer_engine_start, g_handler:0x
3f806aa0, length:1469, profile:0x3f810300
I (24446,tid:3ffbd248) lightduer_ca_conf.c(  38):     duer_conf_get_string: uuid
 = 1609000000000a
I (24454,tid:3ffbd248) lightduer_ca_conf.c(  38):     duer_conf_get_string: serv
erAddr = device.iot.baidu.com
I (24488,tid:3ffbd248) baidu_ca_socket_adp.c( 116): DNS lookup succeeded. IP=180
.97.33.165
I (24521,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
W (24521,tid:3ffbd248) lightduer_events.c(  80): [lightduer_ca] <== event end =
0x40130330, timespent = 87
0x40130330: duer_engine_start at ??:?

I (25223,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
W (25223,tid:3ffbd248) lightduer_events.c(  80): [lightduer_ca] <== event end =
0x40130480, timespent = 605
0x40130480: duer_engine_data_available at ??:?

I (25230,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25239,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25247,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25256,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25264,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25273,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25286,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25290,tid:3ffbd248) lightduer_engine.c( 174): duer_engine_start, g_handler:0x
3f806aa0, length:0, profile:0x0
I (25300,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25308,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25317,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25326,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25334,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25343,tid:3ffbd248) lightduer_connagent.c( 169): will start latter(DUER_ERR_T
RANS_WOULD_BLOCK)
I (25585,tid:3ffbd248) lightduer_connagent.c( 148): connect started!
I (25586,tid:3ffbd248) lightduer_ds_log_cache.c(  67): no cache report
E (25772) DUEROS: event: 0
I (25775) AUDIO_WRAPPER: duer_dcs_init
E (25779) DUEROS: event: DUER_EVENT_STARTED
I (25599,tid:3ffbd248) lightduer_connagent.c( 183): add resource successfully!!
I (25779) DUEROS: Dueros DUER_CMD_CONNECTED, duer_state:1
I (25608,tid:3ffbd248) lightduer_connagent.c( 183): add resource successfully!!
W (25620,tid:3ffbd248) lightduer_system_info.c( 304): Undefined memory type, 0
E (25627,tid:3ffbd248) lightduer_system_info.c( 387): Sys Info: Get disk info fa
iled
I (71974) DUEROS: PERIPH_NOTIFY_KEY_REC
I (71974) REC_ENG: Recorder trigger start
I (72003) DUEROS: --- rec_engine_cb --- REC_EVENT_WAKEUP_START
I (72003) DUEROS: Recv Que DUER_CMD_START
I (72006) AUDIO_WRAPPER: duer_dcs_get_speaker_state
I (72772) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_START
I (74003) REC_ENG: Wakeup time is out
I (74090) DUEROS: PERIPH_NOTIFY_KEY_REC_QUIT
I (74891) REC_ENG: Detect suspend, wakeup:0, flag:1
[... 48 lines cut ...]
I (75889) AUDIO_ELEMENT: [mp3] Element task created
I (75895) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (75895) AUDIO_ELEMENT: [iis] Element task created
I (75901) MP3_DECODER: MP3 opened
I (75906) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1
I (75917) I2S_STREAM: AUDIO_STREAM_WRITER
I (75895) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
E (75955) HTTP_STREAM: Invalid HTTP stream
E (75955) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN
E (75955) ESP_AUDIO_TASK: media_ctrl_task error:ESP_ERR_AUDIO_OPEN
I (75961) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5
I (75961) AUDIO_WRAPPER: esp_auido status:4,err:10100
I (75968) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY, line:485
W (75969) AUDIO_ELEMENT: IN-[mp3] AEL_IO_ABORT
I (75984) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:4
Encountered error reading when MP3 init
E (75969) AUDIO_ELEMENT: [http] AEL_MSG_CMD_ERROR
W (75807,tid:3ffbd248) lightduer_events.c(  80): [lightduer_ca] <== event end =
0x40130480, timespent = 245
0x40130480: duer_engine_data_available at ??:?

E (75995) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN
I (76015) MP3_DECODER: Closed
E (76019) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_ERROR
```

daiyinger, 2018-08-15:
@dtboy1995 只能干很简单的事 比如问个天气 问个百科， 录音键按一次就行了 按了立马靠近话筒大声说话

dtboy1995, 2018-08-15:
@daiyinger，按照您说的。。。超级大声的问了，今天天气怎么样，下面是日志
```log
I (559756) AUDIO_WRAPPER: duer_dcs_get_speaker_state
I (559760,tid:3ffbd248) lightduer_dcs_router.c( 449): Directive name: Speak
I (559946) AUDIO_WRAPPER: Playing speak: http://res.iot.baidu.com:80/api/v1/tts/
Bvan8bXvmdpEJ_JBTF_DGmRAeIOMrIeVseA_QqzuGkxNOg_JYBMMN9aFf5HTXgKKyzXHRHqo6UjCCiPB
1AcTtJaP5H4PfPz1Naot-muULXI.mp3
I (559958) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:301, MEM Total:4167172 Byt
es, Inter:172780 Bytes, Dram:113644 Bytes

        field_set: 0xf, port: 80
        field_data[0]: off: 0, len: 4, part: http
        field_data[1]: off: 7, len: 17, part: res.iot.baidu.com
        field_data[2]: off: 25, len: 2, part: 80
        field_data[3]: off: 27, len: 123, part: /api/v1/tts/Bvan8bXvmdpEJ_JBTF_D
GmRAeIOMrIeVseA_QqzuGkxNOg_JYBMMN9aFf5HTXgKKyzXHRHqo6UjCCiPB1AcTtJaP5H4PfPz1Naot
-muULXI.mp3
        field_data[4]: unset
        field_data[5]: unset
        field_data[6]: unset
I (560006) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5
I (560012) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:398, MEM Total:4167172 Byt
es, Inter:172780 Bytes, Dram:113644 Bytes

I (560012) ESP_AUDIO_TASK: It's a decoder.URI:http://res.iot.baidu.com:80/api/v1
/tts/Bvan8bXvmdpEJ_JBTF_DGmRAeIOMrIeVseA_QqzuGkxNOg_JYBMMN9aFf5HTXgKKyzXHRHqo6Uj
CCiPB1AcTtJaP5H4PfPz1Naot-muULXI.mp3

I (560042) ESP_AUDIO_TASK: 1.IN:[http],CODEC:[mp3],FILTER:[NULL],OUT:[iis],RESAM
PLE:[48000]
I (560052) ESP_AUDIO_TASK: 2.Handles,IN:0x3f80f84c,CODEC:0x3f8196ac,FILTER:0x3f8
1097c,OUT:0x3f80f950
I (560062) ESP_AUDIO_TASK: 2.1 Update all pipelines
I (560067) ESP_AUDIO_TASK: 3. Previous starting...
I (560075) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (560075) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1
I (560079) MP3_DECODER: MP3 opened
I (560085) I2S_STREAM: AUDIO_STREAM_WRITER
I (560075) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
E (560117) HTTP_STREAM: Invalid HTTP stream
E (560118) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN
E (560118) ESP_AUDIO_TASK: media_ctrl_task error:ESP_ERR_AUDIO_OPEN
I (560124) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5
I (560124) AUDIO_WRAPPER: esp_auido status:4,err:10100
E (560138) AUDIO_WRAPPER: duer_dcs_speech_on_finished,0
W (560133) AUDIO_ELEMENT: IN-[mp3] AEL_IO_ABORT
Encountered error reading when MP3 init
I (560131) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY, line:485
E (560151) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN
I (560158) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:4
I (560164) MP3_DECODER: Closed
E (560132) AUDIO_ELEMENT: [http] AEL_MSG_CMD_ERROR
E (560176) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_ERROR
W (559992,tid:3ffbd248) lightduer_events.c(  80): [lightduer_ca] <== event end =
 0x40130480, timespent = 234
0x40130480: duer_engine_data_available at ??:?
```

daiyinger, 2018-08-15:
@dtboy1995 我看服务器有返回语音 返回的语音内容是什么呢？日志可否从按下按键时截取

dtboy1995, 2018-08-15:
您稍等，我先发个启动日志
```
dtboy@DESKTOP-KDJKJGK MINGW32 ~/esp/dueros
$ make monitor
MONITOR
--- idf_monitor on COM14 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5744
ho 0 tail 12 room 4
load:0x40078000,len:0
load:0x40078000,len:14032
entry 0x40078ff0
I (30) boot: ESP-IDF v3.0.1-dirty 2nd stage bootloader
I (30) boot: compile time 10:25:11
I (32) boot: Enabling RNG early entropy source...
I (35) qio_mode: Enabling QIO for flash chip GD
I (40) boot: SPI Speed      : 80MHz
I (44) boot: SPI Mode       : QIO
I (48) boot: SPI Flash Size : 4MB
I (52) boot: Partition Table:
I (56) boot: ## Label            Usage          Type ST Offset   Length
I (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (71) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (78) boot:  2 factory          factory app      00 00 00010000 00200000
I (86) boot: End of partition table
I (90) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x43864 (276
580) map
I (172) esp_image: segment 1: paddr=0x0005388c vaddr=0x3ffb0000 size=0x03818 ( 1
4360) load
I (176) esp_image: segment 2: paddr=0x000570ac vaddr=0x40080000 size=0x00400 (
1024) load
0x40080000: _WindowOverflow4 at E:/esp-adf/esp-idf/components/freertos/xtensa_ve
ctors.S:1685

I (179) esp_image: segment 3: paddr=0x000574b4 vaddr=0x40080400 size=0x08b5c ( 3
5676) load
I (199) esp_image: segment 4: paddr=0x00060018 vaddr=0x400d0018 size=0xfa5d4 (10
25492) map
0x400d0018: _flash_cache_start at ??:?

I (470) esp_image: segment 5: paddr=0x0015a5f4 vaddr=0x40088f5c size=0x08980 ( 3
5200) load
0x40088f5c: xPortGetCoreID at E:/esp-adf/esp-idf/components/freertos/tasks.c:449
4 (discriminator 2)
 (inlined by) xTaskIncrementTick at E:/esp-adf/esp-idf/components/freertos/tasks
.c:2465 (discriminator 2)

I (481) esp_image: segment 6: paddr=0x00162f7c vaddr=0x400c0000 size=0x00000 (
   0) load
I (492) boot: Loaded app from partition at offset 0x10000
I (492) boot: Disabling RNG early entropy source...
I (493) spiram: SPI RAM mode: flash 80m sram 80m
I (498) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (505) cpu_start: Pro cpu up.
I (509) cpu_start: Starting app cpu, entry point is 0x40081420
0x40081420: call_start_cpu1 at E:/esp-adf/esp-idf/components/esp32/cpu_start.c:2
15

[... 160 lines cut ...]
------------------------------------------------------------------------------
I (4195) ESP_AUDIO_CTRL: Func:media_ctrl_create, Line:144, MEM Total:4236820 Byt
es, Inter:204320 Bytes, Dram:145184 Bytes

W (4208) I2S: I2S driver already installed
W (4212) I2S: I2S driver already installed
I (4217) MP3_DECODER: MP3 init
I (4223) AUDIO_WRAPPER: Func:setup_player, Line:167, MEM Total:4223560 Bytes, In
ter:193752 Bytes, Dram:134616 Bytes

I (4232) AUDIO_WRAPPER: esp_audio instance is:0x3f80f6a4
I (4241) AUDIO_WRAPPER: duer_dcs_get_speaker_state
E (4247) DUEROS: event: DUER_EVENT_STARTED
I (4247) DUEROS: Dueros DUER_CMD_CONNECTED, duer_state:1
W (4062,tid:3ffc28c8) lightduer_events.c(  80): [lightduer_ca] <== event end = 0
x40130480, timespent = 115
0x40130480: duer_engine_data_available at ??:?

I (4078,tid:3ffc28c8) lightduer_connagent.c( 183): add resource successfully!!
I (4086,tid:3ffc28c8) lightduer_connagent.c( 183): add resource successfully!!
W (4096,tid:3ffc28c8) lightduer_system_info.c( 304): Undefined memory type, 0
E (4099,tid:3ffc28c8) lightduer_system_info.c( 387): Sys Info: Get disk info fai
led
I (4890) wifi: pm start, type:0
```

dtboy1995, 2018-08-15:
您好，这是我按下rec放开说话的日志
```log
I (142735) DUEROS: PERIPH_NOTIFY_KEY_REC
I (142735) REC_ENG: Recorder trigger start
I (142739) DUEROS: --- rec_engine_cb --- REC_EVENT_WAKEUP_START
I (142740) DUEROS: Recv Que DUER_CMD_START
I (142748) AUDIO_WRAPPER: duer_dcs_get_speaker_state
I (142899) DUEROS: PERIPH_NOTIFY_KEY_REC_QUIT
I (143639) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_START
I (144739) REC_ENG: Wakeup time is out
I (145654,tid:3ffc28c8) lightduer_dcs_router.c( 449): Directive name: Stop

I (145841) AUDIO_WRAPPER: stop_listen, close mic
I (145842) REC_ENG: Recorder trigger stop
I (145877) REC_ENG: Detect suspend, wakeup:0, flag:1
I (145877) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_STOP
I (145877) DUEROS: Dueros DUER_CMD_STOP
I (145877) REC_ENG: Detect suspend, wakeup:0, flag:1
I (145888) DUEROS: --- rec_engine_cb --- REC_EVENT_WAKEUP_END
I (145885) DUEROS: Dueros DUER_CMD_STOP
I (145894) REC_ENG: state idle
I (145756,tid:3ffc28c8) lightduer_dcs_router.c( 449): Directive name: Spea
I (145944) AUDIO_WRAPPER: Playing speak: http://res.iot.baidu.com:80/api/v
Je6FR-7CbuKhxKgZv4k4jUGQ0QHEsUOgnskNvBJKnhTSpx-f-h79P8hS9QeXAqDhbAFwIv9r4e
9BquUbYXmJarno_6eyeCvtRve0w.mp3
I (145955) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:301, MEM Total:41814
es, Inter:192780 Bytes, Dram:133644 Bytes

        field_set: 0xf, port: 80
        field_data[0]: off: 0, len: 4, part: http
        field_data[1]: off: 7, len: 17, part: res.iot.baidu.com
        field_data[2]: off: 25, len: 2, part: 80
        field_data[3]: off: 27, len: 123, part: /api/v1/tts/Je6FR-7CbuKhxK
jUGQ0QHEsUOgnskNvBJKnhTSpx-f-h79P8hS9QeXAqDhbAFwIv9r4eWtKwnA9BquUbYXmJarno
vtRve0w.mp3
        field_data[4]: unset
        field_data[5]: unset
        field_data[6]: unset
I (146005) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5
I (146009) ESP_AUDIO_TASK: It's a decoder.URI:http://res.iot.baidu.com:80/
/tts/Je6FR-7CbuKhxKgZv4k4jUGQ0QHEsUOgnskNvBJKnhTSpx-f-h79P8hS9QeXAqDhbAFwI
tKwnA9BquUbYXmJarno_6eyeCvtRve0w.mp3

I (146010) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:398, MEM Total:41811
es, Inter:192616 Bytes, Dram:133480 Bytes

I (146029) ESP_AUDIO_TASK: 1.IN:[http],CODEC:[mp3],FILTER:[NULL],OUT:[iis]
PLE:[48000]
I (146049) ESP_AUDIO_TASK: 2.Handles,IN:0x3f80f8b4,CODEC:0x3f8196a8,FILTER
25918,OUT:0x3f8193b8
I (146059) ESP_AUDIO_TASK: 2.1 Update all pipelines
I (146073) ESP_AUDIO_TASK: in_rb:0x3f82576c,filter_rb:0x3f819ec8,out_rb:0x
a8
I (146074) ESP_AUDIO_TASK: 3. Previous starting...
I (146078) AUDIO_ELEMENT: [http] Element task created
I (146083) AUDIO_ELEMENT: [mp3] Element task created
I (146089) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (146089) AUDIO_ELEMENT: [iis] Element task created
I (146089) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
I (146089) AUDIO_ELEMENT: [iis] Element task created
I (146095) MP3_DECODER: MP3 opened
I (146107) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1
I (146111) I2S_STREAM: AUDIO_STREAM_WRITER
I (146089) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
E (146159) HTTP_STREAM: Invalid HTTP stream
E (146160) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN
E (146160) ESP_AUDIO_TASK: media_ctrl_task error:ESP_ERR_AUDIO_OPEN
I (146166) AUDIO_WRAPPER: esp_auido status:4,err:10100
I (146166) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5
I (146179) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY, line:485
I (146185) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:4
W (146006,tid:3ffc28c8) lightduer_events.c(  80): [lightduer_ca] <== event
 end = 0x40130480, timespent = 253
0x40130480: duer_engine_data_available at ??:?

W (146218) AUDIO_ELEMENT: IN-[mp3] AEL_IO_ABORT
Encountered error reading when MP3 init
E (146218) AUDIO_ELEMENT: [http] AEL_MSG_CMD_ERROR
E (146218) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN
I (146227) MP3_DECODER: Closed
E (146231) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_ERROR
```

dtboy1995, 2018-08-15:
我插的那个喇叭一点声音都没有。。。

dtboy1995, 2018-08-15:
这是我的dueros profile的内容
```json
{"configures":"{}","bindToken":"b777c9923ed5ec179b2940f8bfef103e","coapPort":443,"token":"T562LstTCFYPDDG3Xb6EnNzjj1Fadgyb","serverAddr":"device.iot.baidu.com","lwm2mPort":443,"uuid":"1609000000000a","rsaCaCrt":"-----BEGIN CERTIFICATE-----\nMIIDUDCCAjgCCQCmVPUErMYmCjANBgkqhkiG9w0BAQUFADBqMQswCQYDVQQGEwJD\nTjETMBEGA1UECAwKU29tZS1TdGF0ZTEOMAwGA1UECgwFYmFpZHUxGDAWBgNVBAMM\nDyouaW90LmJhaWR1LmNvbTEcMBoGCSqGSIb3DQEJARYNaW90QGJhaWR1LmNvbTAe\nFw0xNjAzMTEwMzMwNDlaFw0yNjAzMDkwMzMwNDlaMGoxCzAJBgNVBAYTAkNOMRMw\nEQYDVQQIDApTb21lLVN0YXRlMQ4wDAYDVQQKDAViYWlkdTEYMBYGA1UEAwwPKi5p\nb3QuYmFpZHUuY29tMRwwGgYJKoZIhvcNAQkBFg1pb3RAYmFpZHUuY29tMIIBIjAN\nBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAtbhIeiN7pznzuMwsLKQj2xB02+51\nOvCJ5d116ZFLjecp9qtllqOfN7bm+AJa5N2aAHJtsetcTHMitY4dtGmOpw4dlGqx\nluoz50kWJWQjVR+z6DLPnGE4uELOS8vbKHUoYPPQTT80eNVnl9S9h/l7DcjEAJYC\nIYJbf6+K9x+Ti9VRChvWcvgZQHMRym9j1g/7CKGMCIwkC+6ihkGD/XG40r7KRCyH\nbD53KnBjBO9FH4IL3rGlZWKWzMw3zC6RTS2ekfEsgAtYDvROKd4rNs+uDU9xaBLO\ndXTl5uxgudH2VnVzWtj09OUbBtXcQFD2IhmOl20BrckYul+HEIMR0oDibwIDAQAB\nMA0GCSqGSIb3DQEBBQUAA4IBAQCzTTH91jNh/uYBEFekSVNg1h1kPSujlwEDDf/W\npjqPJPqrZvW0w0cmYsYibNDy985JB87MJMfJVESG/v0Y/YbvcnRoi5gAenWXQNL4\nh2hf08A5wEQfLO/EaD1GTH3OIierKYZ6GItGrz4uFKHV5fTMiflABCdu37ALGjrA\nrIjwjxQG6WwLr9468hkKrWNG3dMBHKvmqO8x42sZOFRJMkqBbKzaBd1uW4xY5XwM\nS1QX56tVrgO0A3S+4dEg5uiLVN4YVP/Vqh4SMtYkL7ZZiZAxD9GtNnhRyFsWlC2r\nOVSdXs1ttZxEaEBGUl7tgsBte556BIvufZX+BXGyycVJdBu3\n-----END CERTIFICATE-----\n","macId":"","version":8974}
```

daiyinger, 2018-08-15:
@dtboy1995 从你的日志看 你那儿语音活动检测(Voice Activity Detection,VAD)没能触发，以我的经验，上电连接服务器成功后，第一次要是没有识别到语音，后面就很难识别到了。只能再次复位。我浏览了代码，只有按下会触发事件，所以短按后松掉直接说话就行。

daiyinger, 2018-08-15:
不知道你用的是不是乐鑫的ESP32-Lyrt开发板，如果喇叭和耳机都没声音的话，可能要检查下硬件了

dtboy1995, 2018-08-15:
@daiyinger 谢谢您，我不太懂这个，重置再说话也是报错，您能和我说一下，能运行dueros这个例子的全过程吗，我看看我遗漏了哪一步，谢谢您。

dtboy1995, 2018-08-15:
@daiyinger，是这个板子，我测试了放音mp3的例子，可以播放啊

daiyinger, 2018-08-15:
@dtboy1995 不好意思哦 我是只是业余爱好者 流程具体的我也没太搞明白

daiyinger, 2018-08-15:
@dtboy1995 反正例子使用方式和板子说明书上描述不太一样，而且识别率低

dtboy1995, 2018-08-15:
@daiyinger ，没事儿，总之谢谢您了，我也就是跑个demo。。。

jason-mao, 2018-10-11:
@dtboy1995 

> 您好，这是我按下rec放开说话的日志
> 
> ```
> I (142735) DUEROS: PERIPH_NOTIFY_KEY_REC
> I (142735) REC_ENG: Recorder trigger start
> I (142739) DUEROS: --- rec_engine_cb --- REC_EVENT_WAKEUP_START
> I (142740) DUEROS: Recv Que DUER_CMD_START
> I (142748) AUDIO_WRAPPER: duer_dcs_get_speaker_state
> I (142899) DUEROS: PERIPH_NOTIFY_KEY_REC_QUIT
> I (143639) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_START
> I (144739) REC_ENG: Wakeup time is out
> I (145654,tid:3ffc28c8) lightduer_dcs_router.c( 449): Directive name: Stop
> 
> I (145841) AUDIO_WRAPPER: stop_listen, close mic
> I (145842) REC_ENG: Recorder trigger stop
> I (145877) REC_ENG: Detect suspend, wakeup:0, flag:1
> I (145877) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_STOP
> I (145877) DUEROS: Dueros DUER_CMD_STOP
> I (145877) REC_ENG: Detect suspend, wakeup:0, flag:1
> I (145888) DUEROS: --- rec_engine_cb --- REC_EVENT_WAKEUP_END
> I (145885) DUEROS: Dueros DUER_CMD_STOP
> I (145894) REC_ENG: state idle
> I (145756,tid:3ffc28c8) lightduer_dcs_router.c( 449): Directive name: Spea
> I (145944) AUDIO_WRAPPER: Playing speak: http://res.iot.baidu.com:80/api/v
> Je6FR-7CbuKhxKgZv4k4jUGQ0QHEsUOgnskNvBJKnhTSpx-f-h79P8hS9QeXAqDhbAFwIv9r4e
> 9BquUbYXmJarno_6eyeCvtRve0w.mp3
> I (145955) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:301, MEM Total:41814
> es, Inter:192780 Bytes, Dram:133644 Bytes
> 
>         field_set: 0xf, port: 80
>         field_data[0]: off: 0, len: 4, part: http
>         field_data[1]: off: 7, len: 17, part: res.iot.baidu.com
>         field_data[2]: off: 25, len: 2, part: 80
>         field_data[3]: off: 27, len: 123, part: /api/v1/tts/Je6FR-7CbuKhxK
> jUGQ0QHEsUOgnskNvBJKnhTSpx-f-h79P8hS9QeXAqDhbAFwIv9r4eWtKwnA9BquUbYXmJarno
> vtRve0w.mp3
>         field_data[4]: unset
>         field_data[5]: unset
>         field_data[6]: unset
> I (146005) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5
> I (146009) ESP_AUDIO_TASK: It's a decoder.URI:http://res.iot.baidu.com:80/
> /tts/Je6FR-7CbuKhxKgZv4k4jUGQ0QHEsUOgnskNvBJKnhTSpx-f-h79P8hS9QeXAqDhbAFwI
> tKwnA9BquUbYXmJarno_6eyeCvtRve0w.mp3
> 
> I (146010) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:398, MEM Total:41811
> es, Inter:192616 Bytes, Dram:133480 Bytes
> 
> I (146029) ESP_AUDIO_TASK: 1.IN:[http],CODEC:[mp3],FILTER:[NULL],OUT:[iis]
> PLE:[48000]
> I (146049) ESP_AUDIO_TASK: 2.Handles,IN:0x3f80f8b4,CODEC:0x3f8196a8,FILTER
> 25918,OUT:0x3f8193b8
> I (146059) ESP_AUDIO_TASK: 2.1 Update all pipelines
> I (146073) ESP_AUDIO_TASK: in_rb:0x3f82576c,filter_rb:0x3f819ec8,out_rb:0x
> a8
> I (146074) ESP_AUDIO_TASK: 3. Previous starting...
> I (146078) AUDIO_ELEMENT: [http] Element task created
> I (146083) AUDIO_ELEMENT: [mp3] Element task created
> I (146089) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
> I (146089) AUDIO_ELEMENT: [iis] Element task created
> I (146089) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1
> I (146089) AUDIO_ELEMENT: [iis] Element task created
> I (146095) MP3_DECODER: MP3 opened
> I (146107) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1
> I (146111) I2S_STREAM: AUDIO_STREAM_WRITER
> I (146089) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1
> E (146159) HTTP_STREAM: Invalid HTTP stream
> E (146160) AUDIO_ELEMENT: [http] AEL_STATUS_ERROR_OPEN
> E (146160) ESP_AUDIO_TASK: media_ctrl_task error:ESP_ERR_AUDIO_OPEN
> I (146166) AUDIO_WRAPPER: esp_auido status:4,err:10100
> I (146166) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5
> I (146179) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY, line:485
> I (146185) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:4
> W (146006,tid:3ffc28c8) lightduer_events.c(  80): [lightduer_ca] <== event
>  end = 0x40130480, timespent = 253
> 0x40130480: duer_engine_data_available at ??:?
> 
> W (146218) AUDIO_ELEMENT: IN-[mp3] AEL_IO_ABORT
> Encountered error reading when MP3 init
> E (146218) AUDIO_ELEMENT: [http] AEL_MSG_CMD_ERROR
> E (146218) AUDIO_ELEMENT: [mp3] AEL_STATUS_ERROR_OPEN
> I (146227) MP3_DECODER: Closed
> E (146231) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_ERROR
> ```

 E (146159) HTTP_STREAM: Invalid HTTP stream 导致了不能播放，这个问题很好再现吗？

closed 2019-12-11 by jason-mao