espressif/esp-adf #53: dueros无法连接

lixiaoxie, 2018-08-10:
我申请了自己的profile
I (223959) DUEROS: duer_start, len:1469
{"configures":"{}","bindToken":"4cfeba812f400c2cd5105a2566cad21b","coapPort":443,"token":"ihLzR544LijFKXxeNBTJLGFyn8LsP7yS","se
rverAddr":"device.iot.baidu.com","lwm2mPort":443,"uuid":"xxxxxxxxxxxx","rsaCaCrt":"-----BEGIN CERTIFICATE-----\nMIIDUDCCAjgCC
QCmVPUErMYmCjANBgkqhkiG9w0BAQUFADBqMQswCQYDVQQGEwJD\nTjETMBEGA1UECAwKU29tZS1TdGF0ZTEOMAwGA1UECgwFYmFpZHUxGDAWBgNVBAMM\nDyouaW90
LmJhaWR1LmNvbTEcMBoGCSqGSIb3DQEJARYNaW90QGJhaWR1LmNvbTAe\nFw0xNjAzMTEwMzMwNDlaFw0yNjAzMDkwMzMwNDlaMGoxCzAJBgNVBAYTAkNOMRMw\nEQY
DVQQIDApTb21lLVN0YXRlMQ4wDAYDVQQKDAViYWlkdTEYMBYGA1UEAwwPKi5p\nb3QuYmFpZHUuY29tMRwwGgYJKoZIhvcNAQkBFg1pb3RAYmFpZHUuY29tMIIBIjAN
\nBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAtbhIeiN7pznzuMwsLKQj2xB02+51\nOvCJ5d116ZFLjecp9qtllqOfN7bm+AJa5N2aAHJtsetcTHMitY4dtGmOpw4
dlGqx\nluoz50kWJWQjVR+z6DLPnGE4uELOS8vbKHUoYPPQTT80eNVnl9S9h/l7DcjEAJYC\nIYJbf6+K9x+Ti9VRChvWcvgZQHMRym9j1g/7CKGMCIwkC+6ihkGD/X
G40r7KRCyH\nbD53KnBjBO9FH4IL3rGlZWKWzMw3zC6RTS2ekfEsgAtYDvROKd4rNs+uDU9xaBLO\ndXTl5uxgudH2VnVzWtj09OUbBtXcQFD2IhmOl20BrckYul+HE
IMR0oDibwIDAQAB\nMA0GCSqGSIb3DQEBBQUAA4IBAQCzTTH91jNh/uYBEFekSVNg1h1kPSujlwEDDf/W\npjqPJPqrZvW0w0cmYsYibNDy985JB87MJMfJVESG/v0Y
/YbvcnRoi5gAenWXQNL4\nh2hf08A5wEQfLO/EaD1GTH3OIierKYZ6GItGrz4uFKHV5fTMiflABCdu37ALGjrA\nrIjwjxQG6WwLr9468hkKrWNG3dMBHKvmqO8x42s
ZOFRJMkqBbKzaBd1uW4xY5XwM\nS1QX56tVrgO0A3S+4dEg5uiLVN4YVP/Vqh4SMtYkL7ZZiZAxD9GtNnhRyFsWlC2r\nOVSdXs1ttZxEaEBGUl7tgsBte556BIvufZ
X+BXGyycVJdBu3\n-----END CERTIFICATE-----\n","macId":"","version":8900}
I (223928,tid:3ffbd110) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f806aa4, length:1469, profile:0x3f8106e4
I (223929,tid:3ffbd110) lightduer_ca_conf.c(  38):     duer_conf_get_string: uuid = xxxxxxxxxxxxxxxx
I (223939,tid:3ffbd110) lightduer_ca_conf.c(  38):     duer_conf_get_string: serverAddr = device.iot.baidu.com
I (223957,tid:3ffbd110) baidu_ca_socket_adp.c( 116): DNS lookup succeeded. IP=180.97.33.165
I (223972,tid:3ffbd110) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (224477,tid:3ffbd110) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f806aa4, length:0, profile:0x0
I (224478,tid:3ffbd110) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (225490,tid:3ffbd110) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f806aa4, length:0, profile:0x0
I (225490,tid:3ffbd110) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (227496,tid:3ffbd110) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f806aa4, length:0, profile:0x0
I (227497,tid:3ffbd110) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (231500,tid:3ffbd110) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f806aa4, length:0, profile:0x0
I (231501,tid:3ffbd110) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (239510,tid:3ffbd110) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f806aa4, length:0, profile:0x0
I (239511,tid:3ffbd110) lightduer_connagent.c( 131): =====> event: 1[DUER_EVT_START], status: -50
E (239517,tid:3ffbd110) lightduer_connagent.c( 178): start fail! status:-50
E (239524,tid:3ffbd110) lightduer_connagent.c( 255): Action failed: event: 1, status: -50
I (239534,tid:3ffbd110) baidu_ca_socket_adp.c( 427): stop timer:0x3f8105a4
I (239544,tid:3ffbd110) baidu_ca_socket_adp.c( 444): start last timer:0x3f8105a4, soc:0x3f8100d8
I (239547,tid:3ffbd110) lightduer_engine.c( 636): no action
I (239552,tid:3ffbd110) lightduer_connagent.c( 131): =====> event: 5[DUER_EVT_STOP], status: -50
E (239561,tid:3ffbd110) lightduer_voice.c( 101): lock failed: rs = -1, status = 1
E (239568,tid:3ffbd110) lightduer_voice.c( 101): lock failed: rs = -1, status = 0
W (239576,tid:3ffbd110) lightduer_engine.c( 332): duer_engine_send not started
E (239582,tid:3ffbd110) lightduer_engine.c( 377): Report failed: rs = -20
W (239589,tid:3ffbd110) lightduer_ds_log.c( 217): CA not connected: -20
I (239595,tid:3ffbd110) lightduer_ds_log_cache.c(  45): cached tracecode: 0x14010009
W (239603,tid:3ffbd110) lightduer_ds_log.c( 228): trace log report fail: {"duer_trace_info":{"code":335609865,"ts":239575}}
I (239614,tid:3ffbd110) lightduer_connagent.c( 228): connect stopped, status:-50!!
E (239857) DUEROS: event: 1
E (239860) DUEROS: event: DUER_EVENT_STOPPED

始终无法连接到dueros

dtboy1995, 2018-08-13:
请问`duer_profile`这个文件中需要配置什么呢，谢谢您

lixiaoxie, 2018-08-13:
按照乐鑫的文档，不需要配置什么。需要再dueros官网申请账号，并建立设备。然后下载profile文件替换就可以了，可是我连接总是失败。。。

jason-mao, 2018-08-13:
@lixiaoxie 失败的log 可以提供上来吗？

jason-mao, 2018-08-13:
@lixiaoxie I (227496,tid:3ffbd110) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f806aa4, length:0, profile:0x0

你的profile 内容的长度为啥是0！

lixiaoxie, 2018-08-13:
@jason-mao 不清楚啊，看之前的log，profile里面的内容解析都是正确的，以下是我proifile里面的内容。我是直接替换的文件

> {"configures":"{}","bindToken":"4cfeba812f400c2cd5105a2566cad21b","coapPort":443,"token":"ihLzR544LijFKXxeNBTJLGFyn8LsP7yS","serverAddr":"device.iot.baidu.com","lwm2mPort":443,"uuid":"15dc0000000001","rsaCaCrt":"-----BEGIN CERTIFICATE-----\nMIIDUDCCAjgCCQCmVPUErMYmCjANBgkqhkiG9w0BAQUFADBqMQswCQYDVQQGEwJD\nTjETMBEGA1UECAwKU29tZS1TdGF0ZTEOMAwGA1UECgwFYmFpZHUxGDAWBgNVBAMM\nDyouaW90LmJhaWR1LmNvbTEcMBoGCSqGSIb3DQEJARYNaW90QGJhaWR1LmNvbTAe\nFw0xNjAzMTEwMzMwNDlaFw0yNjAzMDkwMzMwNDlaMGoxCzAJBgNVBAYTAkNOMRMw\nEQYDVQQIDApTb21lLVN0YXRlMQ4wDAYDVQQKDAViYWlkdTEYMBYGA1UEAwwPKi5p\nb3QuYmFpZHUuY29tMRwwGgYJKoZIhvcNAQkBFg1pb3RAYmFpZHUuY29tMIIBIjAN\nBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAtbhIeiN7pznzuMwsLKQj2xB02+51\nOvCJ5d116ZFLjecp9qtllqOfN7bm+AJa5N2aAHJtsetcTHMitY4dtGmOpw4dlGqx\nluoz50kWJWQjVR+z6DLPnGE4uELOS8vbKHUoYPPQTT80eNVnl9S9h/l7DcjEAJYC\nIYJbf6+K9x+Ti9VRChvWcvgZQHMRym9j1g/7CKGMCIwkC+6ihkGD/XG40r7KRCyH\nbD53KnBjBO9FH4IL3rGlZWKWzMw3zC6RTS2ekfEsgAtYDvROKd4rNs+uDU9xaBLO\ndXTl5uxgudH2VnVzWtj09OUbBtXcQFD2IhmOl20BrckYul+HEIMR0oDibwIDAQAB\nMA0GCSqGSIb3DQEBBQUAA4IBAQCzTTH91jNh/uYBEFekSVNg1h1kPSujlwEDDf/W\npjqPJPqrZvW0w0cmYsYibNDy985JB87MJMfJVESG/v0Y/YbvcnRoi5gAenWXQNL4\nh2hf08A5wEQfLO/EaD1GTH3OIierKYZ6GItGrz4uFKHV5fTMiflABCdu37ALGjrA\nrIjwjxQG6WwLr9468hkKrWNG3dMBHKvmqO8x42sZOFRJMkqBbKzaBd1uW4xY5XwM\nS1QX56tVrgO0A3S+4dEg5uiLVN4YVP/Vqh4SMtYkL7ZZiZAxD9GtNnhRyFsWlC2r\nOVSdXs1ttZxEaEBGUl7tgsBte556BIvufZX+BXGyycVJdBu3\n-----END CERTIFICATE-----\n","macId":"","version":8900}

dtboy1995, 2018-08-14:
profile 文件下载下来是一个压缩包，里面有很多文件，不懂啊。。求解，谢谢

Chenzejun-Dummkopf, 2018-10-09:
我想问下 duer 登陆失败 的问题解决了没有，我也遇到这个问题，log和你的一模一样
我用的esp32lyrat 自带的duer_profile，但是始终无法登陆
@dtboy1995 @lixiaoxie @jason-mao @projectgus @kedars

Chenzejun-Dummkopf, 2018-10-09:
我现在将我的log贴出来，大家看下：
rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5900
load:0x40078000,len:7796
load:0x40080000,len:7440
entry 0x400803a0
\0x1b[0;32mI (29) boot: ESP-IDF  2nd stage bootloader\0x1b[0m
\0x1b[0;32mI (29) boot: compile time 09:55:36\0x1b[0m
\0x1b[0;32mI (31) boot: Enabling RNG early entropy source...\0x1b[0m
\0x1b[0;32mI (33) boot: SPI Speed      : 80MHz\0x1b[0m
\0x1b[0;32mI (37) boot: SPI Mode       : DIO\0x1b[0m
\0x1b[0;32mI (41) boot: SPI Flash Size : 4MB\0x1b[0m
\0x1b[0;32mI (45) boot: Partition Table:\0x1b[0m
\0x1b[0;32mI (49) boot: ## Label            Usage          Type ST Offset   Length\0x1b[0m
\0x1b[0;32mI (56) boot:  0 nvs              WiFi data        01 02 00009000 00006000\0x1b[0m
\0x1b[0;32mI (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000\0x1b[0m
\0x1b[0;32mI (71) boot:  2 avs              WiFi data        01 02 00010000 00006000\0x1b[0m
\0x1b[0;32mI (79) boot:  3 factory          factory app      00 00 00020000 00300000\0x1b[0m
\0x1b[0;32mI (86) boot: End of partition table\0x1b[0m
\0x1b[0;32mI (90) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x11b318 (1159960) map\0x1b[0m
\0x1b[0;32mI (437) esp_image: segment 1: paddr=0x0013b340 vaddr=0x3ffb0000 size=0x041a8 ( 16808) load\0x1b[0m
\0x1b[0;32mI (442) esp_image: segment 2: paddr=0x0013f4f0 vaddr=0x3ffb41a8 size=0x00000 (     0) load\0x1b[0m
\0x1b[0;32mI (443) esp_image: segment 3: paddr=0x0013f4f8 vaddr=0x40080000 size=0x00400 (  1024) load\0x1b[0m
\0x1b[0;32mI (453) esp_image: segment 4: paddr=0x0013f900 vaddr=0x40080400 size=0x00710 (  1808) load\0x1b[0m
\0x1b[0;32mI (462) esp_image: segment 5: paddr=0x00140018 vaddr=0x400d0018 size=0xbdc50 (777296) map\0x1b[0m
\0x1b[0;32mI (696) esp_image: segment 6: paddr=0x001fdc70 vaddr=0x40080b10 size=0x18000 ( 98304) load\0x1b[0m
\0x1b[0;32mI (731) esp_image: segment 7: paddr=0x00215c78 vaddr=0x400c0000 size=0x00000 (     0) load\0x1b[0m
\0x1b[0;32mI (732) esp_image: segment 8: paddr=0x00215c80 vaddr=0x50000000 size=0x00000 (     0) load\0x1b[0m
\0x1b[0;32mI (753) boot: Loaded app from partition at offset 0x20000\0x1b[0m
\0x1b[0;32mI (753) boot: Disabling RNG early entropy source...\0x1b[0m
\0x1b[0;32mI (754) spiram: SPI RAM mode: flash 80m sram 80m\0x1b[0m
\0x1b[0;32mI (758) spiram: PSRAM initialized, cache is in low/high (2-core) mode.\0x1b[0m
\0x1b[0;32mI (765) cpu_start: Pro cpu up.\0x1b[0m
\0x1b[0;32mI (769) cpu_start: Starting app cpu, entry point is 0x40081484\0x1b[0m
\0x1b[0;32mI (0) cpu_start: App cpu up.\0x1b[0m
\0x1b[0;32mI (1269) spiram: SPI SRAM memory test OK\0x1b[0m
\0x1b[0;32mI (1270) heap_init: Initializing. RAM available for dynamic allocation:\0x1b[0m
\0x1b[0;32mI (1270) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM\0x1b[0m
\0x1b[0;32mI (1276) heap_init: At 3FFBB178 len 00024E88 (147 KiB): DRAM\0x1b[0m
\0x1b[0;32mI (1282) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM\0x1b[0m
\0x1b[0;32mI (1289) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM\0x1b[0m
\0x1b[0;32mI (1295) heap_init: At 40098B10 len 000074F0 (29 KiB): IRAM\0x1b[0m
\0x1b[0;32mI (1302) cpu_start: Pro cpu start user code\0x1b[0m
\0x1b[0;32mI (1306) spiram: Adding pool of 4096K of external SPI memory to heap allocator\0x1b[0m
\0x1b[0;32mI (1314) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations\0x1b[0m
\0x1b[0;32mI (217) cpu_start: Starting scheduler on PRO CPU.\0x1b[0m
\0x1b[0;32mI (0) cpu_start: Starting scheduler on APP CPU.\0x1b[0m
\0x1b[0;32mI (218) duer: ==== VERSION: v1 ====\0x1b[0m

 ******* nvs_flash_init ******* 
\0x1b[0;32mI (258) gpio: GPIO[33]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 \0x1b[0m
.........................Initializing audio codec..............................
\0x1b[0;31mE (2268) CODEC_ALC5650: alc5650_read_reg error\0x1b[0m
\0x1b[0;31mE (2268) AudioCodec: Error: Failed to initialize codec.
\0x1b[0m
\0x1b[0;31mE (2268) AudioCodec: Could not read the Device ID.
\0x1b[0m

 ******* alc5650_init ******* 
\0x1b[0;31mE (6268) UI_LED: Led driver init failed\0x1b[0m

 ******* ui_led_init ******* 
\0x1b[0;32mI (6268) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:3 \0x1b[0m
\0x1b[0;32mI (6268) gpio: GPIO[37]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 1| Intr:3 \0x1b[0m
\0x1b[0;32mI (6278) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 \0x1b[0m
\0x1b[0;32mI (6288) gpio: GPIO[33]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 \0x1b[0m
[... 83 lines cut ...]
 ******* EspAudioAllocInner: heap_caps_malloc(n * size, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT); *******

 ******* duer_all_login: len:1447
{"configures":"{}","bindToken":"08c94f9b5b46c1657b11648bd9f8005f","coapPort":9001,"token":"5YxN55cQ","serverAddr":"device.iot.baidu.com","lwm2mPort":9001,"uuid":"037f0000000001","rsaCaCrt":"-----BEGIN CERTIFICATE-----\nMIIDUDCCAjgCCQCmVPUErMYmCjANBgkqhkiG9w0BAQUFADBqMQswCQYDVQQGEwJD\nTjETMBEGA1UECAwKU29tZS1TdGF0ZTEOMAwGA1UECgwFYmFpZHUxGDAWBgNVBAMM\nDyouaW90LmJhaWR1LmNvbTEcMBoGCSqGSIb3DQEJARYNaW90QGJhaWR1LmNvbTAe\nFw0xNjAzMTEwMzMwNDlaFw0yNjAzMDkwMzMwNDlaMGoxCzAJBgNVBAYTAkNOMRMw\nEQYDVQQIDApTb21lLVN0YXRlMQ4wDAYDVQQKDAViYWlkdTEYMBYGA1UEAwwPKi5p\nb3QuYmFpZHUuY29tMRwwGgYJKoZIhvcNAQkBFg1pb3RAYmFpZHUuY29tMI [... cut ...]
I (1391,tid:3ffd6804) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (2391,tid:3ffd6804) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f804e24, length:0, profile:0x0
I (2391,tid:3ffd6804) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (4391,tid:3ffd6804) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f804e24, length:0, profile:0x0
I (4391,tid:3ffd6804) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)

我现在也没有想到解决办法，希望知道答案的同志告知一声

Chenzejun-Dummkopf, 2018-10-10:
I (321,tid:3ffd8cec) baidu_ca_socket_adp.c( 116): DNS lookup succeeded. IP=117.185.16.145
I (325,tid:3ffd8cec) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (825,tid:3ffd8cec) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f800c8c, length:0, profile:0x0
I (825,tid:3ffd8cec) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (1825,tid:3ffd8cec) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f800c8c, length:0, profile:0x0
I (1825,tid:3ffd8cec) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (3825,tid:3ffd8cec) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f800c8c, length:0, profile:0x0
I (3825,tid:3ffd8cec) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (7825,tid:3ffd8cec) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f800c8c, length:0, profile:0x0
I (7825,tid:3ffd8cec) lightduer_connagent.c( 169): will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (15825,tid:3ffd8cec) lightduer_engine.c( 174): duer_engine_start, g_handler:0x3f800c8c, length:0, profile:0x0
I (15825,tid:3ffd8cec) lightduer_connagent.c( 131): =====> event: 1[DUER_EVT_START], status: -50
E (15825,tid:3ffd8cec) lightduer_connagent.c( 178): start fail! status:-50
E (15826,tid:3ffd8cec) lightduer_connagent.c( 255): Action failed: event: 1, status: -50
I (15827,tid:3ffd8cec) baidu_ca_socket_adp.c( 427): stop timer:0x3f80aea4
I (15828,tid:3ffd8cec) baidu_ca_socket_adp.c( 444): start last timer:0x3f80aea4, soc:0x3f801da8
I (15828,tid:3ffd8cec) lightduer_engine.c( 636): no action
I (15829,tid:3ffd8cec) lightduer_connagent.c( 131): =====> event: 5[DUER_EVT_STOP], status: -50
E (15830,tid:3ffd8cec) lightduer_voice.c( 101): lock failed: rs = -1, status = 1
E (15830,tid:3ffd8cec) lightduer_voice.c( 101): lock failed: rs = -1, status = 0
W (15831,tid:3ffd8cec) lightduer_engine.c( 332): duer_engine_send not started
E (15832,tid:3ffd8cec) lightduer_engine.c( 377): Report failed: rs = -20
W (15832,tid:3ffd8cec) lightduer_ds_log.c( 217): CA not connected: -20
I (15833,tid:3ffd8cec) lightduer_ds_log_cache.c(  45): cached tracecode: 0x14010009
W (15834,tid:3ffd8cec) lightduer_ds_log.c( 228): trace log report fail: {"duer_trace_info":{"code":335609865,"ts":15831}}
I (15835,tid:3ffd8cec) lightduer_connagent.c( 228): connect stopped, status:-50!!

 ******* duer_event_hook: begin *******

 ******* duer_event_hook: event->_event: DUER_EVENT_STOPPED;*******

jason-mao, 2018-10-11:
@Chenzejun-Dummkopf 
dueros一直没有发送包出去，最后导致了握手连接失败。
你的IDF用的是ADF默认的吗？建议使用自己申请的profile。`037f0000000001 `不建议使用，因为会有很多人同时用。

Chenzejun-Dummkopf, 2018-10-11:
我的IDF是我自己做的，我想请问您，你们的ADF在原来ESP-IDF的基础上做了什么更新呢，或许这就是我dueros连接不上的原因吧

Chenzejun-Dummkopf, 2018-10-11:
我使用了自己的profile，结果还是出现 dueros 登陆失败的情况，问题如出一辙
@jason-mao

Chenzejun-Dummkopf, 2018-10-11:
解决了这个问题了，谢谢大家

closed 2019-01-03 by jason-mao