espressif/esp-adf #54: 在编译上传时候报错 修改了环境变量了printenv PATH  也能正确显示出完整路径

chenguang3312, 2018-08-13:
pi@raspberrypi:~/esp/play_mp3$ make flash
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
/bin/sh: 1: xtensa-esp32-elf-gcc: Exec format error
WARNING: Failed to find Xtensa toolchain, may need to alter PATH or set one in the configuration menu
CC build/bootloader/bootloader_support/src/bootloader_flash.o
/home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: 1: /home/pi/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc: Syntax error: "(" unexpected
/home/pi/esp/esp-adf/esp-idf/make/component_wrapper.mk:273: recipe for target 'src/bootloader_flash.o' failed
make[2]: *** [src/bootloader_flash.o] Error 2
/home/pi/esp/esp-adf/esp-idf/make/project.mk:450: recipe for target 'component-bootloader_support-build' failed
make[1]: *** [component-bootloader_support-build] Error 2
/home/pi/esp/esp-adf/esp-idf/components/bootloader/Makefile.projbuild:40: recipe for target '/home/pi/esp/play_mp3/build/bootloader/bootloader.bin' failed
make: *** [/home/pi/esp/play_mp3/build/bootloader/bootloader.bin] Error 2

用的树莓派3b+     谢谢

ahhfzhang, 2018-09-11:
I think you need a ESP32 LyraT board first.
https://github.com/espressif/esp-adf/blob/master/README.md

jason-mao, 2019-01-03:
@chenguang3312 这个问题怎么样了？

closed 2019-05-22 by jason-mao