espressif/esp-adf #60: 新购买的ESP32_LyarT开发板 自带的.bin文件是怎么生成的？

xiaoseyihe, 2018-09-10:
新购买的ESP32_LyarT开发板 自带的.bin文件是怎么生成的？按照ESP32_LyarT使用指南，怎么生成audio-esp.bin  esp32-audio-app.bin  partitions_esp_audio.bin  Bootloader.bin文件，能够连网实现语言唤醒功能？按照《ESP-IDF 编程指南》已跑通基本用例

ahhfzhang, 2018-09-11:
自带的bin也是通过make来生成的，你可以尝试去跑一下esp-adf里面的一些example，尝试一些功能，唤醒会在后续的版本中提供。

xiaoseyihe, 2018-09-12:
谢谢回答，你的意思目前版本不支持 唤醒词 唤醒功能？ 只能按键REC 可以进行简单对话？
还有个问题，如果前端对语音数据进行 降噪处理、回声消除等，这些算法可以运行在ESP32-WROVER模组上吗？

jason-mao, 2018-10-11:
@xiaoseyihe 

> 谢谢回答，你的意思目前版本不支持 唤醒词 唤醒功能？ 只能按键REC 可以进行简单对话？
> 还有个问题，如果前端对语音数据进行 降噪处理、回声消除等，这些算法可以运行在ESP32-WROVER模组上吗？

1. 是的，只能REC对话。带唤醒的feature很快会到master分支。
2.可以的

jason-mao, 2019-01-03:
@xiaoseyihe 这个问题怎么样了？

tanhailong, 2019-04-04:
> 新购买的ESP32_LyarT开发板 自带的.bin文件是怎么生成的？按照ESP32_LyarT使用指南，怎么生成audio-esp.bin esp32-audio-app.bin partitions_esp_audio.bin Bootloader.bin文件，能够连网实现语言唤醒功能？按照《ESP-IDF 编程指南》已跑通基本用例

请问你这个固件是哪儿下载得呀，我没有找到这个固件，我也有块这个板子，把默认出厂固件给刷了

tanhailong, 2019-04-04:
> @xiaoseyihe 这个问题怎么样了？

现在 新版本支持语音唤醒了吗？

jason-mao, 2019-04-04:
> > 新购买的ESP32_LyarT开发板 自带的.bin文件是怎么生成的？按照ESP32_LyarT使用指南，怎么生成audio-esp.bin esp32-audio-app.bin partitions_esp_audio.bin Bootloader.bin文件，能够连网实现语言唤醒功能？按照《ESP-IDF 编程指南》已跑通基本用例
> 
> 请问你这个固件是哪儿下载得呀，我没有找到这个固件，我也有块这个板子，把默认出厂固件给刷了

你可以编译example/Dueros， 基本是完整的dueros功能。

> > @xiaoseyihe 这个问题怎么样了？
> 
> 现在 新版本支持语音唤醒了吗？

已经支持了，也在examples/dueros 里面

tanhailong, 2019-04-09:
> 你可以编译example/Dueros， 基本是完整的dueros功能。
>
>
> 已经支持了，也在examples/dueros 里面

我编译这个出来 一直没有连接服务器成功呀，还有一个疑问 我拿到了 这个板子得固件烧录进去 有提示音 比如 网络连接成功，服务器连接成功 这个是通过什么方式做的呀  我猜想音频文件就是audio-esp.bin这个文件呀，还有这个音频文件得烧录位置是怎样确定得呀

jason-mao, 2019-05-22:
> 我编译这个出来 一直没有连接服务器成功呀，还有一个疑问 我拿到了 这个板子得固件烧录进去 有提示音 比如 网络连接成功，服务器连接成功 这个是通过什么方式做的呀 我猜想音频文件就是audio-esp.bin这个文件呀，还有这个音频文件得烧录位置是怎样确定得呀

这些是我们通过商务的方式提供的。

closed 2019-05-22 by jason-mao