espressif/esp-adf #63: 怎么让ADF关联最新的IDF工程？

xiaoseyihe, 2018-09-26:
怎么让ADF关联最新的IDF工程？

krzychb, 2018-09-26:
Hi @xiaoseyihe,

The ADF is tested and released to work with specific version of the IDF.
As of today this is v3.0.1-rc.
If you update IDF to the latest release and try to compile some ADF projects with it, then likely compilation errors will be dropped.

xiaoseyihe, 2018-09-26:
感谢你的回答，我想在ADF 工程中使用fopen函数，例如在dueros测试用例中，但是总是报错，在新版本的IDF工程可以测试通过。

xiaoseyihe, 2018-09-26:
我想把文件保存到SD 卡上，能够把文件读出来，有哪些方法可以实现？

krzychb, 2018-09-26:
Have you checked [recorder/pipeline_wav_sdcard](https://github.com/espressif/esp-adf/tree/4caaf7437e383b34d6d2b5c343ac666a4200399a/examples) example?

It is saving a file to the SD card of ESP32 LyraT board. 
One of underlining functions used by this examples is [fopen()](https://github.com/espressif/esp-adf/blob/4caaf7437e383b34d6d2b5c343ac666a4200399a/components/audio_stream/fatfs_stream.c#L104).

closed 2018-11-09 by jason-mao