espressif/esp-adf #64: some TS AAC not supported

myvobot, 2018-09-26:
save the test.ts into sdcard, and run CLI example, try with 'play 4'
[test.ts.zip](https://github.com/espressif/esp-adf/files/2420735/test.ts.zip)

then, it will output:

```
...
I (20452) AAC_DECODER: New audio song
I (20453) AAC_DECODER: This audio is TS AAC
I (20511) AAC_DECODER: I2S setup: sample_rate = 48000 : channels = 2
I (20513) AAC_DECODER: M4A sampleRate=48000, Channels=2, frameSize=1024
I (20441) AUDIO_ELEMENT: [iis] Element task created
I (20540) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1
I (20542) I2S_STREAM: AUDIO_STREAM_WRITER
I (20551) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (20569) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5
I (20571) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY, line:525
I (20574) CONSOLE_EXAMPLE: esp_auido status:1,err:0

I (20576) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:1
I (20597) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:260, MEM Total:4146200 Bytes, Inter:186424 Bytes, Dram:146672 Bytes

I (20615) AUDIO_PIPELINE: Pipeline started
esp32> E (20840) AAC_DECODER: Decoder encountered error 313 30
E (20843) AAC_DECODER: Decoder encountered error 313 30
E (20846) AAC_DECODER: Decoder encountered error 313 30
E (20849) AAC_DECODER: Decoder encountered error 313 30
...
```
BTW, some m4a not working as well:

```
I (10142) AAC_DECODER: New audio song
I (10146) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1
I (10157) I2S_STREAM: AUDIO_STREAM_WRITER
W (10190) FATFS_STREAM: No more data,ret:0
I (10192) AUDIO_ELEMENT: IN-[file] AEL_IO_DONE,0
I (10193) AUDIO_ELEMENT: IN-[m4a] AEL_IO_DONE,-2
I (10205) AUDIO_ELEMENT: IN-[m4a] AEL_IO_DONE,-2
E (10207) AAC_DECODER: currently not support mvdt ahead of moov case,0
E (10209) AAC_DECODER: error mp4 stream, release
E (10220) AUDIO_ELEMENT: [m4a] AEL_STATUS_ERROR_OPEN
I (10221) AAC_DECODER: Closed
E (10232) AUDIO_ELEMENT: [m4a] AEL_MSG_CMD_ERROR
E (10225) ESP_AUDIO_TASK: media_ctrl_task error:ESP_ERR_AUDIO_OPEN
I (10237) CONSOLE_EXAMPLE: esp_auido status:4,err:10100
```

myvobot, 2018-09-26:
despite the TS problem, even if I extract AAC audio from the TS, it's also not playable. However, some other AAC files are OK.

[test.aac.zip](https://github.com/espressif/esp-adf/files/2420943/test.aac.zip)

```
I (3584) FATFS_STREAM: File size is 485338 byte,pos:0
I (3587) AUDIO_ELEMENT: [aac] AEL_MSG_CMD_RESUME,state:1
I (3599) AAC_DECODER: New audio song
I (3601) AAC_DECODER: This audio is RAW AAC
I (3657) AAC_DECODER: I2S setup: sample_rate = 48000 : channels = 2
I (3659) AAC_DECODER: M4A sampleRate=48000, Channels=2, frameSize=1024
I (3590) AUDIO_ELEMENT: [iis] Element task created
I (3692) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1
I (3693) I2S_STREAM: AUDIO_STREAM_WRITER
I (3749) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
...
Decoder encountered error 503 30
```

dayhhywh, 2019-01-04:
The test.ts in sdcard can be played normally. 
 After testing, there is some errors in the different position of the test.aac  stream. would you tell me where the music is downloaded?
For m4a stream , can you provide audio files  for me to test?

keelanleyser, 2019-03-12:
Ive been working on this issue, and found that some M4A files give this error:

AAC_DECODER: currently not support mvdt ahead of moov case,0

The issue is that the 'moov' atom (header data) is located at the end of the m4a file. ESP-ADF only seems to support files where the 'moov' data is at the beginning of the file for progressive download, ie, start playing while you are still downloading.  

The easy fix for your m4a file is to move the 'moov' atom to the beginning with ffmpeg:

ffmpeg -i input.m4a -c:a copy -movflags +faststart output.m4a

input.m4a being your file.

Hope this helps! 

Dom

NikkSaan, 2021-03-04:
Thank you @keelanleyser for the clarification! 

I found a program which optimises m4a for streaming [here](https://hydrogenaud.io/index.php?topic=45231.0). I don't know if it is better than ffmpeg, but I thought I will post it as an alternative...

commit 185bb35e (2018-06-26): Merge branch 'feature/pipeline_sdcard_control_example' into 'master'

Provided an example how to control a pipeline to play music files from a sdcard.…

Closes #64

See merge request adf/esp-adf-internal!153
closed 2022-02-08 by jason-mao