espressif/esp-adf #66: 当前版本esp-adf跑dueros有问题

diskman88, 2018-10-09:
录音后后出现以下提示后，进入gdb:

input and output channel must be 1 or 2 channelE (10086) RSP_FILTER: Allocate memory failed,line:57
E (10096) AUDIO_ELEMENT: [resample] AEL_STATUS_ERROR_OPEN
I (10107) RSP_FILTER: rsp_filter_close
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC      : 0x40153433  PS      : 0x00060f30  A0      : 0x80152281  A1      : 0x3ffe60f0  
0x40153433: resample_close at /Users/maojianxin/duros/esp-adf-libs-source-dev/esp_codec/./resample/resample.c:602

A2      : 0x00000000  A3      : 0x00000003  A4      : 0x3f41c5e3  A5      : 0x3ffb6bb4  
A6      : 0x00000003  A7      : 0x3ffb6b38  A8      : 0x8008568d  A9      : 0x3ffe6080  
A10     : 0x00000032  A11     : 0x3ffe62b0  A12     : 0x3ffe60c0  A13     : 0x0000000c  
A14     : 0x00000001  A15     : 0x00000000  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000008  LBEG    : 0x40091071  LEND    : 0x40091081  LCOUNT  : 0xfffffffd  
0x40091071: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:84

0x40091081: strlen at /home/jeroen/esp8266/esp32/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/machine/xtensa/../../../../.././newlib/libc/machine/xtensa/strlen.S:96

Backtrace: 0x40153433:0x3ffe60f0 0x4015227e:0x3ffe6110 0x4012c60b:0x3ffe6130 0x4012d411:0x3ffe6160 0x4012cdfc:0x3ffe61a0
0x40153433: resample_close at /Users/maojianxin/duros/esp-adf-libs-source-dev/esp_codec/./resample/resample.c:602

0x4015227e: rsp_filter_close at /home/jiang/share/esp-adf/components/esp-adf-libs/esp_codec/filter_resample.c:97

0x4012c60b: audio_element_on_cmd at /home/jiang/share/esp-adf/components/audio_pipeline/./audio_element.c:681

0x4012d411: audio_event_iface_waiting_cmd_msg at /home/jiang/share/esp-adf/components/audio_pipeline/./audio_event_iface.c:243 (discriminator 1)

0x4012cdfc: audio_element_task at /home/jiang/share/esp-adf/components/audio_pipeline/./audio_element.c:681

Entering gdb stub now.
$T0b#e6GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/jiang/share/esp-adf/ai-esp32/build/esp_dueros.elf...done.
Remote debugging using /dev/ttyUSB0
resample_close (user_data=0x0)
    at /Users/maojianxin/duros/esp-adf-libs-source-dev/esp_codec/./resample/resample.c:602
602	/Users/maojianxin/duros/esp-adf-libs-source-dev/esp_codec/./resample/resample.c: No such file or directory.
(gdb)

diskman88, 2018-10-10:
之前下载的版本没出现这个问题，我把旧版本components下的audio_pipeline替换新版本的，编译后没出现这个问题。我就是用官方的板跑那个dueros例程。

diskman88, 2018-10-10:
另：我用你的录音引擎录音，云端识别率好低。我做下以下改动感觉好很多：
1、发送语音数据前先发wav文件头。
2、因为你的录音引擎做成了库，我不知里面怎么实现，没使用录音引擎，改成定时录音方式：按键触发录音->发语音数据->定时3秒后停止录音。这样也没有使用VAD。

kirbyzjj, 2018-10-10:
我使用中也发现类似的问题
1. 死机是因为rsp_filter_open中src_rate和src_ch都变成0, 导致调用resample_open时死机
2. 用dueros例程，点播歌曲或点儿童故事时，不能播放是因为返回的资源中带有#号， 导致esp_audio把这个#号后面的内容当成iis的tag, 结果查不到这个element. esp_audio不开源也无法修复，只能自己写个类似的。希望官方尽快修复这些问题。

另外提一句，目前在播放dueros返回的资源时无法播放完整，比如这个连接http://od.qingting.fm/m4a/591be05b7cb8913c4b637c39_7329384_24.m4a?from=dumi&logid=dmziu-15d758aeec00000_4#1_0必定在固定的一个位置tcp 连接会断开。很影响使用.
另外建议idf能尽快升级到3.1版本，目前我自己升级过去发现dueros是无法连接了.

jason-mao, 2018-10-11:
@diskman88 
> 另外还有几个问题：
> 1、录音引擎不能完整录完一句话。这个录音结束我不知道是录音引擎发出的还是百度的DCS发出的。
> 2、用44100 48000 16bit 双通道录音后，发回放时语速变快，有杂音，录一段慢节奏的歌，然后回放，很明感觉到。我用例程pipeline_wav_sdcard测的，但用16000 16bit 单通道正常。
> 3、用dueros例程，点播歌曲或点儿童故事时，不能播放。但播放器会提示播放xxx歌。问天气、问时间没问题。
问题1. VAD 有检测，当静音的情况会主动stop
问题2. resmaple 有问题在修复中
问题3. 不能播，具体的log是什么样的？

jason-mao, 2018-10-11:
@kirbyzjj 

> 我使用中也发现类似的问题
>
> 1. 死机是因为rsp_filter_open中src_rate和src_ch都变成0, 导致调用resample_open时死机
> 2. 用dueros例程，点播歌曲或点儿童故事时，不能播放是因为返回的资源中带有#号， 导致esp_audio把这个#号后面的内容当成iis的tag, 结果查不到这个element. [...]

IDF3.1已经在验证了，对应的dueros会升级到3.0.7版本，目前他们之间不能兼容

diskman88, 2018-10-11:
这是我发“请讲小马过河的故事后”打印的log:
[0;32mI (27402) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m
I (27687,tid:3ffbbe4c) lightduer_dcs_router.c( 449): Directive name: Speak
[0;32mI (27814) AUDIO_WRAPPER: Playing speak: http://res.iot.baidu.com:80/api/v1/tts/c_pMbza0rh_sfKrpePvr-7BKwUsXoBej1aSm0Qs7jHj5f0fJISPiAlz8ePSI_0sS6xkiE6QOu8lJfDply_o2Ep3z1x4fs13_CnL8UmzTodE.mp3[0m
[0;32mI (27826) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:301, MEM Total:4229024 Bytes, Inter:202584 Bytes, Dram:140996 Bytes
[0m
	field_set: 0xf, port: 80
	field_data[0]: off: 0, len: 4, part: http
	field_data[1]: off: 7, len: 17, part: res.iot.baidu.com
	field_data[2]: off: 25, len: 2, part: 80
	field_data[3]: off: 27, len: 123, part: /api/v1/tts/c_pMbza0rh_sfKrpePvr-7BKwUsXoBej1aSm0Qs7jHj5f0fJISPiAlz8ePSI_0sS6xkiE6QOu8lJfDply_o2Ep3z1x4fs13_CnL8UmzTodE.mp3
	field_data[4]: unset
	field_data[5]: unset
	field_data[6]: unset
[0;32mI (27873) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5[0m
[0;32mI (27879) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:398, MEM Total:4228716 Bytes, Inter:202420 Bytes, Dram:140832 Bytes
[0m
[0;32mI (27879) ESP_AUDIO_TASK: It's a decoder.URI:http://res.iot.baidu.com:80/api/v1/tts/c_pMbza0rh_sfKrpePvr-7BKwUsXoBej1aSm0Qs7jHj5f0fJISPiAlz8ePSI_0sS6xkiE6QOu8lJfDply_o2Ep3z1x4fs13_CnL8UmzTodE.mp3
[0m
[0;32mI (27914) ESP_AUDIO_TASK: 1.IN:[http],CODEC:[mp3],FILTER:[NULL],OUT:[iis],RESAMPLE:[48000][0m
[0;32mI (27919) ESP_AUDIO_TASK: 2.Handles,IN:0x3f80eb44,CODEC:0x3f8188e8,FILTER:0x3f819b9c,OUT:0x3f818590[0m
[0;32mI (27928) ESP_AUDIO_TASK: 2.1 Update all pipelines[0m
[0;32mI (27935) ESP_AUDIO_TASK: in_rb:0x3f819fa0,filter_rb:0x3f819cbc,out_rb:0x3f819c48[0m
[0;32mI (27949) ESP_AUDIO_TASK: 3. Previous starting...[0m
[0;32mI (27949) AUDIO_ELEMENT: [http] Element task created[0m
[0;32mI (27953) AUDIO_ELEMENT: [mp3] Element task created[0m
[0;32mI (27958) AUDIO_ELEMENT: [mp3] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (27958) AUDIO_ELEMENT: [iis] Element task created[0m
[0;32mI (27964) MP3_DECODER: MP3 opened[0m
[0;32mI (27970) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (27980) I2S_STREAM: AUDIO_STREAM_WRITER[0m
[0;32mI (27959) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (28082) MP3_DECODER: I2S setup: sample_rate = 16000 : channels = 1[0m
[0;32mI (28087) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5[0m
[0;32mI (28089) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY, line:525[0m
[0;32mI (28091) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0[0m
[0;32mI (28097) AUDIO_ELEMENT: [resample] Element task created[0m
[0;32mI (28103) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:260, MEM Total:4128148 Bytes, Inter:175260 Bytes, Dram:113672 Bytes
[0m
[0;32mI (28122) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (28122) RSP_FILTER: rsp_filter_open, decoder, src:16000,1,dest:48000,2,in:1024, out_pcm:1576[0m
[0;32mI (28123) AUDIO_PIPELINE: Pipeline started[0m
W (28013,tid:3ffbbe4c) lightduer_events.c(  80): [lightduer_ca] <== event end = 0x4013077c, timespent = 327
[0;33mW (28144) HTTP_STREAM: No more data,errno:0[0m
I (28025,tid:3ffbbe4c) lightduer_dcs_router.c( 449): Directive name: Play
[0;33mW (28157) HTTP_STREAM: there are no track[0m
I (28033,tid:3ffbbe4c) lightduer_dcs_audio_player.c( 522): token: eyJib3RfaWQiOiJhdWRpb191bmljYXN0X2NoaWxkcmVuIiwicmVzdWx0X3Rva2VuIjoiMDcxNDM2YzA1OTc2ZDAxYTJkNzg3NjVjZmFmMDgzODQiLCJib3RfdG9rZW4iOiI1Nzc3OTYzNTcyNCJ9
[0;32mI (28161) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0[0m
I (28058,tid:3ffbbe4c) lightduer_dcs_router.c( 449): Directive name: RenderHint
E (28067,tid:3ffbbe4c) lightduer_dcs_router.c( 459): unrecognized directive: RenderHint

[0;32mI (28144) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
[0;32mI (28200) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m
[0;32mI (30824) AUDIO_ELEMENT: IN-[mp3] AEL_IO_DONE,-2[0m
[0;32mI (31913) MP3_DECODER: MP3 decodig done[0m
[0;32mI (31913) MP3_DECODER: Closed[0m
[0;32mI (32069) AUDIO_ELEMENT: IN-[resample] AEL_IO_DONE,-2[0m
[0;32mI (32069) RSP_FILTER: rsp_filter_close[0m
[0;32mI (32100) AUDIO_ELEMENT: IN-[iis] AEL_IO_DONE,-2[0m
[0;32mI (32138) AUDIO_WRAPPER: esp_auido status:3,err:0[0m
[0;32mI (32140) AUDIO_WRAPPER: Playing audio offset:0 url:http://od.qingting.fm/vod/00/00/0000000000000000000020706574_24.m4a?from=dumi&logid=n9aks-16d23e326800000_7#1_0[0m
[0;32mI (32148) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:301, MEM Total:4174052 Bytes, Inter:178872 Bytes, Dram:117284 Bytes
[0m
	field_set: 0x3b, port: 0
	field_data[0]: off: 0, len: 4, part: http
	field_data[1]: off: 7, len: 14, part: od.qingting.fm
	field_data[2]: unset
	field_data[3]: off: 21, len: 46, part: /vod/00/00/0000000000000000000020706574_24.m4a
	field_data[4]: off: 68, len: 39, part: from=dumi&logid=n9aks-16d23e326800000_7
	field_data[5]: off: 108, len: 3, part: 1_0
[... 83 lines cut ...]
	field_data[0]: off: 0, len: 4, part: http
	field_data[1]: off: 7, len: 14, part: od.qingting.fm
	field_data[2]: unset
	field_data[3]: off: 21, len: 46, part: /vod/00/00/0000000000000000000020628362_24.m4a
	field_data[4]: off: 68, len: 39, part: from=dumi&logid=n9aks-16d23ede8400001_0
	field_data[5]: off: 108, len: 3, part: 1_0
	field_data[6]: unset
[0;32mI (34338) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:0, Available:5[0m
[0;32mI (34345) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:398, MEM Total:4173068 Bytes, Inter:178948 Bytes, Dram:117360 Bytes
[0m
[... cut ...]
[0;32mI (34382) AUDIO_WRAPPER: esp_auido status:4,err:10002[0m
[0;32mI (34402) AUDIO_WRAPPER: Get play position 0 of 0[0m
[0;32mI (34408) AUDIO_WRAPPER: Get play position 0 of 0[0m
I (34294,tid:3ffbbe4c) lightduer_dcs_router.c( 449): Directive name: RenderHint
[0;32mI (34425) AUDIO_WRAPPER: Get play position 0 of 0[0m
E (34300,tid:3ffbbe4c) lightduer_dcs_router.c( 459): unrecognized directive: RenderHint

[0;31mE (34432) AUDIO_WRAPPER: duer_dcs_audio_on_finished, 2[0m
[0;32mI (34439) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m

diskman88, 2018-10-11:
@jason-mao 

2、用44100 48000 16bit 双通道录音后，发回放时语速变快，有杂音，录一段慢节奏的歌，然后回放，很明感觉到。我用例程pipeline_wav_sdcard测的，但用16000 16bit 单通道正常。

问题2. resmaple 有问题在修复中

这个问题感觉不是resmaple的问题，pipeline_wav_sdcard例程没有resmple，流通道{"i2s", "wav", "file"}，是不是I2S驱动有问题？

diskman88, 2018-10-11:
@kirbyzjj 
死机是因为rsp_filter_open中src_rate和src_ch都变成0, 导致调用resample_open时死机

我看了源代码，src_rate和src_ch都有赋值，不知哪个地方让他们变成0了？另：你的项目做得怎么样了？可以交流一下吗？我的QQ：3119857848

kirbyzjj, 2018-10-12:
> @kirbyzjj
> 死机是因为rsp_filter_open中src_rate和src_ch都变成0, 导致调用resample_open时死机
> 
> 我看了源代码，src_rate和src_ch都有赋值，不知哪个地方让他们变成0了？另：你的项目做得怎么样了？可以交流一下吗？我的QQ：3119857848
是的，所以应该是rec engine哪里导致的问题，rec engine不开源只能等官方修复，临时措施要么录音的时候去掉resample, 要么直接在rsp_filter_open中判断src_rate和src_ch都为0的时候直接设置filter->src_rate = 16000;	filter->src_ch = 1; 这样至少dueros这个demo录音就不会死机了.

另外那个无法播放的问题，你的log里也已经很清楚了
�[0;32mI (34345) ESP_AUDIO_TASK: It's a decoder.URI:http://od.qingting.fm/vod/00/00/0000000000000000000020628362_24.m4a?from=dumi&logid=n9aks-16d23ede8400001_0#1_0
�[0m
�[0;32mI (34371) ESP_AUDIO_TASK: 1.IN:[http],CODEC:[m4a],FILTER:[NULL],OUT:[1_0],RESAMPLE:[48000]�[0m

注意看那个pipeline里 OUT 的tag变成[1_0], 正好是返回的url里#后面的内容，导致pipeline 报1002错误, 猜测是esp audio把这个含有#号的内容当成录音的uri,错误解析了，这个问题需要官方修复，否则播放这块需要自己来实现下，但由于目前问题较多，建议等待官方修复

junbowu, 2018-11-29:
我也碰到这个问题了。。。有好用的老版本么？

jason-mao, 2019-01-03:
> @kirbyzjj
> DCS是通过回调函数void duer_dcs_audio_play_handler(const char _url, const int offset)调用adf的esp_audio_play(player, AUDIO_CODEC_TYPE_DECODER, (const char_)url, offset)播放。
> 我从log里打印出来的信息：
> 0;32mI (34345) ESP_AUDIO_TASK: It's a decoder.URI:http://od.qingting.fm/vod/00/00/0000000000000000000020628362_24.m4a?from=dumi&logid=n9aks-16d23ede8400001_0#1_0
> 提取前一段[http://od.qingting.fm/vod/00/00/0000000000000000000020628362_24.m4a在浏览器里打开，可以播放，在void](http://od.qingting.fm/vod/00/00/0000000000000000000020628362_24.m4a%E5%9C%A8%E6%B5%8F%E8%A7%88%E5%99%A8%E9%87%8C%E6%89%93%E5%BC%80%EF%BC%8C%E5%8F%AF%E4%BB%A5%E6%92%AD%E6%94%BE%EF%BC%8C%E5%9C%A8void) duer_dcs_audio_play_handler()函数里做一下处理，截掉后面的内容，再调用esp_audio_play()，把处理后的uri我传进去，现在放故事可以，昨天放歌的时候也可以，但今天又不行了，好郁闷。但无论放歌还是讲故事，都不能播放完整。
> 
> 估计DCS和ADF都还有问题。

这个问题在release v1.0 里面已修复，支持了`http://od.qingting.fm/vod/00/00/0000000000000000000020628362_24.m4a?from=dumi&logid=n9aks-16d23ede8400001_0#1_0`格式

closed 2019-01-03 by jason-mao