espressif/esp-adf #69: dueros用例 怎么采集两个mic的数据并保存的SD卡中呢？

xiaoseyihe, 2018-10-10:


jason-mao, 2018-10-11:
@xiaoseyihe 　` rec_engine_read`是先用I2S_stream读取数据，然后重采样到16k单通道输出的。你的需求可以使用一个`I2S_stream `->`fatfs_stream`这样的pipeline来实现。

xiaoseyihe, 2018-10-11:
@jason-mao 感谢回复，你的意思是类似record_wav_sdcard_example中的实现吧？在dueros中可以实现吗？目前dueros的pipeline是"i2s"-> "filter"-> "raw"

xiaoseyihe, 2018-10-11:
dueros 能通过修改重采样后的通道数获取两路mic吗？我尝试了一下，采集到的数据信号不对

    i2s_stream_cfg_t i2s_cfg = I2S_STREAM_CFG_DEFAULT();
    i2s_cfg.type = AUDIO_STREAM_READER;
    i2s_stream_reader = i2s_stream_init(&i2s_cfg);
    audio_element_info_t i2s_info = {0};
    audio_element_getinfo(i2s_stream_reader, &i2s_info);
    i2s_info.bits = 16;
    i2s_info.channels = 2;
    i2s_info.sample_rates = 48000;
    audio_element_setinfo(i2s_stream_reader, &i2s_info);

#if 1
    rsp_filter_cfg_t rsp_cfg = DEFAULT_RESAMPLE_FILTER_CONFIG();
    rsp_cfg.src_rate = 48000;
    rsp_cfg.src_ch = 2;
    rsp_cfg.dest_rate = 16000;
    rsp_cfg.dest_ch = 1**_;------------（此处修改为2，采集到的信号不正常啊）_**
    rsp_cfg.type = AUDIO_CODEC_TYPE_ENCODER;
    audio_element_handle_t filter = rsp_filter_init(&rsp_cfg);
#endif

closed 2019-08-08 by jason-mao