espressif/esp-adf #70: Example pipeline_bt_sink broken

errolt, 2018-10-18:
After compiling and uploading the pipeline_bt_sink to the ESP32_LyraT V4.3 board I get an error "the image does not fit in partition"

If I use a new partition with 2MB factory_app size then the board just bootloops.

I (30) boot: ESP-IDF v3.0.1 2nd stage bootloader
I (30) boot: compile time 16:23:47
I (32) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (81) boot: End of partition table
I (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x66298 (418456) map
I (240) esp_image: segment 1: paddr=0x000762c0 vaddr=0x3ffc0000 size=0x02a40 ( 10816) load
I (245) esp_image: segment 2: paddr=0x00078d08 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at C:/msys32/home/Errol/esp/esp-adf/esp-idf/components/freertos/xtensa_vectors.S:1685

I (247) esp_image: segment 3: paddr=0x00079110 vaddr=0x40080400 size=0x06f00 ( 28416) load
I (268) esp_image: segment 4: paddr=0x00080018 vaddr=0x400d0018 size=0x8e290 (582288) map
0x400d0018: _flash_cache_start at ??:?

I (472) esp_image: segment 5: paddr=0x0010e2b0 vaddr=0x40087300 size=0x090e8 ( 37096) load
0x40087300: r_lld_get_mode at ??:?

I (487) esp_image: segment 6: paddr=0x001173a0 vaddr=0x400c0000 size=0x00000 (     0) load
E (488) **esp_image: Image length 1078224 doesn't fit in partition length 1048576**
E (493) boot: Factory app partition is not bootable
E (499) boot: No bootable app partitions in the partition table
user code done

BeancurdPachelbel, 2018-10-26:
set the size of factory app to 2M

1. create file partitions_bluetooth.csv in the root directory 

[image]

2. set the content of partitions_bluetooth.csv as following:

**nvs,      data, nvs,     0x9000,  0x6000,
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 2M,**

3. config the Partition Table
if your environment is esp-adf, 

- **make menuconfig**
- **Partition table**
- choose the following **Custom partition table csv**
- modify the  csv file name as **partitions_bluetooth.csv**
- save 

else depends on your environment.

closed 2018-10-30 by igrr

krzychb, 2018-10-30:
Hi @errolt and @BeancurdPachelbel,
Thank you for reporting this issue and providing a resolution :+1: 
The fix is merged in master now.