espressif/esp-adf #73: Keyword recognition timeframe?

torntrousers, 2018-10-29:
The ESP-ADF features mentions keyword recognition, but as a next release feature. Could you give any   idea of what sort of time that might be in?

Thanks!

jason-mao, 2018-10-29:
@torntrousers 

Thanks for your attention.
The ESP Speech Recognition feature is already under internal review. It will be available in two week at most. BTW, ESP SR can be support single keyword and multi keywords recognition.

Thanks.

torntrousers, 2018-10-29:
Thanks! Very exciting, can't wait.

mgsscorr1, 2018-11-07:
@jason-mao 

is ESP SR going to be an extention of Alexa, Lex, and Google dialog? or its own entity?

Steve

jason-mao, 2018-11-09:
@torntrousers ESP SR libs has been valid on  35e686dc109e19dc799f62a134be33fa1c9177df.

jason-mao, 2018-11-09:
@mgsscorr1 No, ESP SR libs is a local speech recognition on ESP32.

torntrousers, 2018-11-09:
Thanks for letting us know!

From a very quick skim all I found was the dueros example which looks like it uses the "Hi Lexin" keyword, are there any other examples or some doc somewhere on using keyword recognition?

jason-mao, 2018-11-09:
Yes, we will provide some simple examples with LyraTD-MSC and LyraT-V4.3 boards.

mgsscorr1, 2018-11-09:
By local, do you mean completely local or still requires cloud services? do you know when these examples will be posted?

jason-mao, 2018-11-14:
@mgsscorr1 ESP SR can be detect single word and multi words by local. If you say some sentence, need to cloud services to recognition now. ESP SR simple user guide, please refer to https://github.com/espressif/esp-adf-libs/blob/1e04eface358b47d79c650015021f2dd09b3b1b9/esp_sr/include/esp_sr_iface.h#L195

torntrousers, 2018-11-14:
Thanks for that Programming Guide link. Where does the wake word model come from? Can we generate our own for custom wake words?

jason-mao, 2018-11-14:
@torntrousers We have plan to provide some method to customization wake words. Preliminary schedule is valid on March, 2019.

torntrousers, 2018-11-14:
Aww. Thats quite a long time to wait. What are the wake words provided presently?

jason-mao, 2018-11-14:
Please get some information at here https://github.com/espressif/esp-adf-libs/blob/1e04eface358b47d79c650015021f2dd09b3b1b9/esp_sr/READEME.md

torntrousers, 2018-11-14:
I can't even begin to pronounce those! As the method to generate our own customised wake words isn't for some time is there any chance a couple more English wake word models could be provided sooner? For example, "lamp on" and "lights off" would be great for making a good simple demo with ESP32 ADF.

lunarprince, 2018-12-02:
Referring to the esp_sr_iface.h for adding keyword detection in audio stream.
Is this API enabled for use ? It builds and links alright.

int r = model->detect(model_data, buffer);

Is the buffer being passed to detection a MONO stream of 20ms sample @ 16 bit/sample @ 16000 Hz? 

The I2S Clock is configured in example "google_translate_device" to 16k, mono already.
 So can I pass the incoming audio data without resampling to keyword detection?

I am unable to trigger the keyword detection in software by doing so. Do we need resampling as done in esp_avs_sdk (though there the incoming data is sampled at 48k, stereo)

Similar question arises for VAD, in esp_vad.h, 
vad_state = vad_process(vad_inst, buffer);
What is size of buffer to be passed and format? 16k, 30ms, 16bits per sample?.
If I pass audio data received into VAD, VAD is always reported to be detected on every sample.

torntrousers, 2019-03-01:
Hello @jason-mao , any news on the possibility of preliminary support for custom wake words might be available March 2019.

jason-mao, 2019-03-04:
@torntrousers Currently, we have provided commercial custom, if you need that please contact us.

alexwhittemore, 2019-04-02:
> Currently, we have provided commercial custom, if you need that please contact us.

I'm a little confused - doing the work to train custom keywords certainly seems like a commercial service. But is there any documentation (I can't seem to find any) on training custom words? Best I can tell, the models for a handful of wake words and commands are provided as static libs, but I can't find any info on how to train custom words. For instance, if I want my project to respond to "Alexa, do a dance!" - is that possible without commercial support? What about "Jarvis, do a dance!"?

maxgerhardt, 2019-04-28:
@jason-mao I'd also be interested in this.

ludiazv, 2019-04-30:
Also interested on this. A training tool that could generate the models in .a format or other format that can be saved and retrieved in the flash is a must.  But is seems that they are playing the propetary card here in order to promote other income sources.

I think the community should start looking to other alternatives. Just a token. Google tensorflow has launch an experimental branch for mcus. The have published a micro-speech framework to process simple commands. I think porting this to ESPs is feasible as they use c/c++ with gcc tool chain. Parts of ADF could still be used such streams , filters or voice activity detection but the machine learning could be done in tensorflow. Great advantage is that tensorflow learning tool is quite mature an also that you can test your model in your PC as you can run the same code in a PC/mac and the use in the MCU.

maxgerhardt, 2019-04-30:
@ludiazv Reading https://groups.google.com/a/tensorflow.org/forum/#!topic/tflite/OhC25yTgOyE and https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/experimental/micro#how-to-port-tensorflow-lite-micro-to-a-new-platform this might be a solution. Or https://github.com/uTensor/uTensor/issues/137.

lunarprince, 2019-04-30:
> @ludiazv <https://github.com/ludiazv> Reading
> https://groups.google.com/a/tensorflow.org/forum/#!topic/tflite/OhC25yTgOyE
> and
> https://github.com/tensorflow/tensorflow/tree/master/tensorflow/lite/experimental/micro#how-to-port-tensorflow-lite-micro-to-a-new-platform
> this might be a solution. Or uTensor/uTensor#137

I think this is a great idea, we can contribute towards porting. Having a
programmable keyword is the minimum needed feature to deploy any product,
with few commands extra.
Regards,
CK

Mhkgrover, 2020-12-21:
Does ADF voice library support English wake up command?

closed 2022-02-08 by jason-mao