espressif/esp-adf #79: DuerOS: 调用 duer_voice_send 时总是发生内存错误

Chenzejun-Dummkopf, 2018-11-12:
本人使用 ESP32 lyrat的板子多次做实验，发现相同的问题如下
当我使用 duer_voice_send 将我mic采集到的buffer发送给 dueros 时，总是经常性地出现如下错误：

No.1
******* stream_capture_avs : duer_voice_send heap size = [18920] *******（正在发送buffer给dueros）
Guru Meditation Error: Core 0 panic'ed (LoadProhibited)
. Exception was unhandled.
Register dump:
PC : 0x40150c39 PS : 0x00060230 A0 : 0x80146454 A1 : 0x3ffff740
0x40150c39: iir_mem16 at ??:?

A2 : 0x3f85592c A3 : 0x3f8535ec A4 : 0x3f85592c A5 : 0x00000028
A6 : 0x0000000a A7 : 0x00074608 A8 : 0x3f85597c A9 : 0x3ffb05c0
A10 : 0x00000000 A11 : 0x00000000 A12 : 0x3ffb6c34 A13 : 0x0007462c
A14 : 0x00007fff A15 : 0x3f8535fe SAR : 0x00000017 EXCCAUSE: 0x0000001c
EXCVADDR: 0x00074608 LBEG : 0x40146434 LEND : 0x40146440 LCOUNT : 0x00000000
0x40146434: nb_encode at ??:?

0x40146440: nb_encode at ??:?

Backtrace: 0x40150c39:0x3ffff740 0x40146451:0x3ffff770 0x4014b67d:0x3ffff890 0x4014b68d:0x3ffff8b0 0x40149b05:0x3ffff8d0 0x4014b67d:0x3ffff980 0x40141ab6:0x3ffff9a0 0x40141b45:0x3ffff9c0 0x4013a5c3:0x3ffff9f0 0x4013a9c2:0x3ffffa10 0x4012c3d9:0x3ffffa30
0x40150c39: iir_mem16 at ??:?

0x40146451: nb_encode at ??:?

0x4014b67d: speex_encode_int at ??:?

0x4014b68d: speex_encode_native at ??:?

0x40149b05: sb_encode at ??:?

0x4014b67d: speex_encode_int at ??:?

0x40141ab6: duer_speex_encode_frame at lightduer_speex.c:?

0x40141b45: duer_speex_encode at ??:?

0x4013a5c3: duer_voice_send_internal at lightduer_voice.c:?

0x4013a9c2: duer_voice_send at ??:?

0x4012c3d9: stream_capture_avs at /home/......

通过make monitor追踪，发现最后死在iir_mem16() 这个dueros调用的函数上，想请问下有遇到相同问题的同学吗？这是为什么？
求大家救救孩子，这个问题困扰很久了，谢谢大家了

Chenzejun-Dummkopf, 2018-11-14:
求大家 救救孩子，被这个问题困扰很久了，我现在把 录音 和 duer_voice_send 分开处理了，但是还是偶有发生内存crash，请问该如何解救呢？

jason-mao, 2019-05-22:
@Chenzejun-Dummkopf 这个问题如何了？

Chenzejun-Dummkopf, 2019-05-22:
@jason-mao  jason，您好，我是Chenzejun，这个问题是我在实习期间遇到的，直到我离开实习公司也未能很好地解决这个issue，从duerOS反应的log来看，最后是宕在 ii2_mem16 上，估计是和内存页有关的pagefault，至于问题的根源，由于duerOS 闭源也没能继续debug下去，如 jason 您有最新想法，还请不吝赐教哈~ 😄

closed 2019-12-11 by jason-mao