espressif/esp-adf #80: dueros 登陆失败

442617679, 2018-11-12:
W (783) REC_ENG: open rec_eng_2.wav failed,[293]
I (791) I2S: APLL: Req RATE: 48000, real rate: 47999.961, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 12287990.000, SCLK: 1535998.750000, diva: 1, divb: 0
I (794) REC_ENG: ESP SR Engine, chunksize is 480， FRAME_SIZE:960, frequency:16000
I (767) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_RESUME,state:1
E (816) REC_ENG: Recorder Engine Running ..., vad_window=9, wakeup=10000 ms, vad_off=800 ms, threshold=90 ms, sensitivity=0
I (822) RSP_FILTER: rsp_filter_open, encoder, src:48000,2,dest:16000,1,buf_sz:6200, in_pcm:6192
I (834) REC_ENG: state idle
I (1924) wifi: n:11 0, o:1 0, ap:255 255, sta:11 0, prof:1
I (2767) wifi: state: init -> auth (b0)
I (2771) wifi: state: auth -> assoc (0)
I (2776) wifi: state: assoc -> run (10)
I (3027) wifi: connected with XYXYX, channel 11
I (3027) wifi: pm start, type: 1

I (4125) event: sta ip: 192.168.3.4, mask: 255.255.255.0, gw: 192.168.3.1
I (4125) DUEROS: PERIPH_WIFI_CONNECTED [452]
E (4127) DUEROS: Recv Que DUER_CMD_LOGIN
I (4131) DUEROS: duer_start, len:1

E (4034,tid:3ffc9cf4) lightduer_connagent.c( 331):memory alloc fail!
I (4039,tid:3ffc9cf4) lightduer_ds_log_cache.c(  45):cached tracecode: 0x12010400

jason-mao, 2018-11-13:
Could you show me the free memory when system started?

jason-mao, 2018-11-20:
Closed , same with #82

closed 2018-11-20 by jason-mao