espressif/esp-adf #82: dueros无法连接

cc1227858751, 2018-11-20:
rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)

flash read err, 1000

ets_main.c 371 

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)

configsip: 0, SPIWP:0xee

clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00

mode:DIO, clock div:1

load:0x3fff0018,len:4

load:0x3fff001c,len:5964

load:0x40078000,len:7892

load:0x40080000,len:5808

entry 0x40080314

[0;32mI (29) boot: ESP-IDF v3.1.1-6-g2aa9a2118-dirty 2nd stage bootloader[0m

[0;32mI (29) boot: compile time 13:16:41[0m

[0;32mI (31) boot: Enabling RNG early entropy source...[0m

[0;32mI (35) boot: SPI Speed      : 80MHz[0m

[0;32mI (39) boot: SPI Mode       : DIO[0m

[0;32mI (43) boot: SPI Flash Size : 4MB[0m

[0;32mI (47) boot: Partition Table:[0m

[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m

[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m

[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m

[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00200000[0m

[0;32mI (81) boot: End of partition table[0m

[0;32mI (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x8fbe8 (588776) map[0m

[0;32mI (263) esp_image: segment 1: paddr=0x0009fc10 vaddr=0x3ffb0000 size=0x00400 (  1024) load[0m

[0;32mI (263) esp_image: segment 2: paddr=0x000a0018 vaddr=0x400d0018 size=0x1269b4 (1206708) map[0m

[0;32mI (616) esp_image: segment 3: paddr=0x001c69d4 vaddr=0x3ffb0400 size=0x035e8 ( 13800) load[0m

[0;32mI (621) esp_image: segment 4: paddr=0x001c9fc4 vaddr=0x3ffb39e8 size=0x00000 (     0) load[0m

[0;32mI (623) esp_image: segment 5: paddr=0x001c9fcc vaddr=0x40080000 size=0x00400 (  1024) load[0m

[0;32mI (632) esp_image: segment 6: paddr=0x001ca3d4 vaddr=0x40080400 size=0x11c28 ( 72744) load[0m

[0;32mI (665) esp_image: segment 7: paddr=0x001dc004 vaddr=0x400c0000 size=0x00000 (     0) load[0m

[0;32mI (665) esp_image: segment 8: paddr=0x001dc00c vaddr=0x50000000 size=0x00000 (     0) load[0m

[0;32mI (682) boot: Loaded app from partition at offset 0x10000[0m

[... 150 lines cut ...]
I (1699) wifi: n:9 0, o:1 0, ap:255 255, sta:9 0, prof:1

I (2544) wifi: state: init -> auth (b0)

I (2579) wifi: state: auth -> assoc (0)

I (2623) wifi: state: assoc -> run (10)

I (2749) wifi: connected with QJ_YJS, channel 9

I (2749) wifi: pm start, type: 1

[0;32mI (4125) event: sta ip: 192.168.31.211, mask: 255.255.255.0, gw: 192.168.31.1[0m

[0;32mI (4126) DUEROS: PERIPH_WIFI_CONNECTED [452][0m

[0;31mE (4127) DUEROS: Recv Que DUER_CMD_LOGIN[0m

[0;32mI (4132) DUEROS: duer_start, len:1470

{"configures":"{}","bindToken":"1240bb8c6921f4ec546cebbcd17cace0","coapPort":443,"token":"CdAqAuKYUuR2vHrsGNJXzu7Rmf9CYfET","serverAddr":"device.iot.baidu.com","lwm2mPort":443,"uuid":"18ec0000000001","rsaCaCrt":"-----BEGIN CERTIFICATE-----\nMIIDUDCCAjgCCQCmVPUErMYmCjANBgkqhkiG9w0BAQUFADBqMQswCQYDVQQGEwJD\nTjETMBEGA1UECAwKU29tZS1TdGF0ZTEOMAwGA1UECgwFYmFpZHUxGDAWBgNVBAMM\nDyouaW90LmJhaWR1LmNvbTEcMBoGCSqGSIb3DQEJARYNaW90QGJhaWR1LmNvbTAe\nFw0xNjAzMTEwMzMwNDlaFw0yNjAzMDkwMzMwNDlaMGoxCzAJBgNVBAYTAkNOMRMw\nEQYDVQQIDApTb21lLVN0YXRlMQ4wDAYDVQQKDAViYWlkdTEYMBYGA1UEAwwPKi5p\nb3QuYmFpZHUuY29tMRwwGgYJKoZIhvcNAQkBFg1pb3RAYmFpZHUuY29tMIIBIjAN\nBgkqhkiG9w0BAQEFAAOCAQ8AMIIBCgKCAQEAtbhIeiN7pznzuMwsLKQj2xB02+51\nOvCJ5d116ZFLjecp9qtllqOfN7bm+AJa5N2aAHJtsetcTHMitY4dtGmOpw4dlGqx\nluoz50kWJWQjVR+z6DLPnGE4uELOS8vbKHUoYPPQTT80eNVnl9S9h/l7DcjEAJYC\nIYJbf6+K9x+Ti9VRChvWcvgZQHMRym9j1g/7CKGMCIwkC+6ihkGD/XG40r7KRCyH\nbD53KnBjBO9FH4IL3rGlZWKWzMw3zC6RTS2ekfEsgAtYDvROKd4rNs+uDU9xaBLO\ndXTl5uxgudH2VnVzWtj09OUbBtXcQFD2IhmOl20BrckYul+HEIMR0oDibwIDAQAB\nMA0GCSqGSIb3DQEBBQUAA4IBAQCzTTH91jNh/uYBEFekSVNg1h1kPSujlwEDDf/W\npjqPJPqrZvW0w0cmYsYibNDy985JB87MJMfJVESG/v0Y/YbvcnRoi5gAenWXQNL4\nh2hf08A5wEQfLO/EaD1GTH3OIierKYZ6GItGrz4uFKHV5fTMiflABCdu37ALGjrA\nrIjwjxQG6WwLr9468hkKrWNG3dMBHKvmqO8x42sZOFRJMkqBbKzaBd1uW4xY5XwM\nS1QX56tVrgO0A3S+4dEg5uiLVN4YVP/Vqh4SMtYkL7ZZiZAxD9GtNnhRyFsWlC2r\nOVSdXs1ttZxEaEBGUl7tgsBte556BIvufZX+BXGyycVJdBu3\n-----END CERTIFICATE-----\n","macId":"","version":11662}[0m

E (4162,tid:3ffc9cd0) lightduer_connagent.c( 331):memory alloc fail!

I (4169,tid:3ffc9cd0) lightduer_ds_log_cache.c(  45):cached tracecode: 0x12010400

cc1227858751, 2018-11-20:
不止我一个人出现这问题吧，求助啊

jason-mao, 2018-11-20:
@cc1227858751 请参考dueros 下的sdkconfig.defaults.

```c
CONFIG_SPIRAM_BOOT_INIT=y
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_TYPE_ESPPSRAM32=y
CONFIG_SPIRAM_SIZE=4194304
CONFIG_SPIRAM_SPEED_80M=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_CACHE_WORKAROUND=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=16384
```

> E (4162,tid:3ffc9cd0) lightduer_connagent.c( 331):memory alloc fail!

是由于你没有打开PSRAM导致申请内存失败。

cc1227858751, 2018-11-20:
> @cc1227858751 请参考dueros 下的sdkconfig.defaults.
>
> ```c
> CONFIG_SPIRAM_BOOT_INIT=y
> CONFIG_SPIRAM_USE_MALLOC=y
> CONFIG_SPIRAM_TYPE_ESPPSRAM32=y
> CONFIG_SPIRAM_SIZE=4194304
> CONFIG_SPIRAM_SPEED_80M=y [...]

不好意思，如何打开PSRAM呢，在munuconfig中吗？

jason-mao, 2018-11-20:
run command: `make menuconfig`, select item like follow:`Component config`-> `ESP32-specific`->`Support for external, SPI-connected RAM`.

cc1227858751, 2018-11-21:
> run command: `make menuconfig`, select item like follow:`Component config`-> `ESP32-specific`->`Support for external, SPI-connected RAM`.

已解决，多谢！

closed 2019-08-17 by jason-mao