espressif/esp-adf #83: dueros已经连接，语音可以唤醒，但是没有回应

cc1227858751, 2018-11-22:
I (16866,tid:3ffc6c34) lightduer_connagent.c( 212):add resource successfully!!

I (16880,tid:3ffc6c34) lightduer_connagent.c( 212):add resource successfully!!

W (16886,tid:3ffc6c34) lightduer_engine.c( 634):data cache has not sent, pending..., dcache_len:1, qcache_len:4

W (16898,tid:3ffc6c34) lightduer_system_info.c( 306):Undefined memory type, 0

E (16903,tid:3ffc6c34) lightduer_system_info.c( 389):Sys Info: Get disk info failed

W (16961,tid:3ffc6c34) lightduer_events.c(  80):[lightduer_ca] <== event end = 400F6278, timespent = 65

[0;32mI (48166) DUEROS: PERIPH_NOTIFY_KEY_REC[0m

[0;32mI (48167) REC_ENG: Recorder trigger start[0m

[0;32mI (48198) REC_ENG: WAKEUP timer started[0m

[0;32mI (48199) DUEROS: --- rec_engine_cb --- REC_EVENT_WAKEUP_START[0m

[0;32mI (48199) ESP_PERIPH: This peripheral has been added[0m

[0;32mI (49143) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_START[0m

[0;32mI (49143) DUEROS: Recv Que DUER_CMD_START[0m

[0;32mI (49148) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m

I (48848,tid:3ffc3430) lightduer_dcs_local.c( 188):Current dialog id: 18ec000000000140b18ccf0000bece00000004

[0;33mW (51325) REC_ENG: State VAD silence_time >= vad_off_window[0m

[0;32mI (51326) ESP_PERIPH: This peripheral has been added[0m

[0;32mI (51326) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_STOP, state:3[0m

[0;32mI (52124) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_START[0m

[0;32mI (52379) DUEROS: Dueros DUER_CMD_STOP[0m

[0;32mI (52387) DUEROS: Recv Que DUER_CMD_START[0m

E (52085,tid:3ffc6c34) lightduer_voice.c( 169):The qcache is not matched, may be destroyed, item:00000000, stat:3F821754

[0;32mI (52392) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m

I (52096,tid:3ffc3430) lightduer_dcs_local.c( 188):Current dialog id: 18ec0000000001470331290000cb7e00000006

Task watchdog got triggered. The following tasks did not reset the watchdog in time:

 - IDLE (CPU 0)

Tasks currently running:

CPU 0: duerosTask

CPU 1: IDLE

[0;33mW (54270) REC_ENG: State VAD silence_time >= vad_off_window[0m

[0;32mI (54271) ESP_PERIPH: This peripheral has been added[0m

[0;32mI (54271) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_STOP, state:3[0m

[0;33mW (55099) REC_ENG: State VAD silence_time >= vad_off_window[0m

[0;33mW (55899) REC_ENG: State VAD silence_time >= vad_off_window[0m

[0;32mI (56793) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_START[0m

[0;33mW (58782) REC_ENG: State VAD silence_time >= vad_off_window[0m

[0;32mI (58783) ESP_PERIPH: This peripheral has been added[0m

[0;32mI (58783) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_STOP, state:3[0m

Task watchdog got triggered. The following tasks did not reset the watchdog in time:

 - IDLE (CPU 0)

Tasks currently running:

CPU 0: duerosTask

CPU 1: IDLE

[0;33mW (59581) REC_ENG: State VAD silence_time >= vad_off_window[0m

I (16866,tid:3ffc6c34) lightduer_connagent.c( 212):add resource successfully!!

I (16880,tid:3ffc6c34) lightduer_connagent.c( 212):add resource successfully!!

W (16886,tid:3ffc6c34) lightduer_engine.c( 634):data cache has not sent, pending..., dcache_len:1, qcache_len:4

W (16898,tid:3ffc6c34) lightduer_system_info.c( 306):Undefined memory type, 0

E (16903,tid:3ffc6c34) lightduer_system_info.c( 389):Sys Info: Get disk info failed

W (16961,tid:3ffc6c34) lightduer_events.c(  80):[lightduer_ca] <== event end = 400F6278, timespent = 65

[0;32mI (48166) DUEROS: PERIPH_NOTIFY_KEY_REC[0m

[0;32mI (48167) REC_ENG: Recorder trigger start[0m

[0;32mI (48198) REC_ENG: WAKEUP timer started[0m

[0;32mI (48199) DUEROS: --- rec_engine_cb --- REC_EVENT_WAKEUP_START[0m

[0;32mI (48199) ESP_PERIPH: This peripheral has been added[0m

[0;32mI (49143) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_START[0m

[0;32mI (49143) DUEROS: Recv Que DUER_CMD_START[0m

[0;32mI (49148) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m

I (48848,tid:3ffc3430) lightduer_dcs_local.c( 188):Current dialog id: 18ec000000000140b18ccf0000bece00000004

[0;33mW (51325) REC_ENG: State VAD silence_time >= vad_off_window[0m

[0;32mI (51326) ESP_PERIPH: This peripheral has been added[0m

[0;32mI (51326) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_STOP, state:3[0m

[0;32mI (52124) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_START[0m

[0;32mI (52379) DUEROS: Dueros DUER_CMD_STOP[0m

[0;32mI (52387) DUEROS: Recv Que DUER_CMD_START[0m

E (52085,tid:3ffc6c34) lightduer_voice.c( 169):The qcache is not matched, may be destroyed, item:00000000, stat:3F821754

[0;32mI (52392) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m

I (52096,tid:3ffc3430) lightduer_dcs_local.c( 188):Current dialog id: 18ec0000000001470331290000cb7e00000006

Task watchdog got triggered. The following tasks did not reset the watchdog in time:

 - IDLE (CPU 0)

Tasks currently running:

CPU 0: duerosTask

CPU 1: IDLE

[0;33mW (54270) REC_ENG: State VAD silence_time >= vad_off_window[0m

[0;32mI (54271) ESP_PERIPH: This peripheral has been added[0m

[0;32mI (54271) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_STOP, state:3[0m

[0;33mW (55099) REC_ENG: State VAD silence_time >= vad_off_window[0m

[0;33mW (55899) REC_ENG: State VAD silence_time >= vad_off_window[0m

[0;32mI (56793) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_START[0m

[0;33mW (58782) REC_ENG: State VAD silence_time >= vad_off_window[0m

[0;32mI (58783) ESP_PERIPH: This peripheral has been added[0m

[0;32mI (58783) DUEROS: --- rec_engine_cb --- REC_EVENT_VAD_STOP, state:3[0m

Task watchdog got triggered. The following tasks did not reset the watchdog in time:

 - IDLE (CPU 0)

Tasks currently running:

CPU 0: duerosTask

CPU 1: IDLE

[0;33mW (59581) REC_ENG: State VAD silence_time >= vad_off_window[0m

jason-mao, 2018-11-22:
@cc1227858751 你log没看到`event: DUER_EVENT_STARTED`，这个是表示连接成功的log。请提供一份完整的log。

cc1227858751, 2018-11-22:
> @cc1227858751 你log没看到`event: DUER_EVENT_STARTED`，这个是表示连接成功的log。请提供一份完整的log。

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5964
load:0x40078000,len:7892
load:0x40080000,len:5904
entry 0x4008031c
[0;32mI (29) boot: ESP-IDF v3.1.1-6-g2aa9a2118-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 13:33:59[0m
[0;32mI (31) boot: Enabling RNG early entropy source...[0m
[0;32mI (35) boot: SPI Speed      : 80MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00300000[0m
[0;32mI (81) boot: End of partition table[0m
[0;32mI (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x910c0 (594112) map[0m
[0;32mI (264) esp_image: segment 1: paddr=0x000a10e8 vaddr=0x3ffb0000 size=0x040d0 ( 16592) load[0m
[0;32mI (270) esp_image: segment 2: paddr=0x000a51c0 vaddr=0x3ffb40d0 size=0x00000 (     0) load[0m
[0;32mI (271) esp_image: segment 3: paddr=0x000a51c8 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (280) esp_image: segment 4: paddr=0x000a55d0 vaddr=0x40080400 size=0x0aa40 ( 43584) load[0m
[0;32mI (303) esp_image: segment 5: paddr=0x000b0018 vaddr=0x400d0018 size=0x13121c (1249820) map[0m
[0;32mI (663) esp_image: segment 6: paddr=0x001e123c vaddr=0x4008ae40 size=0x0d144 ( 53572) load[0m
[0;32mI (681) esp_image: segment 7: paddr=0x001ee388 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (681) esp_image: segment 8: paddr=0x001ee390 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (701) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (701) boot: Disabling RNG early entropy source...[0m
[0;32mI (702) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (707) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (714) cpu_start: Pro cpu up.[0m
[0;32mI (718) cpu_start: Starting app cpu, entry point is 0x40081504[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1187) spiram: SPI SRAM memory test OK[0m
[0;32mI (1187) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1188) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1194) heap_init: At 3FFBB718 len 000248E8 (146 KiB): DRAM[0m
[0;32mI (1200) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1207) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1213) heap_init: At 40097F84 len 0000807C (32 KiB): IRAM[0m
[0;32mI (1219) cpu_start: Pro cpu start user code[0m
[0;32mI (1224) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (1232) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (247) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (362) DUEROS: ADF version is v1.0-beta1-49-g5d73886-dirty[0m
[0;33mW (363) ESP_PERIPH: Peripherals have been initialized already[0m
[0;32mI (364) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;32mI (373) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;33mW (409) PERIPH_TOUCH: _touch_init[0m
[0;32mI (409) SDCARD: Trying to mount with base path=/sdcard[0m
[0;32mI (456) SDCARD: CID name SD   !
[0m
I (379,tid:3ffc6c34) lightduer_session.c(  44):random = 75933
[0;32mI (685) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;31mE (688) gpio: gpio_install_isr_service(394): GPIO isr service already installed[0m
I (698) wifi: wifi driver task: 3ffbde0c, prio:23, stack:3584, core=0
[... 100 lines cut ...]

[0m
[0;33mW (16833) I2S: I2S driver already installed[0m
[0;33mW (16838) I2S: I2S driver already installed[0m
[0;32mI (16843) STAGEFRIGHTMP3_DECODER: MP3 init[0m
[0;32mI (16850) AUDIO_WRAPPER: Func:setup_player, Line:167, MEM Total:4174464 Bytes, Inter:146392 Bytes, Dram:113536 Bytes

[0m
[0;32mI (16858) AUDIO_WRAPPER: esp_audio instance is:0x3f8165d0[0m
I (16566,tid:3ffc6c34) lightduer_dcs_router.c( 283):namespace: ai.dueros.private.protocol
I (16567,tid:3ffc6c34) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.screen_extended_card
I (16581,tid:3ffc6c34) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.system
I (16586,tid:3ffc6c34) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.voice_input
I (16594,tid:3ffc6c34) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.voice_output
I (16603,tid:3ffc6c34) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.speaker_controller
I (16613,tid:3ffc6c34) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.audio_player
[0;32mI (16928) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m
[... cut ...]
W (16640,tid:3ffc6c34) lightduer_events.c(  80):[lightduer_ca] <== event end = 400F0DF8, timespent = 197
I (16648,tid:3ffc6c34) lightduer_connagent.c( 212):add resource successfully!!
I (16659,tid:3ffc6c34) lightduer_connagent.c( 212):add resource successfully!!
W (16662,tid:3ffc6c34) lightduer_engine.c( 634):data cache has not sent, pending..., dcache_len:1, qcache_len:4
W (16674,tid:3ffc6c34) lightduer_system_info.c( 306):Undefined memory type, 0
E (16679,tid:3ffc6c34) lightduer_system_info.c( 389):Sys Info: Get disk info failed

Chenzejun-Dummkopf, 2018-12-03:
你试着唤醒之后说 “天气” 或者 “播放音乐”，看有没有回应，你可以注意下dueros的两个函数：
duer_dcs_speaker_handler(const char *url)
duer_dcs_audio_play_handler(const char *url , const int offset)
看看有没有获取到 url？

cc1227858751, 2019-01-04:
> @cc1227858751 这个问题怎么样了？有进展帮更新一下状态，感谢。

不好意思，这个问题已经解决，谢谢！

closed 2019-01-04 by cc1227858751

jinshun0225, 2019-03-11:
@cc1227858751    
我的和你有点相似  老是出现
W (16898,tid:3ffc6c34) lightduer_system_info.c( 306):Undefined memory type, 0

E (16903,tid:3ffc6c34) lightduer_system_info.c( 389):Sys Info: Get disk info failed
同意也出现过看门狗溢出过
你是怎么解决的  是什么原因？
我的出现上述情况   按键唤醒识别没问题  ，但是唤醒不了  alexa 唤醒不了  （menuconfig配置用这个唤醒）

TangHuanQiao, 2019-09-11:
> > @cc1227858751 这个问题怎么样了？有进展帮更新一下状态，感谢。
> 
> 不好意思，这个问题已经解决，谢谢！

请问一下可以分享一下是如何解决的吗？我也遇到这个问题，感谢

jason-mao, 2019-09-12:
> > > @cc1227858751 这个问题怎么样了？有进展帮更新一下状态，感谢。
> > 
> > 
> > 不好意思，这个问题已经解决，谢谢！
> 
> 请问一下可以分享一下是如何解决的吗？我也遇到这个问题，感谢

把你的问题log也贴出来一下吧

TangHuanQiao, 2019-09-19:
> > > > @cc1227858751 这个问题怎么样了？有进展帮更新一下状态，感谢。
> > > 
> > > 
> > > 不好意思，这个问题已经解决，谢谢！
> > 
> > 
> > 请问一下可以分享一下是如何解决的吗？我也遇到这个问题，感谢
> 
> 把你的问题log也贴出来一下吧

rst:0x1 (POWERON_RESET),boot:0x1f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6076
load:0x40078000,len:8812
ho 0 tail 12 room 4
load:0x40080400,len:6380
entry 0x4008072c
I (30) boot: ESP-IDF v3.2.2 2nd stage bootloader
I (30) boot: compile time 19:24:33
I (30) boot: Enabling RNG early entropy source...
I (34) boot: SPI Speed      : 80MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00300000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x6eb30 (453424) map
I (223) esp_image: segment 1: paddr=0x0007eb58 vaddr=0x3ffb0000 size=0x014b8 (  5304) load
I (225) esp_image: segment 2: paddr=0x00080018 vaddr=0x400d0018 size=0x127b5c (1211228) map
0x400d0018: _flash_cache_start at ??:?

I (578) esp_image: segment 3: paddr=0x001a7b7c vaddr=0x3ffb14b8 size=0x02704 (  9988) load
I (581) esp_image: segment 4: paddr=0x001aa288 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at I:/Train/ESP/msys32/home/Administrator/esp/esp-idf/components/freertos/xtensa_vectors.S:1779

I (585) esp_image: segment 5: paddr=0x001aa690 vaddr=0x40080400 size=0x1775c ( 96092) load
I (639) boot: Loaded app from partition at offset 0x10000
I (639) boot: Disabling RNG early entropy source...
I (640) spiram: Found 64MBit SPI RAM device
I (645) spiram: SPI RAM mode: flash 80m sram 80m
I (650) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (657) cpu_start: Pro cpu up.
I (661) cpu_start: Starting app cpu, entry point is 0x40081524
0x40081524: call_start_cpu1 at I:/Train/ESP/msys32/home/Administrator/esp/esp-idf/components/esp32/cpu_start.c:246

I (0) cpu_start: App cpu up.
I (1146) spiram: SPI SRAM memory test OK
I (1146) heap_init: Initializing. RAM available for dynamic allocation:
I (1146) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1153) heap_init: At 3FFBB150 len 00024EB0 (147 KiB): DRAM
I (1159) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1165) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1172) heap_init: At 40097B5C len 000084A4 (33 KiB): IRAM
I (1178) cpu_start: Pro cpu start user code
I (1183) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (85) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (86) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (127) DUEROS: ADF version is v2.0-beta1-47-gdf76962-dirty
I (128) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
I (132) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3
W (167) PERIPH_TOUCH: _touch_init
I (167) SDCARD: Trying to mount with base path=/sdcard
I (215) SDCARD: CID name NCard!
[... 102 lines cut ...]
I (3938,tid:3ffbc76c) lightduer_connagent.c( 198):will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (3946,tid:3ffbc76c) lightduer_connagent.c( 198):will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (3954,tid:3ffbc76c) lightduer_connagent.c( 198):will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (3964,tid:3ffbc76c) lightduer_connagent.c( 198):will start latter(DUER_ERR_TRANS_WOULD_BLOCK)
I (4238,tid:3ffbc76c) lightduer_connagent.c( 177):connect started!
I (4240,tid:3ffbc76c) lightduer_voice.c( 697):Mutex initializing
I (4240,tid:3ffbc76c) lightduer_ds_log_cache.c(  67):no cache report
E (4331) DUEROS: event: 0
I (4334) DUEROS: duer_dcs_init
I (4254,tid:3ffbc76c) lightduer_dcs_router.c( 283):namespace: ai.dueros.private.protocol
I (4262,tid:3ffbc76c) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.screen_extended_card
I (4271,tid:3ffbc76c) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.system
I (4279,tid:3ffbc76c) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.voice_input
I (4287,tid:3ffbc76c) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.voice_output
I (4296,tid:3ffbc76c) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.speaker_controller
I (4306,tid:3ffbc76c) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.audio_player
I (4402) AUDIO_WRAPPER: duer_dcs_get_speaker_state
E (4410) DUEROS: event: DUER_EVENT_STARTED
I (4410) DUEROS: Dueros DUER_CMD_CONNECTED, duer_state:2
W (4325,tid:3ffbc76c) lightduer_events.c(  80):[lightduer_ca] <== event end = 400F9D48, timespent = 91
I (4340,tid:3ffbc76c) lightduer_connagent.c( 212):add resource successfully!!
I (4348,tid:3ffbc76c) lightduer_connagent.c( 212):add resource successfully!!
W (4417) DUEROS: duer_callback: type:0, source:0x3f806b8c data:3, data_len:0
W (4354,tid:3ffbc76c) lightduer_engine.c( 634):data cache has not sent, pending..., dcache_len:1, qcache_len:4
W (4375,tid:3ffbc76c) lightduer_system_info.c( 306):Undefined memory type, 0

jason-mao, 2019-09-20:
@cc1227858751 那句log不影响，最多算个warning信息。你的log看起来都是正常的。你能通过`hi，乐鑫`唤醒吗？

cc1227858751, 2019-09-20:
> @cc1227858751 那句log不影响，最多算个warning信息。你的log看起来都是正常的。你能通过`hi，乐鑫`唤醒吗？

hi 乐鑫可以唤醒 alxa不可以

TangHuanQiao, 2019-09-20:
> > @cc1227858751 那句log不影响，最多算个warning信息。你的log看起来都是正常的。你能通过`hi，乐鑫`唤醒吗？
> 
> hi 乐鑫可以唤醒 alxa不可以

可以用语音hi 乐鑫唤醒，但是说播放音乐 新闻 天气 任何反应都没有，唤醒后也没有声音应答。灯亮一会儿就休眠了，可以继续唤醒。 和这个贴子楼主的问题一样：dueros已经连接，语音可以唤醒，但是没有回应。看楼上他说解决了，但是没有在这里分享解决办法。郁闷中

jason-mao, 2019-09-20:
@cc1227858751  你登陆你的百度开发者账号，在【设备控制台】找到你的设备，然后进入编译，在【运行日志】里面是否可以找到你这次的语音交互内容呢？

jason-mao, 2019-09-20:
@cc1227858751 另外把你唤醒然后询问天气这样的 串口打印发出来一下

TangHuanQiao, 2019-09-21:
> @cc1227858751 另外把你唤醒然后询问天气这样的 串口打印发出来一下

解决了， 解决方法是，重新copy一份例程， 重新编译烧录。目前只有小度音乐没有授权，新闻天气故事可以试用