espressif/esp-adf #85: Remerge esp_http_client changes

maxgerhardt, 2018-11-23:
The code for the `esp_http_client` component is duplicated from the ESP-IDF component but yet was changed locally here without it being merged back.

I'm talking specifically about the added function `esp_http_client_set_redirection` which exists here [here](https://github.com/espressif/esp-adf/blob/master/components/esp_http_client/include/esp_http_client.h#L337-L360) but not in [ESP-IDF](https://github.com/espressif/esp-idf/blob/master/components/esp_http_client/include/esp_http_client.h#L364-L374). 

Why have diverging library code for this component? It makes building harder when using other build systems (e.g. PlatformIO) because I have to un-include the ESP-IDF lib and include this version of the library. 

Why not merge https://github.com/espressif/esp-adf/commit/a0cba1164a00116b276a712eeebc8fecbbbf7726 into ESP-IDF?

tuanpmt, 2018-11-24:
Hi @maxgerhardt 

Because ESP-ADF is using ESP-IDF v3.1, which is not have esp_http_client component. And some specific features for ADF, if it's necessary we will send the MR to IDF soon

Thanks

closed 2019-01-09 by jason-mao

maxgerhardt, 2019-04-28:
@tuanpmt So I see that ESP-IDF (https://github.com/espressif/esp-idf/tree/release/v3.2/components/esp_http_client) is now way ahead with their changes on esp_http_client, but here, esp_http_client still exists. Will the esp_http_client be deleted from here because all necessary changes are in main ESP-IDF?

vikramdattu, 2019-04-28:
@maxgerhardt
Yes. This will be done once we update to IDF release/v3.2.