espressif/esp-adf #89: SSL fail after 3.1 merge

robertoimai, 2018-12-03:
Hi,

I'm unable to connect to using ssl anymore after I updated ADF and IDF 3.1 merge.
It was fine before.
This is what I get:
E (15355) aws_iot: failed! mbedtls_ssl_handshake returned -0x6800
E (15355) WIFI: Error(-4) connecting to ******-ats.iot.us-east-2.amazonaws.com:8883
I remember I had to make a small change on the prior IDF to make ssl connections, but I was in a rush yesterday and discarded those changes when I merged the latest master branch.
Can this be fixed or I should dig up and apply my small patch again on the 3.1 IDF?

robertoimai, 2018-12-04:
Wrong report.
Sorry.

robertoimai, 2018-12-06:
My apologies.
I am still having this problem. I thought I had it fixed, but it was when I reverted back to my previous IDF release.
This is what I get:

I (14454) WIFI: RAM left 4182760
I (14454) WIFI: AWS IoT SDK Version 2.2.1-
I (14484) WIFI: certificate.pem.crt size: 1220
I (14494) WIFI: private.pem.key size: 1679
I (14494) WIFI: Connecting to AWS...
W (14534) wifi: alloc eb len=36 type=3 fail, heap:4115072

W (14534) wifi: mem fail
W (14564) wifi: alloc eb len=24 type=3 fail, heap:4115932

W (14564) wifi: m f null

W (14614) wifi: alloc eb len=24 type=3 fail, heap:4118008

W (14614) wifi: m f null

E (21054) aws_iot: failed! mbedtls_net_connect returned -0x52
E (21054) WIFI: Error(-23) connecting to a1f3ynjyzalf3r-ats.iot.us-east-2.amazonaws.com:8883
E (29054) aws_iot: failed! mbedtls_net_connect returned -0x52
E (29054) WIFI: Error(-23) connecting to a1f3ynjyzalf3r-ats.iot.us-east-2.amazonaws.com:8883
E (37054) aws_iot: failed! mbedtls_net_connect returned -0x52
E (37054) WIFI: Error(-23) connecting to a1f3ynjyzalf3r-ats.iot.us-east-2.amazonaws.com:8883
E (45054) aws_iot: failed! mbedtls_net_connect returned -0x52
E (45054) WIFI: Error(-23) connecting to a1f3ynjyzalf3r-ats.iot.us-east-2.amazonaws.com:8883

jason-mao, 2019-01-03:
@robertoimai which IDF you used? and did you test the release v1.0 version?

robertoimai, 2019-01-04:
I'm using ESP-IDF v3.1.1-6-g2aa9a2118 2nd stage bootloader on ADF 1.0 and I'm still having problems connecting to anything that uses tls.
Both HTTP client and AWS IoT fail to connect.
This is the log:
I (14941) WIFI: AWS IoT SDK Version 2.2.1-
I (14951) WIFI: Connecting to googleapis.com server....
I (14951) WIFI: certificate.pem.crt size: 1220
I (14961) WIFI: private.pem.key size: 1675
I (14961) WIFI: Connecting to AWS...
W (15061) wifi: alloc eb len=36 type=3 fail, heap:4127908

W (15061) wifi: mem fail
W (15271) wifi: alloc eb len=24 type=3 fail, heap:4074276

W (15281) wifi: m f null

W (15421) wifi: alloc eb len=24 type=3 fail, heap:4071520

W (15421) wifi: m f null

E (21981) esp-tls: mbedtls_ssl_handshake returned -0x4c
I (21981) esp-tls: Certificate verified.
E (21991) esp-tls: Failed to open new connection
E (21991) TRANS_SSL: Failed to open a new connection
E (21991) HTTP_CLIENT: Connection failed, sock < 0
E (21991) WIFI: HTTP POST request failed: 7002
I (22001) WIFI: HTTP_EVENT_DISCONNECTED
I (22001) WIFI: Geolocation data:
I (22011) WIFI: Geolocation task stopping...
E (23651) aws_iot: failed! mbedtls_ssl_handshake returned -0x6800
E (23661) WIFI: Error(-4) connecting to a1f3ynjyzalf3r-ats.iot.us-west-2.amazonaws.com:8883

krzychb, 2019-01-05:
I think the reason of this issue may be as discussed in https://github.com/espressif/esp-idf/issues/2184#issuecomment-408200660

robertoimai, 2019-01-05:
Ok, so if it is memory issue, can you help me free up some memory? What kind of memory is that I should be freeing up?
How can I check how much I have left?
As you can see in the log above, I have 4071520 bytes free on external RAM.
I'm using a 4MB PSRAM ESP32 WROVER module.
Also, why is that 3.0.1 doesn't give me those errors?
It only presents itself when I updated to 3.1?
I was forced to downgrade my IDF and ADF to be able to have a stable system.

krzychb, 2019-01-05:
Wow, a lot of good questions :)

I think that troubleshooting of Wi-Fi connection problems due to memory limitation is outside of the scope of ESP-ADF repository (use esp32.com forum instead), but since I started I here are my thoughts:

> What kind of memory is that I should be freeing up?

Internal memory normally used by the Wi-Fi driver, see [Wi-Fi Buffer Usage](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/wifi.html#wi-fi-buffer-usage).

> How can I check how much I have left?

Use [heap_caps_get_free_size()](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/system/mem_alloc.html#_CPPv223heap_caps_get_free_size8uint32_t)

> As you can see in the log above, I have 4071520 bytes free on external RAM.
I'm using a 4MB PSRAM ESP32 WROVER module.

To use this memory instead, if not done already, try https://github.com/espressif/esp-idf/issues/2184#issuecomment-409107469

Another option is trying https://github.com/espressif/esp-idf/issues/2184#issuecomment-408209411 (i.e. decreasing CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL in menuconfig to 256)

> Also, why is that 3.0.1 doesn't give me those errors?
It only presents itself when I updated to 3.1?
I was forced to downgrade my IDF and ADF to be able to have a stable system.

Each new release brings new features. This usually on the expense of extra memory used.

If "Try to allocate memories of WiFi and LWIP in SPIRAM firstly", etc. does not help, then if possible, try establishing only a single "HTTP client" or "AWS IoT" connection at a time and check when memory issue is reported. This is to verify if the problem is caused by one particular connection or rather because two of them are present at the same time.

robertoimai, 2019-01-05:
Agree.
I'll direct IDF questions there.

closed 2019-01-05 by robertoimai