espressif/esp-adf #90: How to gracefully stop and start pipeline?

robertoimai, 2018-12-05:
I'm getting AEL_IO_ABORT when I try to stop and start the pipeline mid through playback.
If I let the pipeline finish playing the file by itself, everything works beautifully, but if I want to interrupt a playback to start a new one, I keep getting errors.
How do I gracefully stop and start playback?

jason-mao, 2019-01-03:
@robertoimai Pipeline response stop command mechanism  is set connected ring buffer abort flag. So you can get the `AEL_IO_ABORT ` log.

robertoimai, 2019-01-04:
Ok, I see that the AEL_IO_ABORT is not the issue here.
The problem is that I keep getting this when I run the pipeline after I terminate it.
I (4151) AUDIO_ELEMENT: [mp3] RESUME: Element has finished, state:5
My code is like this:
            ESP_LOGI(TAG, "[ * ] New playback");
            ESP_LOGI(TAG, "[ * ] Playing [%d]: %s",msg.cmd,mp3_file_array[msg.cmd]);
            audio_pipeline_terminate(pipeline);
            fclose(sound_file);
            sound_file = fopen(mp3_file_array[msg.cmd], "r");
            audio_pipeline_run(pipeline);
And this is the log I get:
I (4031) SOUND: [ * ] New playback
I (4031) SOUND: [ * ] Playing [3]: /spiffs/enter_pwd.mp3
W (4071) AUDIO_ELEMENT: OUT-[filter] AEL_IO_ABORT
I (4071) RSP_FILTER: rsp_filter_close
W (4071) AUDIO_ELEMENT: OUT-[mp3] AEL_IO_ABORT
W (4071) STAGEFRIGHTMP3_DECODER: output aborted -3
I (4081) STAGEFRIGHTMP3_DECODER: Closed
I (4131) AUDIO_ELEMENT: [mp3] Element task created
I (4131) AUDIO_ELEMENT: [i2s] Element task created
I (4131) AUDIO_ELEMENT: [filter] Element task created
I (4141) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:278, MEM Total:4284836 Bytes, Inter:121008 Bytes, Dram:111704 Bytes

I (4151) AUDIO_ELEMENT: [mp3] RESUME: Element has finished, state:5
I (4151) AUDIO_PIPELINE: Pipeline started
E (4161) SOUND: [ * ] Event interface error : -1
E (4161) SOUND: [ * ] Event interface error : -1
E (4171) SOUND: [ * ] Event interface error : -1
E (4171) SOUND: [ * ] Event interface error : -1
E (4181) SOUND: [ * ] Event interface error : -1
E (4191) SOUND: [ * ] Event interface error : -1
I (4191) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (4201) I2S_STREAM: AUDIO_STREAM_WRITER
I (4201) AUDIO_ELEMENT: [filter] AEL_MSG_CMD_RESUME,state:1
I (4211) RSP_FILTER: sample rate of source data : 44100, channel ofsource data : 1, sample rate of destination data : 48000, channel of destination data : 1
W (4221) AUDIO_ELEMENT: IN-[filter] AEL_IO_ABORT
I (4231) RSP_FILTER: rsp_filter_close
I (4241) AUDIO_ELEMENT: IN-[i2s] AEL_IO_DONE,-2
I (4321) SOUND: [ * ] Finished, stopping pipeline

What would cause mp3 audio element to be in state 5?
It should be 1 like the other elements, right?

robertoimai, 2019-01-05:
Ok, I tried resetting the state to 1 and although it does now show state 1, it is still not playing the file.
So, I just used a workaround where I don't stop the pipeline at all if I need to play another file and pipeline is already running.
I simply close and open a new file with the pipeline still in the running state. That seemed to have worked for now.
Is there any drawback of doing this?

hallgrim, 2019-01-07:
I have the same issue. Sometimes when I stop the playback and change to another file the mp3 decoder is left in a finish state, and I get no playback.

My workaround was to call ```audio_element_reset_state```:
```
  audio_pipeline_stop(pipeline);
  audio_pipeline_wait_for_stop(pipeline);
  audio_element_set_uri(fatfs_stream_reader, uri);
  audio_element_reset_state(mp3_decoder); // <--
  audio_pipeline_run(pipeline);
```

robertoimai, 2019-01-15:
After several tests, I found out that the drawback of doing this is that when I do this right at the end of the first playback, the second one thinks that it is the end too and doesn't play.
I still need a reliable solution to start a playback and move to a different playback without issues.

HanYangZhao, 2019-02-13:
any update on this issue?

jason-mao, 2019-05-22:
@robertoimai @hallgrim @HanYangZhao 
Hi guys
Could you help to test with the latest version 93963a35c0fd57b40bacf229c90895ec715554ab?

jason-mao, 2019-08-17:
Closed due to no more information.

closed 2019-08-17 by jason-mao

hallgrim, 2019-08-27:
I can verify that this problem is fixed. I can now start and stop a pipeline without having to manually reset the mp3 decoder. Thank you @jason-mao and team.