espressif/esp-adf #92: 更新后dueros例程有的时候可以完整播放儿歌或者笑话，但是播放速率不正常，比正常音频慢，不知道你们测试的时候有没有。

cc1227858751, 2018-12-17:
更新后dueros例程有的时候可以完整播放儿歌或者笑话，但是播放速率不正常，比正常音频慢，不知道你们测试的时候有没有。

jason-mao, 2018-12-17:
@cc1227858751 你更新有使用如下步骤吗，“git pull, git submodule update”.都有操作的话，帮提供一下log。

cc1227858751, 2018-12-17:
> @cc1227858751 你更新有使用如下步骤吗，“git pull, git submodule update”.都有操作的话，帮提供一下log。

原有adf丢失，直接clone的最新版，log没有保存，下午测试完再提供，谢谢您及时回复。

cc1227858751, 2018-12-17:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5960
load:0x40078000,len:7892
load:0x40080000,len:5904
entry 0x4008031c
[0;32mI (29) boot: ESP-IDF v3.1.1-6-g2aa9a2118 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 15:26:45[0m
[0;32mI (33) boot: Enabling RNG early entropy source...[0m
[0;32mI (35) boot: SPI Speed      : 40MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00300000[0m
[0;32mI (80) boot: End of partition table[0m
[0;32mI (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x916c0 (595648) map[0m
[0;32mI (300) esp_image: segment 1: paddr=0x000a16e8 vaddr=0x3ffb0000 size=0x040d0 ( 16592) load[0m
[0;32mI (306) esp_image: segment 2: paddr=0x000a57c0 vaddr=0x3ffb40d0 size=0x00000 (     0) load[0m
[0;32mI (307) esp_image: segment 3: paddr=0x000a57c8 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (316) esp_image: segment 4: paddr=0x000a5bd0 vaddr=0x40080400 size=0x0a440 ( 42048) load[0m
[0;32mI (341) esp_image: segment 5: paddr=0x000b0018 vaddr=0x400d0018 size=0x1331c0 (1257920) map[0m
[0;32mI (777) esp_image: segment 6: paddr=0x001e31e0 vaddr=0x4008a840 size=0x0d750 ( 55120) load[0m
[0;32mI (799) esp_image: segment 7: paddr=0x001f0938 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (799) esp_image: segment 8: paddr=0x001f0940 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (820) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (820) boot: Disabling RNG early entropy source...[0m
[0;32mI (820) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (825) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (832) cpu_start: Pro cpu up.[0m
[0;32mI (836) cpu_start: Starting app cpu, entry point is 0x40081508[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1305) spiram: SPI SRAM memory test OK[0m
[0;32mI (1306) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1306) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1312) heap_init: At 3FFBB710 len 000248F0 (146 KiB): DRAM[0m
[0;32mI (1318) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1325) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1331) heap_init: At 40097F90 len 00008070 (32 KiB): IRAM[0m
[0;32mI (1338) cpu_start: Pro cpu start user code[0m
[0;32mI (1342) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (1350) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (30) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (82) DUEROS: ADF version is v1.0-beta1-74-gb58ecad-dirty[0m
[0;33mW (82) ESP_PERIPH: Peripherals have been initialized already[0m
[0;32mI (84) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;32mI (93) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;33mW (128) PERIPH_TOUCH: _touch_init[0m
[0;32mI (128) SDCARD: Trying to mount with base path=/sdcard[0m
[0;32mI (176) SDCARD: CID name SD   !
[0m
I (347,tid:3ffc04dc) lightduer_session.c(  44):random = 75933
[0;32mI (404) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;31mE (407) gpio: gpio_install_isr_service(394): GPIO isr service already installed[0m
I (417) wifi: wifi driver task: 3ffc3280, prio:23, stack:3584, core=0
I (421) wifi: wifi firmware version: b2c9a19
I (425) wifi: config NVS flash: enabled
[... 249 lines cut ...]
[0;32mI (29409) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (29415) I2S_STREAM: AUDIO_STREAM_WRITER[0m
[0;32mI (29784) AAC_DECODER: new aduio song[0m
[0;32mI (34205) AAC_DECODER: I2S setup: sample_rate = 44100 : channels = 2[0m
[0;32mI (34206) AAC_DECODER: M4A sampleRate = 44100, Channels = 2, frameSize = 2048[0m
[0;32mI (34207) RSP_FILTER: reset sample rate of source data : 44100, reset channel of source data : 2[0m
[0;32mI (34219) ESP_AUDIO_CTRL: Request_CMD_Queue CMD:1, Available:5[0m
[0;32mI (34225) ESP_AUDIO_TASK: Send MEDIA_CTRL_EVT_PLAY, line:521[0m
[0;32mI (34232) ESP_AUDIO_TASK: MEDIA_CTRL_EVT_PLAY, status:0[0m
[0;32mI (34240) AUDIO_ELEMENT: [resample] Element task created[0m
[0;32mI (34245) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:278, MEM Total:3949240 Bytes, Inter:115504 Bytes, Dram:82660 Bytes

[0m
[0;32mI (34257) AUDIO_ELEMENT: [resample] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (34263) RSP_FILTER: sample rate of source data : 44100, channel of source data : 2, sample rate of destination data : 48000, channel of destination data : 2[0m
[0;32mI (34258) AUDIO_PIPELINE: Pipeline started[0m
[... cut ...]
I (34245,tid:3ffc04dc) lightduer_dcs_router.c( 407):Directive name: RenderHint
E (34249,tid:3ffc04dc) lightduer_dcs_router.c( 433):unrecognized directive: RenderHint, namespace: ai.dueros.device_interface.screen

[0;32mI (34319) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m
[0;32mI (34325) AUDIO_WRAPPER: Get play position 213339 of 0[0m
I (34293,tid:3ffc04dc) lightduer_dcs_router.c( 407):Directive name: DialogueFinished
[0;32mI (34390) AUDIO_WRAPPER: esp_auido status:1,err:0[0m
[0;32mI (64290) AUDIO_WRAPPER: Get play position 242546 of 0[0m

cc1227858751, 2018-12-17:
> @cc1227858751 你更新有使用如下步骤吗，“git pull, git submodule update”.都有操作的话，帮提供一下log。

这是播放速率不正常的log，确认是最新版的adf

Chenzejun-Dummkopf, 2018-12-17:
适配 LyraTD 的 sdk，默认接受环境变量 IDF_PATH，而不是 ADF_PATH ，但是官网的教程却是设置 ADF_PATH。
另外，git clone了最新的adf下来之后，出现了dueros无法登陆的情况(board ： LyraTD)，请教大家

cc1227858751, 2018-12-17:
> 适配 LyraTD 的 sdk，默认接受环境变量 IDF_PATH，而不是 ADF_PATH ，但是官网的教程却是设置 ADF_PATH。
> 另外，git clone了最新的adf下来之后，出现了dueros无法登陆的情况(board ： LyraTD)，请教大家
ADF下的例程用的应该是ADF中自带的IDF，无法登陆的log贴出来大家看看么。

Chenzejun-Dummkopf, 2018-12-17:
I (1473) DUEROS: PERIPH_NOTIFY_KEY_REC
I (1295,tid:3ffc6994) lightduer_session.c(  44):random = 75933
E (1620) ES8388_DRIVER: es_write_reg error
E (1621) ES8388_DRIVER: es_write_reg error
E (1624) ES8388_DRIVER: es_write_reg error
E (1629) ES8388_DRIVER: es_write_reg error
E (1635) ES8388_DRIVER: es_write_reg error
E (1638) ES8388_DRIVER: es_write_reg error
E (1643) ES8388_DRIVER: es_write_reg error
E (1648) ES8388_DRIVER: es_write_reg error
E (1653) ES8388_DRIVER: es_write_reg error
E (1658) ES8388_DRIVER: es_write_reg error
E (1662) ES8388_DRIVER: es_write_reg error
E (1667) ES8388_DRIVER: es_write_reg error
E (1672) ES8388_DRIVER: es_write_reg error
E (1677) ES8388_DRIVER: es_write_reg error
E (1682) ES8388_DRIVER: es_write_reg error
E (1686) ES8388_DRIVER: es_write_reg error
E (1691) ES8388_DRIVER: es_write_reg error
E (1696) ES8388_DRIVER: es_write_reg error
I (1697) wifi: wifi driver task: 3ffbdd5c, prio:23, stack:3584, core=0
E (1701) ES8388_DRIVER: es_write_reg error
I (1708) wifi: wifi firmware version: b2c9a19
E (1712) ES8388_DRIVER: es_write_reg error
I (1716) wifi: config NVS flash: enabled
E (1721) ES8388_DRIVER: es_write_reg error
I (1725) wifi: config nano formating: disabled
E (1729) ES8388_DRIVER: es_write_reg error
I (1734) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (1738) ES8388_DRIVER: es_write_reg error
I (1748) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE
E (1753) ES8388_DRIVER: es_write_reg error
E (1767) ES8388_DRIVER: es_write_reg error
I (1771) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
I (1781) ES8388_DRIVER: init,out:02, in:00
I (1786) wifi: Init dynamic tx buffer num: 32
I (1789) wifi: Init data frame dynamic rx buffer num: 32
E (1786) ES8388_DRIVER: es_read_reg error
I (1795) wifi: Init management frame dynamic rx buffer num: 32
E (1800) ES8388_DRIVER: es_write_reg error
I (1806) wifi: Init static tx buffer num: 16
E (1810) ES8388_DRIVER: es_read_reg error
I (1814) wifi: Init static rx buffer size: 1600
E (1818) ES8388_DRIVER: es_write_reg error
I (1823) wifi: Init static rx buffer num: 10
E (1828) ES8388_DRIVER: es_read_reg error
I (1832) wifi: Init dynamic rx buffer num: 32
E (1836) ES8388_DRIVER: es_write_reg error
E (1845) ES8388_DRIVER: es_read_reg error
W (1846) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
E (1850) ES8388_DRIVER: es_write_reg error
E (1864) ES8388_DRIVER: es_write_reg error
E (1869) ES8388_DRIVER: es_write_reg error
E (1873) ES8388_DRIVER: es_write_reg error
E (1878) ES8388_DRIVER: es_write_reg error
I (1883) AUDIO_HAL: Codec mode is 1, Ctrl:1
E (1888) ES8388_DRIVER: es_read_reg error
E (1892) ES8388_DRIVER: es_write_reg error
E (1897) ES8388_DRIVER: es_read_reg error
E (1902) ES8388_DRIVER: es_write_reg error
I (1907) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (1914) I2S: DMA Malloc info, datalen=blocksize=1200, dma_buf_count=3
I (1923) I2S: APLL: Req RATE: 44100, real rate: 43945.238, BITS: 16, CLKM: 1, BCK_M: 8, MCLK: 11249981.000, SCLK: 1406247.625000, diva: 1, divb: 0
I (1936) AUDIO_PIPELINE: audio_pipeline_link:0x3f807e10, i2s, 0x3f808098
I (1942) AUDIO_PIPELINE: audio_pipeline_link:0x3f807ecc, filter, 0x3f80a0e8
I (1951) AUDIO_ELEMENT: [i2s] Element task created
I (2050) phy: phy_version: 4000, b6198fa, Sep  3 2018, 15:11:06, 0, 2
I (2051) AUDIO_ELEMENT: [filter] Element task created
I (2055) AUDIO_ELEMENT: [raw] Element task created
I (2058) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:278, MEM Total:4299464 Bytes, Inter:191024 Bytes, Dram:158136 Bytes
[... 28 lines cut ...]
I (9120,tid:3ffc6994) lightduer_connagent.c( 160):=====> event: 1[DUER_EVT_START], status: -18
E (9127,tid:3ffc6994) lightduer_connagent.c( 207):start fail! status:-18
E (9134,tid:3ffc6994) lightduer_connagent.c( 286):Action failed: event: 1, status: -18
W (9141,tid:3ffc6994) lightduer_engine.c( 812):duer_engine_stop has stopped
I (9148,tid:3ffc6994) lightduer_connagent.c( 160):=====> event: 5[DUER_EVT_STOP], status: -18
I (9156,tid:3ffc6994) lightduer_connagent.c( 257):connect stopped, status:-18!!
E (9489) DUEROS: event: 1
E (9491) DUEROS: event: DUER_EVENT_STOPPED
I (9497) DUEROS: Dueros DUER_CMD_QUIT reconnect, retry_num:2
W (9173,tid:3ffc6994) lightduer_events.c(  80):[lightduer_ca] <== event end = 400F0C98, timespent = 78
E (11504) DUEROS: Func:retry_login_timer_cb
E (11504) DUEROS: Recv Que DUER_CMD_LOGIN
I (11504) DUEROS: duer_start, len:1

I (11182,tid:3ffc6994) lightduer_engine.c( 220):duer_engine_start, g_handler:3F807614, length:1, profile:3F8076CC
E (11192,tid:3ffc6994) lightduer_ca.c( 645):    create the configuation failed...
I (11199,tid:3ffc6994) lightduer_ds_log_cache.c(  45):cached tracecode: 0x12010301
I (11207,tid:3ffc6994) lightduer_connagent.c( 160):=====> event: 1[DUER_EVT_START], status: -18
E (11215,tid:3ffc6994) lightduer_connagent.c( 207):start fail! status:-18
E (11221,tid:3ffc6994) lightduer_connagent.c( 286):Action failed: event: 1, status: -18
W (11229,tid:3ffc6994) lightduer_engine.c( 812):duer_engine_stop has stopped
I (11237,tid:3ffc6994) lightduer_connagent.c( 160):=====> event: 5[DUER_EVT_STOP], status: -18
I (11244,tid:3ffc6994) lightduer_connagent.c( 257):connect stopped, status:-18!!
E (11577) DUEROS: event: 1
E (11580) DUEROS: event: DUER_EVENT_STOPPED

Chenzejun-Dummkopf, 2018-12-17:
我用的板子： LyraTD MSC v2.2
image：最新ADF中example中的dueros例程
问题：duer无法登陆

Chenzejun-Dummkopf, 2018-12-17:
好的好的，确实是这样的，多谢，懂了 👍 🥇

Chenzejun-Dummkopf, 2018-12-17:
哦对了，我发现这个圆板的唤醒速度和识别速度很快，url播放也很快，相比LyraT，效率极高，请问这是为什么，为什么圆板唤醒识别播放，一气呵成，效率惊人？

cc1227858751, 2018-12-18:
> 哦对了，我发现这个圆板的唤醒速度和识别速度很快，url播放也很快，相比LyraT，效率极高，请问这是为什么，为什么圆板唤醒识别播放，一气呵成，效率惊人？

你圆板里烧的是哪个程序

Chenzejun-Dummkopf, 2018-12-20:
我原版是烧的 TDesp-audio-sdk-release-ver0.93.tar 这个文件里面编译生成的代码，目前在更换其配套的 DSP代码，使之适应我自己的DSP平台

closed 2019-05-22 by jason-mao