espressif/esp-adf #94: Feature request: UDP stream

valioiv, 2018-12-18:
Hi there! Would be very useful to add support for bidirectional UDP stream similar to your current HTTP stream. Do you guys plan to implement that?

jason-mao, 2018-12-24:
@valioiv We have no plan to do that currently.

matiaspl, 2019-01-03:
Not my question but what I'm looking for is a low latency bidirectional audio intercom. Since most VoIP protocols utilize UDP a simple raw UDP data stream (or even better RTP) would be enough.

valioiv, 2021-03-17:
> Not my question but what I'm looking for is a low latency bidirectional audio intercom. Since most VoIP protocols utilize UDP a simple raw UDP data stream (or even better RTP) would be enough.

This is what I'm also looking for - intercom usage! RTP stream would be great!
@jason-mao , any plans to implement that?

denmuellner, 2021-05-26:
+1 here! (RTP on top of) UDP is used by the vast majority of media transfer protocols. Having a network enabled device and an audio framework like ESP-ADF basically screams for UDP/RTP Stream support. I'm currently trying to port the GNU ccRTP library to the ESP32 and as of now, it doesn't seem to be that big of a stretch.

valioiv, 2021-06-22:
@jason-mao , anyone, any updates?

chitarrastonata, 2022-01-09:
Hi, I'm also interested to UDP. I have a NOT-standard application that use UDP to send multichannel audio. If it is not planned from Espressif to have a UDP, there is some support and some tutorial to implement it by myself? 
tcp_stream is not so different to a UDP implementation. Unfortunately the implementation of TCP stream is really critical to me.
Thanks. Luca

baptabl, 2022-06-02:
+1 really missing UDP (any of RTP, [SRT](https://en.wikipedia.org/wiki/Secure_Reliable_Transport), [RIST](https://en.wikipedia.org/wiki/Reliable_Internet_Stream_Transport)) audio support here :+1:

JeremyXXJ, 2025-02-07:
现在依然没有UDP(RTP)的支持吗

jason-mao, 2025-02-18:
UDP related protocols examples:
https://github.com/espressif/esp-webrtc-solution/tree/main/solutions
https://github.com/espressif/esp-adf/tree/master/examples/protocols