espressif/esp-adf #95: LINE IN => sound fx => EARPHONE

fsicre, 2018-12-20:
Hi,
first thanks a lot for this great library, I had much much fun using it with my ESP32-A1S board.

Could you please explain how one could make sound FX (like guitar pedal, or voice modifier) ?

setting up a pipeline like this one :

LINE IN (not the mic) => sound fx (like reverb, echo, distortion etc) => EARPHONE

jason-mao, 2019-01-03:
@fsicre I think you can do like follow: `receive line in data from one i2s stream`->`sound filter(sonic-voice change,)` ->`playback on i2s stream`

fsicre, 2019-01-24:
Thank you jason, that's exactly what I was thinking.

Unfortunately I can't figure out how make such a filter.
I do manage setting up an i2s input stream, an i2s output stream but how setup such a filter ?

Any pointer / example ?

Thanks a lot

commit 627f1b5f (2019-04-02): Merge branch 'bugfix/playlist_parser' into 'master'

Fixed playlist parsing algo + Added more `content_type` checks + Fixed some mem leaks.

Closes #95

See merge request adf/esp-adf-internal!263
closed 2019-04-02 by igrr