espressif/esp-adf #96: rec_engine_data_read 返回值，经常有非常大的值比如16584336

junbowu, 2018-12-25:
/**
 * @brief Read voice data after REC_EVENT_VAD_START.
 *
 * @param buffer        data pointer
 * @param buffer_size   Size of buffer, must be equal to REC_ONE_BLOCK_SIZE.
 * @param waiting_time  Timeout for reading data. Default time of REC_ONE_BLOCK_SIZE is 100ms, larger than 100ms is recommended.
 *
 * @return
 *      - -2    : timeout of read
 *      - -1    : parameters invalid or task not running.
 *      - 0     : last voice block.
 *      - others: voice block index.
 */
int rec_engine_data_read(uint8_t *buffer, int buffer_size, int waiting_time);
实际在用dueros这个例子时，
```
int ret = rec_engine_data_read(voiceData, REC_ONE_BLOCK_SIZE, 110 / portTICK_PERIOD_MS);
ESP_LOGI(TAG, "index = %d", ret);
```
### I (483264) DUEROS: index = 16584336
I (483267) DUEROS: index = 16584336
I (483271) DUEROS: index = 16584336
I (483275) DUEROS: index = 16584336
I (483279) DUEROS: index = 16584336
I (483283) DUEROS: index = 16584336
I (483288) DUEROS: index = 16584336
I (483294) DUEROS: index = 16584336
I (483296) DUEROS: index = 16584336
I (483301) DUEROS: index = 16584336
I (483304) DUEROS: index = 16584336
I (483310) DUEROS: index = 16584336
I (483313) DUEROS: index = 1
I (483554) DUEROS: index = 2
I (483653) DUEROS: index = 3
I (483701) DUEROS: index = 4
I (483772) DUEROS: index = 5
I (483830) DUEROS: index = 6
I (483872) DUEROS: index = 7
I (483961) DUEROS: index = 8
I (484031) DUEROS: index = 9
I (484130) DUEROS: index = 10
I (484272) DUEROS: index = 11
I (484425) DUEROS: index = 12
I (484538) DUEROS: index = 13
I (484684) DUEROS: index = 14**

jason-mao, 2018-12-26:
@junbowu 我们正在修复这个问题

jason-mao, 2018-12-28:
@junbowu This issue has been fixed on d9901407f8c4ff1ca6e3f3ab948bb391558e00b5

commit ec413e02 (2019-08-16): Merge branch 'bugfix/cmake_build_error' into 'master'

Bugfix/cmake build error

Closes #96

See merge request adf/esp-adf-internal!341
closed 2018-12-29 by jason-mao