espressif/esp-adf #97: Building problem with latest update

RemyEE, 2019-01-02:
I just update the ESP-ADF library to latest update @c02e209.
Building process (make menuconfig, make all, make flash...) now not working properly.
Anyone else facing same problem with me?
Here is the logs:
`root@Useed-DungNH:/mnt/d/esp/workspace/play_mp3# make menuconfig
make[1]: Entering directory '/mnt/d/esp/esp-adf/esp-idf/tools/kconfig'
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/check-lxdialog.sh: line 3: $'\r': command not found
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/check-lxdialog.sh: line 5: syntax error near unexpected token `$'\r''
'mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/check-lxdialog.sh: line 5: `ldflags()
cc -c   -DLOCALE -MD  /mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf.c -o mconf.o
In file included from /mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf.c:23:0:
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:38:10: error: #include expects "FILENAME" or <FILENAME>
 #include CURSES_LOC
          ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:103:2: error: unknown type name ‘chtype’
  chtype atr; /* Color attribute */
  ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:200:16: error: unknown type name ‘WINDOW’
 int on_key_esc(WINDOW *win);
                ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:221:17: error: unknown type name ‘WINDOW’
 void attr_clear(WINDOW * win, int height, int width, chtype attr);
                 ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:221:54: error: unknown type name ‘chtype’
 void attr_clear(WINDOW * win, int height, int width, chtype attr);
                                                      ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:223:21: error: unknown type name ‘WINDOW’
 void print_autowrap(WINDOW * win, const char *prompt, int width, int y, int x);
                     ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:224:19: error: unknown type name ‘WINDOW’
 void print_button(WINDOW * win, const char *label, int y, int x, int selected);
                   ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:225:18: error: unknown type name ‘WINDOW’
 void print_title(WINDOW *dialog, const char *title, int width);
                  ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:226:15: error: unknown type name ‘WINDOW’
 void draw_box(WINDOW * win, int y, int x, int height, int width, chtype box,
               ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:226:66: error: unknown type name ‘chtype’
 void draw_box(WINDOW * win, int y, int x, int height, int width, chtype box,
                                                                  ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:227:8: error: unknown type name ‘chtype’
        chtype border);
        ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:228:18: error: unknown type name ‘WINDOW’
 void draw_shadow(WINDOW * win, int y, int x, int height, int width);
                  ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf.c: In function ‘conf’:
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:97:29: error: ‘KEY_MAX’ undeclared (first use in this function)
 #define ERRDISPLAYTOOSMALL (KEY_MAX + 1)
                             ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf.c:671:45: note: in expansion of macro ‘ERRDISPLAYTOOSMALL’
   if (res == 1 || res == KEY_ESC || res == -ERRDISPLAYTOOSMALL)
                                             ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:97:29: note: each undeclared identifier is reported only once for each function it appears in
 #define ERRDISPLAYTOOSMALL (KEY_MAX + 1)
                             ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf.c:671:45: note: in expansion of macro ‘ERRDISPLAYTOOSMALL’
   if (res == 1 || res == KEY_ESC || res == -ERRDISPLAYTOOSMALL)
                                             ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf.c: In function ‘show_help’:
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf.c:793:19: warning: implicit declaration of function ‘getmaxx’ [-Wimplicit-function-declaration]
  help.max_width = getmaxx(stdscr) - 10;
                   ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf.c:793:27: error: ‘stdscr’ undeclared (first use in this function)
  help.max_width = getmaxx(stdscr) - 10;
                           ^
In file included from /mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf.c:23:0:
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf.c: In function ‘conf_choice’:
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/lxdialog/dialog.h:97:29: error: ‘KEY_MAX’ undeclared (first use in this function)
 #define ERRDISPLAYTOOSMALL (KEY_MAX + 1)
                             ^
/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf.c:855:9: note: in expansion of macro ‘ERRDISPLAYTOOSMALL’
   case -ERRDISPLAYTOOSMALL:
         ^
Makefile:171: recipe for target 'mconf.o' failed
make[1]: *** [mconf.o] Error 1
make[1]: Leaving directory '/mnt/d/esp/esp-adf/esp-idf/tools/kconfig'
make: *** No rule to make target '/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/conf-idf', needed by '/mnt/d/esp/esp-adf/esp-idf/tools/kconfig/mconf-idf'.  Stop.`

jason-mao, 2019-01-02:
@RemyEE Hi, could you delete follow files and test again.`esp/esp-adf/esp-idf/tools/kconfig/*.o `, `esp/esp-adf/esp-idf/tools/kconfig/*.d`, executable files-`mconf` and`conf`, `esp/esp-adf/esp-idf/tools/kconfig/lxdialog/*.o`, `esp/esp-adf/esp-idf/tools/kconfig/lxdialog/*.d`.

igrr, 2019-01-03:
> line 5: syntax error near unexpected token $'\r''

This points to the issue with line endings. Perhaps you have cloned the repository on Windows (which uses CRLF) and building the code on Linux (which requires LF only)?

RemyEE, 2019-01-03:
@igrr You are right. I cloned on Windows and build on Subsystem Linux. I cloned the repository again on Subsystem Linux. Now it works. Thank you.

closed 2019-01-03 by RemyEE