espressif/esp-adf #104: face error when run DLNA example by Lyrat

spiritfire, 2019-01-20:
i am using a 3rd party DLNA service called "stream what you hear"
[image]

- *1* i firstly open reset the lyrat, and it says DLNA started, no error.
- *2* and i open the "stream what you hear" now, i think the "stream what you hear" is going to search the DLNA renderer.
at this moment, the monitor screen will show some errors like below:
[image]
and the "stream what you hear" service could not find the lyrat renderer.

instead, if i use the universal media player, and do as what the readme,md file says, the DLNA works normally.

How could i solve this problem?what is the reason of this error? i need cast all sound from computer to DLNA using this so "stream what you hear" software
 please help

tuanpmt, 2019-01-20:
Hi @spiritfire 

Maybe there is a problem with ADF SSDP with another DLNA client. While working with DLNA, I found a lot of DLNA Server working slightly differently. We just tried on popular servers like Kodi and Universal Media Server.

We will try SWYH soon and find out what the problem is, thank you for reporting the error

spiritfire, 2019-01-21:
> Hi @spiritfire
>
> Maybe there is a problem with ADF SSDP with another DLNA client. While working with DLNA, I found a lot of DLNA Server working slightly differently. We just tried on popular servers [...]

thank you , hope you can solve this soon. Really need this function now for a new product development.
- _FYI, i also tried the software called "simpledlna" , it has the same issue with SWYH._

closed 2020-07-09 by espressif-bot