espressif/esp-adf #106: VoIP support available in next release?

songshan99, 2019-01-23:
Will next release have examples that supports VoIP? When are you planning to release it?
We are evaluating ESP32 as a VoIP device, this will help us a lot. Thanks for answering

tuanpmt, 2019-01-24:
Hi @songshan99 

The SIP stack and VoIP will be supported by ADF soon, We planned at the end of Q1-2019

Thank you for interesting about this

songshan99, 2019-04-01:
Hi, is there any progress on the VoIP support and the Q1 release? We are looking forward to evaluate it for our new product. Thanks in advance.

Niek, 2019-06-27:
I see the code just got merged, including a sample: https://github.com/espressif/esp-adf/blob/master/examples/advanced_examples/voip/README.md

jason-mao, 2019-06-27:
@Niek Thanks. @songshan99 @glennhuber SIP has been released on 49d73308bb12f555ea3eebe8f4085a5ed7f206cf

closed 2019-06-27 by jason-mao