espressif/esp-adf #107: DLNA m4a support

arlaor, 2019-01-28:
Is possible DLNA example m4a support? and foobar2000 upnp media server support?

tuanpmt, 2019-01-28:
Hi @arlaor 

The DLNA can be supported m4a by implementing an m4a player https://github.com/espressif/esp-adf/blob/master/examples/advanced_examples/dlna/main/audio_player.c#L166

With foobar2000, could you please give a try?

thanks

arlaor, 2019-01-28:
Hi @tuanpmt 

then I must include something like this [code](https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_living_stream/main/play_living_stream_example.c) inside the [file](https://github.com/espressif/esp-adf/blob/master/examples/advanced_examples/dlna/main/audio_player.c#L166) ?

and tried to use foobar2000 with UPNP server and selecting my esp32-Lyrat as audio output, but it stops at 2 seconds, without emitting sound

tuanpmt, 2019-01-30:
Could you please provide the log

Thanks

arlaor, 2019-01-31:
Hi @tuanpmt

send various logs

for foobar2000 with my [code modify](https://github.com/arlaor/DLNA_AAC_SUPPORT/blob/master/audio_player_modified.c), this is the log for AAC file in foobar2000

I must clarify that the code works partially on the universal media server, but for some reason the mp3 format is disabled, some lines of code must be missing to enable it when recognizing the file extension to be played
Also, when a song is paused or stopped in the universal media server, there are no options to play it again, until you reset the esp32-lyrat and play a new song again.
The play button disappears and only have the option to stop

In foobar2000 with AAC file not working
 ```javascript
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5876
load:0x40078000,len:9288
load:0x40080000,len:6008
entry 0x4008032c
[0;32mI (29) boot: ESP-IDF v3.1.1-6-g2aa9a2118 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 23:04:12[0m
[0;32mI (31) boot: Enabling RNG early entropy source...[0m
[0;32mI (35) boot: SPI Speed      : 40MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (50) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (80) boot: End of partition table[0m
[0;32mI (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x68a8c (428684) map[0m
[0;32mI (243) esp_image: segment 1: paddr=0x00078ab4 vaddr=0x3ffb0000 size=0x03874 ( 14452) load[0m
[0;32mI (249) esp_image: segment 2: paddr=0x0007c330 vaddr=0x3ffb3874 size=0x00000 (     0) load[0m
[0;32mI (250) esp_image: segment 3: paddr=0x0007c338 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (260) esp_image: segment 4: paddr=0x0007c740 vaddr=0x40080400 size=0x038d0 ( 14544) load[0m
[0;32mI (274) esp_image: segment 5: paddr=0x00080018 vaddr=0x400d0018 size=0xc0cc8 (789704) map[0m
[0;32mI (553) esp_image: segment 6: paddr=0x00140ce8 vaddr=0x40083cd0 size=0x1508c ( 86156) load[0m
[0;32mI (589) esp_image: segment 7: paddr=0x00155d7c vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (590) esp_image: segment 8: paddr=0x00155d84 vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (611) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (611) boot: Disabling RNG early entropy source...[0m
[0;32mI (613) spiram: SPI RAM mode: flash 40m sram 40m[0m
[0;32mI (616) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (623) cpu_start: Pro cpu up.[0m
[0;32mI (627) cpu_start: Starting app cpu, entry point is 0x4008146c[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1539) spiram: SPI SRAM memory test OK[0m
[0;32mI (1540) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1540) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1546) heap_init: At 3FFB9BC0 len 00026440 (153 KiB): DRAM[0m
[0;32mI (1552) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1559) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1565) heap_init: At 40098D5C len 000072A4 (28 KiB): IRAM[0m
[0;32mI (1572) cpu_start: Pro cpu start user code[0m
[0;32mI (1576) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (1584) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (264) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
I (391) wifi: wifi driver task: 3ffc1a50, prio:23, stack:3584, core=0
I (391) wifi: wifi firmware version: b2c9a19
[... 165 lines cut ...]
this is the log in universal media server with MP3 file not working
 ```javascript
[0;32mI (1847121) DLNA_EXAMPLE: SetAVTransportURI=http://192.168.1.5:5001/get/108$Temp/D%3A%5Cm4a%5Camanece.mp3[0m
[0;32mI (1847431) DLNA_EXAMPLE: Play with speed=1[0m
[0;32mI (1847431) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:278, MEM Total:4282832 Bytes, Inter:151536 Bytes, Dram:122224 Bytes

[0m
[0;32mI (1847441) AUDIO_ELEMENT: [aac] AEL_MSG_CMD_RESUME,state:4[0m
[0;32mI (1847451) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1[0m
[0;32mI (1847451) I2S_STREAM: AUDIO_STREAM_WRITER[0m
[0;33mW (1847461) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT[0m
[0;32mI (1847461) AUDIO_PIPELINE: Pipeline started[0m
[0;33mW (1847461) AUDIO_ELEMENT: IN-[aac] AEL_IO_ABORT[0m
[0;31mE (1847471) AAC_DECODER: empty aac file[0m
[0;31mE (1847481) AUDIO_ELEMENT: [aac] AEL_STATUS_ERROR_OPEN[0m
[0;32mI (1847481) AAC_DECODER: Closed[0m
[0;32mI (1847481) AUDIO_PLAYER: Stop pipeline[0m
[0;32mI (1847491) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:4[0m
[0;32mI (1847491) DLNA_EXAMPLE: _avt_get_trans_state STOPPED[0m
[0;33mW (1847751) UPNP_NOTIFY: Invalid subscription, Unsubscribe=uuid:6931589d-e07e-bb77-16c6-1dee132f7726[0m
[0;32mI (1853161) DLNA_EXAMPLE: _avt_get_trans_state STOPPED[0m
[0;32mI (1864631) DLNA_EXAMPLE: _avt_get_trans_state STOPPED[0m
[0;32mI (1875301) DLNA_EXAMPLE: _avt_get_trans_state STOPPED[0m
 ```
I must also clarify that I am not a programmer or something similar, I am an amateur that I have only made the union of code trying to carry a logical sequence but i have failed

tuanpmt, 2019-01-31:
Hi @arlaor 

The DLNA still working good, there are some points:
- If you want to use mp3/aac, you should link the pipeline by the correct decoder. Ex: `audio_pipeline_link(ap->pipeline, (const char *[]) {"http", "mp3", "i2s"}, 3);`  for mp3. Each pipeline can only work with a single link at a time
- Foobar2000 send the playlist to the Renderer, your player must support playlist parser before playing the audio

arlaor, 2019-01-31:
> * audio_pipeline_link(ap->pipeline, (const char *[]) {"http", "mp3", "i2s"}, 3);

Wich is the way to go? that I must modify or include in my code, I am not clear about what is missing

tuanpmt, 2019-01-31:
If you want to play the mp3 file, you must remove the red rectangle 
[image]

arlaor, 2019-01-31:
the idea is to be able to reproduce the two formats or more, if possible

I am reviewing this [code](https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_http_select_decoder/main/play_http_select_decoder_example.c) to try to adapt it to DLNA. Is it possible that it will be recognized and work?

arlaor, 2019-01-31:
Hi @tuanpmt 

I have adapted part of this [code](https://github.com/espressif/esp-adf/blob/master/examples/player/pipeline_http_select_decoder/main/play_http_select_decoder_example.c) to my [DLNA code](https://github.com/arlaor/DLNA_AAC_SUPPORT/blob/master/Adapt_select.c), but I can not verify selected_decoder to make the change of decoder
The aac file working
The MP3 file don't working
can you verify it please

This is The log with this [code](https://github.com/arlaor/DLNA_AAC_SUPPORT/blob/master/Adapt_select.c) modified
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:5876
load:0x40078000,len:9288
load:0x40080000,len:6008
0x40080000: _iram_start at D:/ARDUINO/espressif/lyrat/esp-adf/esp-idf/components
/freertos/xtensa_vectors.S:1685

entry 0x4008032c
0x4008032c: _KernelExceptionVector at ??:?

I (29) boot: ESP-IDF v3.1.1-6-g2aa9a2118 2nd stage bootloader
I (29) boot: compile time 23:04:12
I (31) boot: Enabling RNG early entropy source...
I (35) boot: SPI Speed      : 40MHz
I (39) boot: SPI Mode       : DIO
I (43) boot: SPI Flash Size : 4MB
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00200000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x657d4 (415
700) map
I (239) esp_image: segment 1: paddr=0x000757fc vaddr=0x3ffb0000 size=0x03874 ( 1
4452) load
I (245) esp_image: segment 2: paddr=0x00079078 vaddr=0x3ffb3874 size=0x00000 (
   0) load
I (246) esp_image: segment 3: paddr=0x00079080 vaddr=0x40080000 size=0x00400 (
1024) load
0x40080000: _iram_start at D:/ARDUINO/espressif/lyrat/esp-adf/esp-idf/components
/freertos/xtensa_vectors.S:1685

I (255) esp_image: segment 4: paddr=0x00079488 vaddr=0x40080400 size=0x06b88 ( 2
7528) load
I (275) esp_image: segment 5: paddr=0x00080018 vaddr=0x400d0018 size=0xbaa4c (76
4492) map
0x400d0018: _stext at ??:?

I (543) esp_image: segment 6: paddr=0x0013aa6c vaddr=0x40086f88 size=0x11dd4 ( 7
3172) load
0x40086f88: i2c_isr_handler_default at D:/ARDUINO/espressif/lyrat/esp-adf/esp-id
f/components/driver/i2c.c:1074

I (574) esp_image: segment 7: paddr=0x0014c848 vaddr=0x400c0000 size=0x00000 (
   0) load
I (574) esp_image: segment 8: paddr=0x0014c850 vaddr=0x50000000 size=0x00000 (
   0) load
I (595) boot: Loaded app from partition at offset 0x10000
I (595) boot: Disabling RNG early entropy source...
I (597) spiram: SPI RAM mode: flash 40m sram 40m
[... 212 lines cut ...]
I (70391) AUDIO_ELEMENT: [http] AEL_MSG_CMD_RESUME,state:4
E (70571) HTTP_STREAM: Invalid HTTP stream
W (70571) HTTP_STREAM: No more data,errno:0, total_bytes:0
I (70571) AUDIO_ELEMENT: IN-[http] AEL_IO_DONE,0
I (72871) DLNA_EXAMPLE: SetAVTransportURI=http://192.168.1.5:5001/get/null$Temp/
C%3A%5CProgram+Files+%28x86%29%5CUniversal+Media+Server%5Camanece
I (73181) DLNA_EXAMPLE: Play with speed=1
I (73181) AUDIO_ELEMENT: [aac] Element task created
I (73181) AUDIO_PIPELINE: Func:audio_pipeline_run, Line:278, MEM Total:4282580 B
ytes, Inter:151000 Bytes, Dram:121688 Bytes

I (73201) AUDIO_ELEMENT: [http] RESUME: Element has finished, state:5
I (73201) AUDIO_PIPELINE: Pipeline started
I (73201) AUDIO_ELEMENT: [i2s] AEL_MSG_CMD_RESUME,state:1
I (73211) I2S_STREAM: AUDIO_STREAM_WRITER
I (73201) AUDIO_ELEMENT: [aac] AEL_MSG_CMD_RESUME,state:1
I (73221) AUDIO_ELEMENT: IN-[aac] AEL_IO_DONE,-2
E (73231) AAC_DECODER: empty aac file
E (73231) AUDIO_ELEMENT: [aac] AEL_STATUS_ERROR_OPEN
W (73241) AUDIO_ELEMENT: IN-[i2s] AEL_IO_ABORT
I (73241) AAC_DECODER: Closed
I (73271) HTTP_SELECT_AAC_EXAMPLE: Stop pipeline
W (73271) HTTP_SELECT_AAC_EXAMPLE: [ * ] Stop event received
I (73271) DLNA_EXAMPLE: _avt_get_trans_state STOPPED
I (76771) DLNA_EXAMPLE: _avt_get_trans_state STOPPED

commit cbc6bb17 (2020-10-13): Merge branch 'bugfix/build_error_idf_3_3' into 'master'

build error because updata idf3.3.2 to idf3.3.3

Closes #107

See merge request adf/esp-adf-internal!639
closed 2020-07-09 by espressif-bot