espressif/esp-adf #108: pipeline_bt_sink audio example audio quality

MrityunjaiKumar, 2019-01-28:
Target platform  -  lyraT 4.3  

Facing a problem where the audio quality is degraded compared to original audio. Here are some screenshots of logs.

[image]

[image]

[image]

Any clue?

mgsscorr1, 2019-01-31:
I am having the same issue with my program. After a recent update of the ADF, the resampling feature was removed and my audio Quality has diminished.

theronic, 2019-03-10:
The Bluetooth sink audio quality on the ESP32-LYRATD-MSC v2.2 is also bad despite resampling from 44100Hz to 48000Hz in my pipeline (ZL38063 doesn't support 44.1kHz). If I set the volume to 100%, it sounds terrible. Suspect ADC offset issue. At 70% volume Bluetooth sounds soft and tinny at best.

Any advice, @jason-mao? I saw something about having to forcefully flash the ZL38063 firmware on v2.1, but don't know if that's relevant, or how.

steenou, 2019-08-08:
What is the bad audio quality like? Are you hearing glitching due to zeros in the output like in this issue I posted in the idf repo? https://github.com/espressif/esp-idf/issues/3775
Does it occur when you first start a stream?

theronic, 2019-08-09:
It's been a while, so I don't recall exactly, but I remember it wasn't glitchy, just low fidelity. It sounded like an undersampled recording - maybe a noisy DAC or quantization error.

Due to lack of support, I have since given up on this hardware and started making my own PCBs.

jason-mao, 2019-08-17:
@steenou I have note your report issue on IDF.

jason-mao, 2019-08-17:
@theronic You're right. 44.1kHz music should be resample to 48kHz for The ESP32-LYRATD-MSC board. We will test as your said.  BTW, if you meet this issue again, please let me know.

pad52, 2019-09-20:
> The Bluetooth sink audio quality on the ESP32-LYRATD-MSC v2.2 is also bad despite resampling from 44100Hz to 48000Hz in my pipeline (ZL38063 doesn't support 44.1kHz). If I set the volume to 100%, it [...]

Hello @theronic I got the same problem but resolved with just one line change:

FROM:
        .communication_format = I2S_COMM_FORMAT_I2S_MSB,
TO:
        .communication_format = I2S_COMM_FORMAT_I2S | I2S_COMM_FORMAT_I2S_MSB,

Altough I don't understand the real difference, it solved my problems,
check if this could work for you.

Tarun-Narain, 2019-12-13:
I got clean Audio from LyraTD-MSC board in pipeline_bt_sink, please refer
https://github.com/espressif/esp-adf/pull/279/files

closed 2022-02-08 by jason-mao