espressif/esp-adf #109: downmix example not working

HanYangZhao, 2019-01-30:
When compiling the downmix example, I get the following error msg
`CC build/main/downmix_pipeline_main.o
/home/hanz/esp/esp-adf/examples/advanced_examples/downmix_pipeline/main/downmix_pipeline_main.c:24:21: fatal error: downmix.h: No such file or directory
compilation terminated.
/home/hanz/akan/mf-esp32-project/esp-idf/make/component_wrapper.mk:285: recipe for target 'downmix_pipeline_main.o' failed
make[1]: *** [downmix_pipeline_main.o] Error 1
/home/hanz/akan/mf-esp32-project/esp-idf/make/project.mk:468: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2`

Seems like it can't find the `downmix.h` file

Any idea?

krzychb, 2019-01-30:
Hi @HanYangZhao,
Try old good solution https://github.com/espressif/esp-adf/issues/1#issuecomment-379539070

HanYangZhao, 2019-02-08:
Temporarily fixed by adding `EXTRA_COMPONENT_DIRS += $(ADF_PATH)/esp-adf/components/esp-adf-libs/esp_codec` to Makefile

closed 2019-02-08 by HanYangZhao