espressif/esp-adf #113: VAD detecting any and every noise

mgsscorr1, 2019-02-14:
While attempting to run the VAD example, it has come to my conclusion that the "Voice" activity detector fires with any and ever sound that occurs. 

Through testing of every single combination of the following settings VAD_MODE_#, VAD_SAMPLE_RATE_HZ, VAD_FRAME_LENGTH_MS
Nothing seems to solve the issue.

Are there any plans to have the VAD up and running properly or more efficiently?

Yohannfra, 2022-01-07:
Hi,
I'm having the same issue, does anyone have a solution or an alternative ?

Thanks

jason-mao, 2025-02-18:
The new VAD and NS of ESP-SR have better performance than before. And, this topic has become inactive, so I'm closing the issue. Please reopen it if you have any questions or need any further assistance.

closed 2025-02-18 by jason-mao