espressif/esp-adf #117: ESP32_Lyrat跑dueros例程，语音没有没有反应

yaoshila, 2019-02-20:
之前能跑通的，昨天突然就不行了，可以语音唤醒，提问说话没有反应，检查了wifi、麦克风、录音、sd卡、播放等单独都没有问题...
这是登录的log:
`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x3f (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5828
load:0x40078000,len:7892
ho 0 tail 12 room 4
load:0x40080000,len:5948
entry 0x40080320
[0;32mI (31) boot: ESP-IDF v3.1.1-6-g2aa9a2118 2nd stage bootloader[0m
[0;32mI (31) boot: compile time 21:46:31[0m
[0;32mI (35) boot: Enabling RNG early entropy source...[0m
[0;32mI (37) boot: SPI Speed      : 80MHz[0m
[0;32mI (41) boot: SPI Mode       : DIO[0m
[0;32mI (45) boot: SPI Flash Size : 4MB[0m
[0;32mI (49) boot: Partition Table:[0m
[0;32mI (52) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (60) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (67) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (75) boot:  2 factory          factory app      00 00 00010000 00300000[0m
[0;32mI (82) boot: End of partition table[0m
[0;32mI (86) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x8f598 (587160) map[0m
[0;32mI (264) esp_image: segment 1: paddr=0x0009f5c0 vaddr=0x3ffb0000 size=0x00a50 (  2640) load[0m
[0;32mI (265) esp_image: segment 2: paddr=0x000a0018 vaddr=0x400d0018 size=0x12ebcc (1240012) map[0m
[0;32mI (627) esp_image: segment 3: paddr=0x001cebec vaddr=0x3ffb0a50 size=0x035ec ( 13804) load[0m
[0;32mI (631) esp_image: segment 4: paddr=0x001d21e0 vaddr=0x3ffb403c size=0x00000 (     0) load[0m
[0;32mI (634) esp_image: segment 5: paddr=0x001d21e8 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (643) esp_image: segment 6: paddr=0x001d25f0 vaddr=0x40080400 size=0x17c0c ( 97292) load[0m
[0;32mI (684) esp_image: segment 7: paddr=0x001ea204 vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (684) esp_image: segment 8: paddr=0x001ea20c vaddr=0x50000000 size=0x00000 (     0) load[0m
[0;32mI (704) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (704) boot: Disabling RNG early entropy source...[0m
[0;32mI (705) spiram: SPI RAM mode: flash 80m sram 80m[0m
[0;32mI (710) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
[0;32mI (717) cpu_start: Pro cpu up.[0m
[0;32mI (721) cpu_start: Starting app cpu, entry point is 0x40081508[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (1190) spiram: SPI SRAM memory test OK[0m
[0;32mI (1191) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (1191) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (1197) heap_init: At 3FFBB688 len 00024978 (146 KiB): DRAM[0m
[0;32mI (1203) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (1210) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (1216) heap_init: At 4009800C len 00007FF4 (31 KiB): IRAM[0m
[0;32mI (1222) cpu_start: Pro cpu start user code[0m
[0;32mI (1227) spiram: Adding pool of 4096K of external SPI memory to heap allocator[0m
[0;32mI (1235) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations[0m
[0;32mI (138) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (221) DUEROS: ADF version is v1.0-4-g01f45d8[0m
[0;33mW (221) ESP_PERIPH: Peripherals have been initialized already[0m
[0;32mI (222) gpio: GPIO[36]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;32mI (231) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
[0;33mW (266) PERIPH_TOUCH: _touch_init[0m
[0;32mI (266) SDCARD: Trying to mount with base path=/sdcard[0m
[0;32mI (313) SDCARD: CID name SD!
[0m
[0;32mI (541) gpio: GPIO[19]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 [0m
I (362,tid:3ffc6bac) lightduer_session.c(  44):random = 75933
[0;31mE (542) gpio: gpio_install_isr_service(394): GPIO isr service already installed[0m
[... 100 lines cut ...]

[0m
[0;33mW (9439) I2S: I2S driver already installed[0m
[0;33mW (9444) I2S: I2S driver already installed[0m
[0;32mI (9449) STAGEFRIGHTMP3_DECODER: MP3 init[0m
[0;32mI (9456) AUDIO_WRAPPER: Func:setup_player, Line:167, MEM Total:4178960 Bytes, Inter:143808 Bytes, Dram:111088 Bytes

[0m
[0;32mI (9465) AUDIO_WRAPPER: esp_audio instance is:0x3f8148c4[0m
I (9301,tid:3ffc6bac) lightduer_dcs_router.c( 283):namespace: ai.dueros.private.protocol
I (9302,tid:3ffc6bac) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.screen_extended_card
I (9310,tid:3ffc6bac) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.system
I (9317,tid:3ffc6bac) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.voice_input
I (9326,tid:3ffc6bac) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.voice_output
I (9334,tid:3ffc6bac) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.speaker_controller
I (9344,tid:3ffc6bac) lightduer_dcs_router.c( 283):namespace: ai.dueros.device_interface.audio_player
[0;32mI (9533) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m
[0;32mI (9544) DUEROS: Dueros DUER_CMD_CONNECTED, duer_state:1[0m
[0;31mE (9544) DUEROS: event: DUER_EVENT_STARTED[0m
W (9373,tid:3ffc6bac) lightduer_events.c(  80):[lightduer_ca] <== event end = 400F0F48, timespent = 194
I (9379,tid:3ffc6bac) lightduer_connagent.c( 212):add resource successfully!!
I (9388,tid:3ffc6bac) lightduer_connagent.c( 212):add resource successfully!!
W (9393,tid:3ffc6bac) lightduer_engine.c( 634):data cache has not sent, pending..., dcache_len:1, qcache_len:4
W (9405,tid:3ffc6bac) lightduer_system_info.c( 306):Undefined memory type, 0
E (9409,tid:3ffc6bac) lightduer_system_info.c( 389):Sys Info: Get disk info failed`

yaoshila, 2019-02-20:
这是唤醒和语音的log：
[0;33mW (63952) REC_ENG: ### spot keyword ###[0m
[0;32mI (63953) DUEROS: rec_engine_cb - REC_EVENT_WAKEUP_START[0m
[0;32mI (63953) ESP_PERIPH: This peripheral has been added[0m
[0;33mW (64706) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;32mI (64888) DUEROS: rec_engine_cb - REC_EVENT_VAD_START[0m
[0;32mI (64889) DUEROS: Recv Que DUER_CMD_START[0m
[0;32mI (64891) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m
I (64716,tid:3ffc33a0) lightduer_dcs_local.c( 188):Current dialog id: 1ad3000000000440b18ccf0000fccc00000004
[0;33mW (66969) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;32mI (66970) ESP_PERIPH: This peripheral has been added[0m
[0;32mI (66970) DUEROS: rec_engine_cb - REC_EVENT_VAD_STOP, state:3[0m
[0;33mW (67768) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;32mI (67819) DUEROS: Dueros DUER_CMD_STOP[0m
[0;33mW (68574) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (69387) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (70199) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (71006) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (71818) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (72631) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (73445) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (74249) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (75062) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (75874) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (76680) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;32mI (76970) REC_ENG: Wakeup time is out[0m
[0;31mE (76980) REC_ENG: State FETCH_DATA, wakeup_time_out:1, vad_speech_on:0, vad_time_mode:0[0m
[0;33mW (76981) REC_ENG: State WAKEUP_END[0m
[0;32mI (76982) ESP_PERIPH: This peripheral has been added[0m
[0;32mI (76988) DUEROS: rec_engine_cb - REC_EVENT_WAKEUP_END[0m
[0;32mI (76994) REC_ENG: state idle[0m
[0;33mW (91640) REC_ENG: ### spot keyword ###[0m
[0;32mI (91640) DUEROS: rec_engine_cb - REC_EVENT_WAKEUP_START[0m
[0;32mI (91641) ESP_PERIPH: This peripheral has been added[0m
[0;33mW (92462) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;32mI (93368) DUEROS: rec_engine_cb - REC_EVENT_VAD_START[0m
[0;32mI (93369) DUEROS: Recv Que DUER_CMD_START[0m
[0;32mI (93372) AUDIO_WRAPPER: duer_dcs_get_speaker_state[0m
I (93196,tid:3ffc33a0) lightduer_dcs_local.c( 188):Current dialog id: 1ad300000000044703312900016c0b00000006
[0;33mW (97044) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;32mI (97045) ESP_PERIPH: This peripheral has been added[0m
[0;32mI (97046) DUEROS: rec_engine_cb - REC_EVENT_VAD_STOP, state:3[0m
[0;33mW (97856) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;32mI (97915) DUEROS: Dueros DUER_CMD_STOP[0m
[0;33mW (98668) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (99474) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (100287) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (101100) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (101912) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (102718) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (103530) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (104343) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (105149) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (105962) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;33mW (106774) REC_ENG: State VAD silence_time >= vad_off_window[0m
[0;32mI (107045) REC_ENG: Wakeup time is out[0m
[0;31mE (107074) REC_ENG: State FETCH_DATA, wakeup_time_out:1, vad_speech_on:0, vad_time_mode:0[0m
[0;33mW (107074) REC_ENG: State WAKEUP_END[0m
[0;32mI (107076) ESP_PERIPH: This peripheral has been added[0m
[0;32mI (107082) DUEROS: rec_engine_cb - REC_EVENT_WAKEUP_END[0m
[0;32mI (107088) REC_ENG: state idle[0m

jason-mao, 2019-02-20:
> 这是唤醒和语音的log：
> �[0;33mW (63952) REC_ENG: ### spot keyword ###�[0m
> �[0;32mI (63953) DUEROS: rec_engine_cb - REC_EVENT_WAKEUP_START�[0m
> �[0;32mI (63953) ESP_PERIPH: This peripheral has been added�[0m [...]

`REC_ENG: Wakeup time is out`, 是在timeout时间之内，VAD没检测到声音。你之前是好的，到现在有哪些改动吗？

yaoshila, 2019-02-20:
VAD和REC_ENG都没有改动，后来又拷贝了一份dueros的程序重新下载进去，也是这样的

yaoshila, 2019-02-20:
单独测试VAD也没问题...

jason-mao, 2019-02-20:
sdkconfig 有变动吗？

yaoshila, 2019-02-20:
没有，只在menuconfig里面配置了wifi

jabadabado, 2019-02-20:
I bought it...it speaks Chinese...!!! A waste of money...? Any English version?

yaoshila, 2019-02-21:
问题可能是解决了，但是不明所以...
我复现这个问题的操作：把故事机的duer_profile(此时语音正常)变更为音箱的duer_profile(变更后语音VAD就出现检测超时)。
有问题之后解决的操作：删除sdkconfig文件，make menuconfig重新生成sdkconfig文件，重新下载又可以语音了...
还没有对比这两个sdkconfig之间有什么区别，我下次再试试看，比较懵

homeoflittlewolf, 2019-02-21:
@jabadabado  you can test this : https://github.com/espressif/esp-va-sdk

closed 2019-02-22 by yaoshila