espressif/esp-adf #119: This error should be a problem with ESP-ADF?

coollofty, 2019-03-03:
This error occurred when I used both the amrwb encoder and the wav encoder. If I remove the amrwb encoder, there is no problem.

```
Python requirements from E:/esp32/esp-adf/esp-idf/requirements.txt are satisfied.
LD build/demo3.elf
D:/projects/esp32/demo3/build/lwip\liblwip.a(mem.o): In function `mem_malloc':
E:/esp32/esp-adf/esp-idf/components/lwip/lwip/src/core/mem.c:123: multiple definition of `mem_malloc'
E:/esp32/esp-adf/components/esp-adf-libs/esp_codec/lib\libesp-amrwbenc.a(mem_align.o):/home/houhaiyan/esp-audio-adf/esp-adf-libs-source/esp_codec/esp-amrwbenc/src/mem_align.c:41: first defined here
D:/projects/esp32/demo3/build/lwip\liblwip.a(mem.o): In function `mem_free':
mem.c:(.text.mem_free+0x0): multiple definition of `mem_free'
E:/esp32/esp-adf/components/esp-adf-libs/esp_codec/lib\libesp-amrwbenc.a(mem_align.o):/home/houhaiyan/esp-audio-adf/esp-adf-libs-source/esp_codec/esp-amrwbenc/src/mem_align.c:41: first defined here
collect2.exe: error: ld returned 1 exit status
make: *** [E:/esp32/esp-adf/esp-idf/make/project.mk:458：/d/projects/esp32/demo3/build/demo3.elf] error 1
```

jason-mao, 2019-03-04:
@coollofty 
Which IDF version you used?

coollofty, 2019-03-04:
The master branch newest version :)

coollofty, 2019-03-04:
And ESP-ADF also the newest version

coollofty, 2019-03-04:
> @coollofty
> Which IDF version you used?

ESP-IDF and ADF are all the master branch newest version

closed 2019-03-08 by coollofty

ichenxing, 2019-03-22:
I am faced with the same problem with @coollofty , finally how to solve this problem？@jason-mao