espressif/esp-adf #120: How can CMake be used with esp-adf?

aguynamedben, 2019-03-07:
Hi, I'm new to ESP32, so sorry if this is a dumb question.

I noticed the esp-idf uses CMake. esp-adf is an extension of esp-idf, but here CMake does not appear to be used.

I tried a few things such as:
* In CMakeLists.txt setting changing the line with `IDF_PATH` to `include($ENV{ADF_PATH}/esp-idf/tools/cmake/project.cmake)`. This gives an error `Internal error, toolchain has not been set correctly by project` in `idf_verify_environment`
* Adjusting `IDF_PATH` and `ADF_PATH` in different ways... i.e. pointing to an independent `IDF_PATH`, or pointing it to esp-adf/esp-idf.

I can't seem to get CMake to play nicely with esp-adf. Any guidance on this?

jason-mao, 2019-03-25:
@aguynamedben We have plan to support CMake, will be available soon.

closed 2019-03-25 by aguynamedben

cor3ntin, 2019-05-10:
That would be greatly appreciated, although I'd prefer that adf be merged intoo idf:)