espressif/esp-adf #121: "Play activity timeout" blocked the entire program

coollofty, 2019-03-07:
I (857) ESP_AUDIO_TASK: It's a decoder.URI:file://sdcard/sound/welcome.amr

I (857) ESP_AUDIO_TASK: 1.IN:[file],CODEC:[auto],FILTER:[NULL],OUT:[iis],RESAMPLE:[0]
I (877) ESP_AUDIO_TASK: 2.Handles,IN:0x3f808c10,CODEC:0x3f809224,FILTER:0x0,OUT:0x3f808f60
I (877) ESP_AUDIO_TASK: 2.1 Update all pipelines
I (877) ESP_AUDIO_TASK: in_rb:0x3f81d190,filter_rb:0x0,out_rb:0x3f81f9d0
I (887) ESP_AUDIO_TASK: 3. Previous starting...
I (927) AUDIO_ELEMENT: [file] Element task created
I (927) AUDIO_ELEMENT: [file] AEL_MSG_CMD_RESUME,state:1
I (927) FATFS_STREAM: File size is 0 byte,pos:0
W (927) FATFS_STREAM: No more data,ret:0
I (937) AUDIO_ELEMENT: IN-[file] AEL_IO_DONE,0
I (967) AUDIO_ELEMENT: [auto] Element task created
I (967) AUDIO_ELEMENT: [auto] AEL_MSG_CMD_RESUME,state:1
I (967) ESP_DECODER: Ready to do audio type check (line 121)
I (997) AUDIO_ELEMENT: [iis] Element task created
I (997) AUDIO_ELEMENT: [iis] AEL_MSG_CMD_RESUME,state:1
I (997) I2S_STREAM: AUDIO_STREAM_WRITER
I (1367) wifi: n:7 0, o:1 0, ap:255 255, sta:7 0, prof:1
I (2057) wifi: state: init -> auth (b0)
I (3057) wifi: state: auth -> init (200)
I (3057) wifi: n:7 0, o:7 0, ap:255 255, sta:7 0, prof:1
I (5467) ESP_AUDIO_CTRL: Func:media_ctrl_play, Line:301, MEM Total:4220264 Bytes, Inter:201908 Bytes, Dram:166112 Bytes

W (80847) ESP_AUDIO_CTRL: Play activity timeout
I (80847) ESP_AUDIO_CTRL: Type:1, ext:auto len:4
--------------------------------------------------------------------------------------------------------
First, I made a WiFi connection and then initialized the audio module and play a "init.amr" file.
But, before the warning "ESP_AUDIO_CTRL: Play activity timeout" printed, the entire program hangs for about 20 seconds, then the warning printed and program continuing to run.

coollofty, 2019-03-08:
There is no WiFi connection action, this problem also occurred. I have deleted a lot of code, but it still have. It seems to be after updating the ESP-ADF & IDF. I don’t know what is changed to led this problem, can any helps?

jason-mao, 2019-03-08:
@coollofty Did you test the cli with AMR file?

coollofty, 2019-03-09:
This is an unexplained problem, because I have not done any changes, and I started the device again today. This problem no longer exists, so it disappears...

coollofty, 2019-03-09:
Could you tell me under what circumstances will that warning appear? I can't know this because there is no source code.

jason-mao, 2019-03-27:
@coollofty Yes, this is an unexplained problem. But we can't reproduced this phenomenon，it's not easy to find out exactly reason. If you have more information,  please share. Most reason of` ESP_AUDIO_CTRL: Play activity timeout` is not found the music information(sample rate, channels).

closed 2019-12-11 by jason-mao