espressif/esp-adf #122: ESP-ADF SDK连上WIFI的时候，播放蓝牙音乐，长时间运行出现死机或者蓝牙断开，再也连接不上。

xiaoming10757, 2019-03-11:
使用官方例子    esp-adf\examples\player\pipeline_bt_sink
和WIFI连接例子 esp-adf\esp-idf\examples\wifi\simple_wifi
先初始化WIFI，修改SSID和PASSWD，连上路由，然后初始化蓝牙，手机连上蓝牙，并播放音乐，代码见附件。
运行几个小时（时间不确定）后，出现死机，如下LOG（有时候不出现如下内存错误，只是蓝牙断开，但是再也连接不上，必须重启才可以）。
[main.zip](https://github.com/espressif/esp-adf/files/2950579/main.zip)

如下log:
[0m
ASSERT_PARAM(1024 0), in rwbt.c at line 311
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0)

Core 0 register dump:
PC      : 0x400847c1  PS      : 0x00060034  A0      : 0x80088a64  A1      : 0x3ffc05b0  
A2      : 0x00000001  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x60008054  
A6      : 0x3ffc1028  A7      : 0x60008050  A8      : 0x800847c1  A9      : 0x3ffc0590  
A10     : 0x00000004  A11     : 0x00000000  A12     : 0x6000804c  A13     : 0xffffffff  
A14     : 0x00000000  A15     : 0xfffffffc  SAR     : 0x00000004  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x400846f9  LEND    : 0x40084700  LCOUNT  : 0x00000000  
Core 0 was running in ISR context:
EPC1    : 0x4019cc52  EPC2    : 0x4019cc52  EPC3    : 0x00000000  EPC4    : 0x400847c1

Backtrace: 0x400847c1:0x3ffc05b0 0x40088a61:0x3ffc05d0 0x40089153:0x3ffc05f0 0x40082695:0x3ffc0610 0x4019cc4f:0x00000000

Core 1 register dump:
PC      : 0x4019cc52  PS      : 0x00060234  A0      : 0x800924e6  A1      : 0x3ffdafc0  
A2      : 0x00000008  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x3ffda900  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x3ffc4c28  A9      : 0x3ffc4bec  
A10     : 0x00000000  A11     : 0x80000001  A12     : 0x00000000  A13     : 0x00000001  
A14     : 0x00060021  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000005  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4019cc52:0x3ffdafc0 0x400924e3:0x3ffdafe0
Rebooting...
[0;32mI (10) boot: ESP-IDF v3.1.1-6-g2aa9a2118 2nd stage bootloader[0m
[0;32mI (11) boot: compile time 11:46:14[0m
[0;32mI (20) boot: Enabling RNG early entropy source...[0m
[0;32mI (20) boot: SPI Speed      : 40MHz[0m
[0;32mI (20) boot: SPI Mode       : DIO[0m
[0;32mI (23) boot: SPI Flash Size : 4MB[0m
[0;32mI (27) boot: Partition Table:[0m
[0;32mI (31) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (38) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
[0;32mI (46) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
[0;32mI (53) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (61) boot:  3 ota_0            OTA app          00 10 00010000 00190000[0m
[0;32mI (68) boot:  4 ota_1            OTA app          00 11 001a0000 00190000[0m
[0;32mI (76) boot:  5 storage          Unknown data     01 82 00330000 000d0000[0m
[0;32mI (83) boot: End of partition table[0m
＃[0;＃＃pu_?[0;pu_?[0;pu_?[0;pu_?[0;pu_?[0;pu_?[0;pu_?[0;pu_?[0;pu_?[0;pu_?[0;31mE (3358) boot: OTA app partition slot 0 is not bootable[0m
[0;31mE (3363) esp_image: image at 0x1a0000 has invalid magic byte[0m
[0;33mW (3370) esp_image: image at 0x1a0000 has invalid SPI mode 255[0m
[0;33mW (3376) esp_image: image at 0x1a0000 has invalid SPI size 15[0m
[0;31mE (3383) boot: OTA app partition slot 1 is not bootable[0m
[0;31mE (3389) boot: No bootable app partitions in the partition table[0m
Fatal exception (29): StoreProhibited
epc1=0x40080391, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000006, depc=0x00000000

jason-mao, 2019-08-17:
@xiaoming10757 这个问题现在如何了？

esp-xiaodong, 2019-10-08:
@xiaoming10757 
1. reboot看起来是core0 wdt timeout了。可以通过menuconfige  esp32-specific->task watchdog timeout period的时间，默认是5s，可以改为8s。还有wifi和bt并存，可以将bt放到core1上，在menuconfig bluetooth里面，将controller和bluedroid都放到core1上面。
2. 异常断开，原因会有很多，需要给出log才能判断。
3. 断开后连不上，可以检查下你断开时候的处理，需要esp_bt_gap_set_scan_mode(ESP_BT_SCAN_MODE_CONNECTABLE_DISCOVERABLE); 这个处理。

jason-mao, 2020-06-19:
Closed by no more informations.

closed 2020-06-19 by jason-mao