espressif/esp-adf #124: Recorder example pipeline_wav_sdcard not working properly

RemyEE, 2019-03-12:
I am trying to use the pipeline_wav_sdcard example. But it seem not working properly. Output file should be 20 seconds length, in my case output file length is 1 second with 1411kbps bit rate.
Amr recording examples (pipeline_amr_sdcard) works fine.

zhaofx1234, 2019-03-13:
I also met the same problem, who can tell me what shall i do?

scruffynerf, 2019-03-29:
Weird, I'm having the opposite problem: pipeline_wav_sdcard works great, pipeline_amr_sdcard doesn't.  Did someone change something to fix that and break the other?

What board are you using?

RemyEE, 2019-04-01:
@scruffynerf I am using ESP32-Lyrat v4.3

jason-mao, 2019-04-01:
@RemyEE We can't reproduced your issue, how is size of the wav file ? @zhaofx1234 Is it same with RemyEE? Only 1 second wav file in Sdcard with `pipeline_wav_sdcard ` example.

zhaofx1234, 2019-04-01:
@jason-mao my wav file is only 8 seconds.

jason-mao, 2019-04-01:
@zhaofx1234 could you help to provide your log file

RemyEE, 2019-04-01:
The file size is 182,316 (bytes).
[REC.zip](https://github.com/espressif/esp-adf/files/3027749/REC.zip)
Here is the log
```
I (0) cpu_start: App cpu up.
I (247) heap_init: Initializing. RAM available for dynamic allocation:
I (254) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (260) heap_init: At 3FFB34A8 len 0002CB58 (178 KiB): DRAM
I (266) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (272) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (279) heap_init: At 4008B6A4 len 0001495C (82 KiB): IRAM
I (285) cpu_start: Pro cpu start user code
I (303) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (304) REC_WAV_SDCARD: [ 1 ] Mount sdcard
I (404) REC_WAV_SDCARD: [ 2 ] Start codec chip
E (404) gpio: gpio_install_isr_service(394): GPIO isr service already installed
I (424) REC_WAV_SDCARD: [3.0] Create audio pipeline for recording
I (424) REC_WAV_SDCARD: [3.1] Create fatfs stream to write data to sdcard
I (424) REC_WAV_SDCARD: [3.2] Create i2s stream to read audio data from codec ch
ip
I (434) REC_WAV_SDCARD: [3.3] Create wav encoder to encode wav format
I (444) REC_WAV_SDCARD: [3.4] Register all elements to audio pipeline
I (444) REC_WAV_SDCARD: [3.5] Link it together [codec_chip]-->i2s_stream-->wav_e
ncoder-->fatfs_stream-->[sdcard]
I (454) REC_WAV_SDCARD: [3.6] Setup uri (file as fatfs_stream, wav as wav encode
r)
I (464) REC_WAV_SDCARD: [ 4 ] Setup event listener
I (474) REC_WAV_SDCARD: [4.1] Listening event from pipeline
I (474) REC_WAV_SDCARD: [4.2] Listening event from peripherals
I (484) REC_WAV_SDCARD: [ 5 ] Start audio_pipeline
I (494) REC_WAV_SDCARD: [ 6 ] Listen for all pipeline events, record for 20 Seco
nds
I (1654) REC_WAV_SDCARD: [ * ] Recording ... 1
I (2744) REC_WAV_SDCARD: [ * ] Recording ... 2
I (3814) REC_WAV_SDCARD: [ * ] Recording ... 3
I (4914) REC_WAV_SDCARD: [ * ] Recording ... 4
I (5934) REC_WAV_SDCARD: [ * ] Recording ... 5
I (7024) REC_WAV_SDCARD: [ * ] Recording ... 6
I (8024) REC_WAV_SDCARD: [ * ] Recording ... 7
I (9044) REC_WAV_SDCARD: [ * ] Recording ... 8
I (10084) REC_WAV_SDCARD: [ * ] Recording ... 9
I (11084) REC_WAV_SDCARD: [ * ] Recording ... 10
I (12104) REC_WAV_SDCARD: [ * ] Recording ... 11
I (13134) REC_WAV_SDCARD: [ * ] Recording ... 12
I (14144) REC_WAV_SDCARD: [ * ] Recording ... 13
I (15164) REC_WAV_SDCARD: [ * ] Recording ... 14
I (16174) REC_WAV_SDCARD: [ * ] Recording ... 15
I (17184) REC_WAV_SDCARD: [ * ] Recording ... 16
I (18294) REC_WAV_SDCARD: [ * ] Recording ... 17
I (19304) REC_WAV_SDCARD: [ * ] Recording ... 18
I (20404) REC_WAV_SDCARD: [ * ] Recording ... 19
I (21404) REC_WAV_SDCARD: [ * ] Recording ... 20
I (21404) REC_WAV_SDCARD: [ 7 ] Stop audio_pipeline
W (21914) AUDIO_ELEMENT: OUT-[wav] AEL_IO_ABORT
W (21914) AUDIO_ELEMENT: OUT-[i2s] AEL_IO_ABORT
W (21914) AUDIO_PIPELINE: There are no listener registered
```

RemyEE, 2019-04-01:
I am using 124MB sd card, FAT32 format.

jason-mao, 2019-04-01:
@RemyEE  Did you test with other size SDCARD, e.g. 4G 8G sdcard FAT32 format.

RemyEE, 2019-04-01:
@jason-mao 
I think the SD card format caused the problem.
After checking the SD card format, I changed it to FAT. It's seems working properly now.
Here is the result and the log.
[REC.zip](https://github.com/espressif/esp-adf/files/3027764/REC.zip)
```
I (0) cpu_start: App cpu up.
I (247) heap_init: Initializing. RAM available for dynamic allocation:
I (254) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (260) heap_init: At 3FFB34A8 len 0002CB58 (178 KiB): DRAM
I (266) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM
I (272) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (279) heap_init: At 4008B6A4 len 0001495C (82 KiB): IRAM
I (285) cpu_start: Pro cpu start user code
I (303) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (304) REC_WAV_SDCARD: [ 1 ] Mount sdcard
I (404) REC_WAV_SDCARD: [ 2 ] Start codec chip
E (404) gpio: gpio_install_isr_service(394): GPIO isr service already installed
I (424) REC_WAV_SDCARD: [3.0] Create audio pipeline for recording
I (424) REC_WAV_SDCARD: [3.1] Create fatfs stream to write data to sdcard
I (424) REC_WAV_SDCARD: [3.2] Create i2s stream to read audio data from codec chip
I (434) REC_WAV_SDCARD: [3.3] Create wav encoder to encode wav format
I (444) REC_WAV_SDCARD: [3.4] Register all elements to audio pipeline
I (444) REC_WAV_SDCARD: [3.5] Link it together [codec_chip]-->i2s_stream-->wav_encoder-->fatfs_stream-
->[sdcard]
I (454) REC_WAV_SDCARD: [3.6] Setup uri (file as fatfs_stream, wav as wav encoder)
I (464) REC_WAV_SDCARD: [ 4 ] Setup event listener
I (474) REC_WAV_SDCARD: [4.1] Listening event from pipeline
I (474) REC_WAV_SDCARD: [4.2] Listening event from peripherals
I (484) REC_WAV_SDCARD: [ 5 ] Start audio_pipeline
I (494) REC_WAV_SDCARD: [ 6 ] Listen for all pipeline events, record for 20 Seconds
I (1514) REC_WAV_SDCARD: [ * ] Recording ... 1
I (2514) REC_WAV_SDCARD: [ * ] Recording ... 2
I (3514) REC_WAV_SDCARD: [ * ] Recording ... 3
I (4514) REC_WAV_SDCARD: [ * ] Recording ... 4
I (5514) REC_WAV_SDCARD: [ * ] Recording ... 5
I (6514) REC_WAV_SDCARD: [ * ] Recording ... 6
I (7514) REC_WAV_SDCARD: [ * ] Recording ... 7
I (8514) REC_WAV_SDCARD: [ * ] Recording ... 8
I (9514) REC_WAV_SDCARD: [ * ] Recording ... 9
I (10514) REC_WAV_SDCARD: [ * ] Recording ... 10
I (11514) REC_WAV_SDCARD: [ * ] Recording ... 11
I (12514) REC_WAV_SDCARD: [ * ] Recording ... 12
I (13514) REC_WAV_SDCARD: [ * ] Recording ... 13
I (14514) REC_WAV_SDCARD: [ * ] Recording ... 14
I (15514) REC_WAV_SDCARD: [ * ] Recording ... 15
I (16514) REC_WAV_SDCARD: [ * ] Recording ... 16
I (17514) REC_WAV_SDCARD: [ * ] Recording ... 17
I (18514) REC_WAV_SDCARD: [ * ] Recording ... 18
I (19514) REC_WAV_SDCARD: [ * ] Recording ... 19
I (20514) REC_WAV_SDCARD: [ * ] Recording ... 20
I (20514) REC_WAV_SDCARD: [ 7 ] Stop audio_pipeline
W (20534) AUDIO_PIPELINE: There are no listener registered
```

zhaofx1234, 2019-04-01:
I am using 2G sd card, FAT32 format. The following is my log and wav files.

[log_wav.zip](https://github.com/espressif/esp-adf/files/3027781/log_wav.zip)

RemyEE, 2019-04-01:
@jason-mao Don't think I have 4GB or 8GB SD card for testing now.
But I did test on 16GB SD card with FAT32 format. The result last 15s instead of 20s.
[REC.zip](https://github.com/espressif/esp-adf/files/3027782/REC.zip)

zhaofx1234, 2019-04-01:
@jason-mao I have tried the FAT and FAT32 format, and both of them still no work well.

brendenvogt, 2019-04-17:
Try a faster speed card. I tried with a 2gb very cheap card, and had same issue as @RemyEE , then tried with a 32gb class 10 samsung evo card and it worked first try. Also FAT32 vs FAT doesnt seem to affect anything, both work just fine on my samsung card.

closed 2022-02-08 by jason-mao